openwrt/packages #5730: acme: please support dns-01 challenge

bjmgeek, 2018-03-08:
Please support the dns-01 challenge for acme.  

Maintainer: @tohojo 
Environment:
BUILD_ID="r4882-f295db6e46"
LEDE_BOARD="ramips/mt76x8"
LEDE_ARCH="mipsel_24kc"
(router is a vocore2)

Description:

I'd like to support the dns-01 challenge.

tohojo, 2018-03-08:
Brian Minton <notifications@github.com> writes:

> Please support the dns-01 challenge for acme.  
>
> Maintainer: @tohojo 
> Environment:
> BUILD_ID="r4882-f295db6e46"
> LEDE_BOARD="ramips/mt76x8"
> LEDE_ARCH="mipsel_24kc"
> (router is a vocore2)
>
>
> Description:
>
> I'd like to support the dns-01 challenge.

Patches are very welcome :)

-Toke

bjmgeek, 2018-03-08:
I'll take a look :-)

closed 2018-08-18 by dibdot