openwrt/packages #5741: wifidog-ng: build failure with x86 and x86_64

KazukiShimada, 2018-03-11:
Maintainer: @zhaojh329
Environment: x86, x86_64, OpenWrt version 4e8f1e9f4ca088542fd2b861ea2f1a9dca0d845f

Description:
failed to build wifidog-ng with x86 and x86_64. Other targets, ar71xx, ixp4xx, ramips mt7621 are OK.
Big differences between failure and OK are gcc version and kernel version. 

P.S. Wifidog-ng is nice hack!

```
make[4]: Entering directory '/home/zukky/openwrt/openwrt/build_dir/target-x86_64_musl/libuhttpd-nossl/libuhttpd-2.0.0'
make[5]: Entering directory '/home/zukky/openwrt/openwrt/build_dir/target-x86_64_musl/libuhttpd-nossl/libuhttpd-2.0.0'
make[6]: Entering directory '/home/zukky/openwrt/openwrt/build_dir/target-x86_64_musl/libuhttpd-nossl/libuhttpd-2.0.0'
make[6]: Leaving directory '/home/zukky/openwrt/openwrt/build_dir/target-x86_64_musl/libuhttpd-nossl/libuhttpd-2.0.0'
make[6]: Entering directory '/home/zukky/openwrt/openwrt/build_dir/target-x86_64_musl/libuhttpd-nossl/libuhttpd-2.0.0'
[  8%] Building C object src/CMakeFiles/uhttpd.dir/lua_template.c.o
/home/zukky/openwrt/openwrt/build_dir/target-x86_64_musl/libuhttpd-nossl/libuhttpd-2.0.0/src/lua_template.c: In function 'uh_lua_send':
/home/zukky/openwrt/openwrt/build_dir/target-x86_64_musl/libuhttpd-nossl/libuhttpd-2.0.0/src/lua_template.c:555:14: error: cast to pointer from integer of different size [-Werror=int-to-pointer-cast]
         cl = (struct uh_client *)lua_tointeger(L, -1);
              ^
cc1: all warnings being treated as errors
src/CMakeFiles/uhttpd.dir/build.make:206: recipe for target 'src/CMakeFiles/uhttpd.dir/lua_template.c.o' failed
make[6]: *** [src/CMakeFiles/uhttpd.dir/lua_template.c.o] Error 1
make[6]: Leaving directory '/home/zukky/openwrt/openwrt/build_dir/target-x86_64_musl/libuhttpd-nossl/libuhttpd-2.0.0'
CMakeFiles/Makefile2:85: recipe for target 'src/CMakeFiles/uhttpd.dir/all' failed
make[5]: *** [src/CMakeFiles/uhttpd.dir/all] Error 2
make[5]: Leaving directory '/home/zukky/openwrt/openwrt/build_dir/target-x86_64_musl/libuhttpd-nossl/libuhttpd-2.0.0'
Makefile:129: recipe for target 'all' failed
make[4]: *** [all] Error 2
make[4]: Leaving directory '/home/zukky/openwrt/openwrt/build_dir/target-x86_64_musl/libuhttpd-nossl/libuhttpd-2.0.0'
Makefile:79: recipe for target '/home/zukky/openwrt/openwrt/build_dir/target-x86_64_musl/libuhttpd-nossl/libuhttpd-2.0.0/.built' failed
make[3]: *** [/home/zukky/openwrt/openwrt/build_dir/target-x86_64_musl/libuhttpd-nossl/libuhttpd-2.0.0/.built] Error 2
make[3]: Leaving directory '/home/zukky/openwrt/openwrt/feeds/packages/libs/libuhttpd'
package/Makefile:107: recipe for target 'package/feeds/packages/libuhttpd/compile' failed
make[2]: *** [package/feeds/packages/libuhttpd/compile] Error 2
make[2]: Leaving directory '/home/zukky/openwrt/openwrt'
package/Makefile:103: recipe for target '/home/zukky/openwrt/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile' failed
make[1]: *** [/home/zukky/openwrt/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/zukky/openwrt/openwrt'
/home/zukky/openwrt/openwrt/include/toplevel.mk:216: ターゲット 'world' のレシピで失敗しました
make: *** [world] エラー 2

```

hnyman, 2018-03-11:
Possibly fixed by #5745 (but I am not quite sure if the underlying fix (  https://github.com/zhaojh329/libuhttpd/commit/2a204e7dace502c844fe0b122becbfe073aefad0  ) is quite safe as that simply removes the warning but leaves the actual behaviour as it was. )

zhaojh329, 2018-03-12:
But I need to convert the integer to the structure pointer. I haven't found any other ways. Can you help me?

zhaojh329, 2018-03-12:
Fixed.
See https://github.com/zhaojh329/libuhttpd/commit/d2439b6f5c0be4a60dce094bd1fbdde6c79f40b0

KazukiShimada, 2018-03-12:
@zhaojh329
@hnyman

Thanks to all.
Unfortunalely, another error happened...
`
make[5]: Leaving directory '/home/zukky/openwrt/openwrt/build_dir/target-x86_64_musl/wifidog-ng-nossl/wifidog-ng-1.1.3'
make[4]: Leaving directory '/home/zukky/openwrt/openwrt/build_dir/target-x86_64_musl/wifidog-ng-nossl/wifidog-ng-1.1.3'
make -C /home/zukky/openwrt/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-4.14.25 HOSTCFLAGS="-O2 -I/home/zukky/openwrt/openwrt/staging_dir/host/include -I/home/zukky/openwrt/openwrt/staging_dir/hostpkg/include -I/home/zukky/openwrt/openwrt/staging_dir/target-x86_64_musl/host/include -Wall -Wmissing-prototypes -Wstrict-prototypes" CROSS_COMPILE="x86_64-openwrt-linux-musl-" ARCH="x86" KBUILD_HAVE_NLS=no KBUILD_BUILD_USER="" KBUILD_BUILD_HOST="" KBUILD_BUILD_TIMESTAMP="Sun Mar 11 22:42:38 2018" KBUILD_BUILD_VERSION="0" HOST_LOADLIBES="-L/home/zukky/openwrt/openwrt/staging_dir/host/lib" CONFIG_SHELL="bash" V=''  cmd_syscalls= KERNELRELEASE=4.14.25 CC="x86_64-openwrt-linux-musl-gcc" SUBDIRS="/home/zukky/openwrt/openwrt/build_dir/target-x86_64_musl/wifidog-ng-nossl/wifidog-ng-1.1.3"/kmod modules
make[4]: Entering directory '/home/zukky/openwrt/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-4.14.25'
  CC [M]  /home/zukky/openwrt/openwrt/build_dir/target-x86_64_musl/wifidog-ng-nossl/wifidog-ng-1.1.3/kmod/main.o
/home/zukky/openwrt/openwrt/build_dir/target-x86_64_musl/wifidog-ng-nossl/wifidog-ng-1.1.3/kmod/main.c: In function 'wifidog_init':
/home/zukky/openwrt/openwrt/build_dir/target-x86_64_musl/wifidog-ng-nossl/wifidog-ng-1.1.3/kmod/main.c:233:11: error: implicit declaration of function 'nf_register_hooks'; did you mean 'nf_register_net_hooks'? [-Werror=implicit-function-declaration]
     ret = nf_register_hooks(wifidog_ops, ARRAY_SIZE(wifidog_ops));
           ^~~~~~~~~~~~~~~~~
           nf_register_net_hooks
/home/zukky/openwrt/openwrt/build_dir/target-x86_64_musl/wifidog-ng-nossl/wifidog-ng-1.1.3/kmod/main.c: In function 'wifidog_exit':
/home/zukky/openwrt/openwrt/build_dir/target-x86_64_musl/wifidog-ng-nossl/wifidog-ng-1.1.3/kmod/main.c:259:5: error: implicit declaration of function 'nf_unregister_hooks'; did you mean 'nf_unregister_net_hooks'? [-Werror=implicit-function-declaration]
     nf_unregister_hooks(wifidog_ops, ARRAY_SIZE(wifidog_ops));
     ^~~~~~~~~~~~~~~~~~~
     nf_unregister_net_hooks
cc1: some warnings being treated as errors
scripts/Makefile.build:320: recipe for target '/home/zukky/openwrt/openwrt/build_dir/target-x86_64_musl/wifidog-ng-nossl/wifidog-ng-1.1.3/kmod/main.o' failed
make[5]: *** [/home/zukky/openwrt/openwrt/build_dir/target-x86_64_musl/wifidog-ng-nossl/wifidog-ng-1.1.3/kmod/main.o] Error 1
Makefile:1506: recipe for target '_module_/home/zukky/openwrt/openwrt/build_dir/target-x86_64_musl/wifidog-ng-nossl/wifidog-ng-1.1.3/kmod' failed
make[4]: *** [_module_/home/zukky/openwrt/openwrt/build_dir/target-x86_64_musl/wifidog-ng-nossl/wifidog-ng-1.1.3/kmod] Error 2
make[4]: Leaving directory '/home/zukky/openwrt/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-4.14.25'
Makefile:98: recipe for target '/home/zukky/openwrt/openwrt/build_dir/target-x86_64_musl/wifidog-ng-nossl/wifidog-ng-1.1.3/.built' failed
make[3]: *** [/home/zukky/openwrt/openwrt/build_dir/target-x86_64_musl/wifidog-ng-nossl/wifidog-ng-1.1.3/.built] Error 2
make[3]: Leaving directory '/home/zukky/openwrt/openwrt/feeds/packages/net/wifidog-ng'
package/Makefile:107: recipe for target 'package/feeds/packages/wifidog-ng/compile' failed
make[2]: *** [package/feeds/packages/wifidog-ng/compile] Error 2
make[2]: Leaving directory '/home/zukky/openwrt/openwrt'
package/Makefile:103: recipe for target '/home/zukky/openwrt/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile' failed
make[1]: *** [/home/zukky/openwrt/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/zukky/openwrt/openwrt'
/home/zukky/openwrt/openwrt/include/toplevel.mk:216: ターゲット 'world' のレシピで失敗しました
make: *** [world] エラー 2
`

zhaojh329, 2018-03-12:
Fixed.
See https://github.com/zhaojh329/wifidog-ng/commit/9ce158d9f1ee9d5d5ecbd73edde481f308285540

PR #5748 (2018-03-12): libuhttpd: Update to 2.0.2

Maintainer: me
Compile tested: (mipsel,miwifi-mini, master)
Run tested: (mipsel,miwifi-mini, master)

Description:
Fix #5741
PR #5749 (2018-03-12): wifidog-ng: Update to 1.1.4

Maintainer: me
Compile tested: (mipsel,miwifi-mini, master)
Run tested: (mipsel,miwifi-mini, master)

Description:
Fix #5741
closed 2018-03-12 by hnyman