openwrt/packages #5753: haveged:

knacky, 2018-03-12:
Maintainer: @\hnyman
Environment: (ar71xx, wndr3800, OpenWrt master)
Host: Debian 8.10 (Linux ts140 3.16.0-5-amd64 #1 SMP Debian 3.16.51-3+deb8u1 (2018-01-08) x86_64 GNU/Linux)

Description: haveged package fails to build under new GCC 7.3.  Previously built ok under older GCC 5.4 and 5.5 when compiling from master.

[dotconfig.txt](https://github.com/openwrt/packages/files/1804162/dotconfig.txt)

```
make[3]: Leaving directory '/home/matt/openwrt/package/boot/uboot-envtools'
make[3]: Entering directory '/home/matt/openwrt/feeds/packages/utils/haveged'
rm -f /home/matt/openwrt/build_dir/target-mips_24kc_musl/haveged-1.9.2/.built
touch /home/matt/openwrt/build_dir/target-mips_24kc_musl/haveged-1.9.2/.built_check
CFLAGS="-Os -pipe -mno-branch-likely -mips32r2 -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -mips16 -minterlink-mips16 -iremap/home/matt/openwrt/build_dir/target-mips_24kc_musl/haveged-1.9.2:haveged-1.9.2 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro  -I/home/matt/openwrt/staging_dir/target-mips_24kc_musl/usr/include -I/home/matt/openwrt/staging_dir/target-mips_24kc_musl/include -I/home/matt/openwrt/staging_dir/toolchain-mips_24kc_gcc-7.3.0_musl/usr/include -I/home/matt/openwrt/staging_dir/toolchain-mips_24kc_gcc-7.3.0_musl/include/fortify -I/home/matt/openwrt/staging_dir/toolchain-mips_24kc_gcc-7.3.0_musl/include " CXXFLAGS="-Os -pipe -mno-branch-likely -mips32r2 -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -mips16 -minterlink-mips16 -iremap/home/matt/openwrt/build_dir/target-mips_24kc_musl/haveged-1.9.2:haveged-1.9.2 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro  -I/home/matt/openwrt/staging_dir/target-mips_24kc_musl/usr/include -I/home/matt/openwrt/staging_dir/target-mips_24kc_musl/include -I/home/matt/openwrt/staging_dir/toolchain-mips_24kc_gcc-7.3.0_musl/usr/include -I/home/matt/openwrt/staging_dir/toolchain-mips_24kc_gcc-7.3.0_musl/include/fortify -I/home/matt/openwrt/staging_dir/toolchain-mips_24kc_gcc-7.3.0_musl/include " LDFLAGS="-L/home/matt/openwrt/staging_dir/target-mips_24kc_musl/usr/lib -L/home/matt/openwrt/staging_dir/target-mips_24kc_musl/lib -L/home/matt/openwrt/staging_dir/toolchain-mips_24kc_gcc-7.3.0_musl/usr/lib -L/home/matt/openwrt/staging_dir/toolchain-mips_24kc_gcc-7.3.0_musl/lib -znow -zrelro " make  -C /home/matt/openwrt/build_dir/target-mips_24kc_musl/haveged-1.9.2/. AR="mips-openwrt-linux-musl-gcc-ar" AS="mips-openwrt-linux-musl-gcc -c -Os -pipe -mno-branch-likely -mips32r2 -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -iremap/home/matt/openwrt/build_dir/target-mips_24kc_musl/haveged-1.9.2:haveged-1.9.2 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" LD=mips-openwrt-linux-musl-ld NM="mips-openwrt-linux-musl-gcc-nm" CC="mips-openwrt-linux-musl-gcc" GCC="mips-openwrt-linux-musl-gcc" CXX="mips-openwrt-linux-musl-g++" RANLIB="mips-openwrt-linux-musl-gcc-ranlib" STRIP=mips-openwrt-linux-musl-strip OBJCOPY=mips-openwrt-linux-musl-objcopy OBJDUMP=mips-openwrt-linux-musl-objdump SIZE=mips-openwrt-linux-musl-size CROSS="mips-openwrt-linux-musl-" ARCH="mips" ;
make[4]: Entering directory '/home/matt/openwrt/build_dir/target-mips_24kc_musl/haveged-1.9.2'
make  all-recursive
make[5]: Entering directory '/home/matt/openwrt/build_dir/target-mips_24kc_musl/haveged-1.9.2'
Making all in src
make[6]: Entering directory '/home/matt/openwrt/build_dir/target-mips_24kc_musl/haveged-1.9.2/src'
/bin/bash ../libtool  --tag=CC   --mode=compile mips-openwrt-linux-musl-gcc -DHAVE_CONFIG_H -I. -I..   -I/home/matt/openwrt/staging_dir/target-mips_24kc_musl/usr/include -I/home/matt/openwrt/staging_dir/target-mips_24kc_musl/include -I/home/matt/openwrt/staging_dir/toolchain-mips_24kc_gcc-7.3.0_musl/usr/include -I/home/matt/openwrt/staging_dir/toolchain-mips_24kc_gcc-7.3.0_musl/include/fortify -I/home/matt/openwrt/staging_dir/toolchain-mips_24kc_gcc-7.3.0_musl/include  -Wall -I.. -Os -pipe -mno-branch-likely -mips32r2 -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -mips16 -minterlink-mips16 -iremap/home/matt/openwrt/build_dir/target-mips_24kc_musl/haveged-1.9.2:haveged-1.9.2 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro  -c -o libhavege_la-havegecollect.lo `test -f 'havegecollect.c' || echo './'`havegecollect.c
libtool: compile:  mips-openwrt-linux-musl-gcc -DHAVE_CONFIG_H -I. -I.. -I/home/matt/openwrt/staging_dir/target-mips_24kc_musl/usr/include -I/home/matt/openwrt/staging_dir/target-mips_24kc_musl/include -I/home/matt/openwrt/staging_dir/toolchain-mips_24kc_gcc-7.3.0_musl/usr/include -I/home/matt/openwrt/staging_dir/toolchain-mips_24kc_gcc-7.3.0_musl/include/fortify -I/home/matt/openwrt/staging_dir/toolchain-mips_24kc_gcc-7.3.0_musl/include -Wall -I.. -Os -pipe -mno-branch-likely -mips32r2 -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -mips16 -minterlink-mips16 -iremap/home/matt/openwrt/build_dir/target-mips_24kc_musl/haveged-1.9.2:haveged-1.9.2 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -c havegecollect.c  -fPIC -DPIC -o .libs/libhavege_la-havegecollect.o
{standard input}: Assembler messages:
{standard input}:31447: Error: operand value out of range for instruction
{standard input}:31482: Error: operand value out of range for instruction
{standard input}:31518: Error: operand value out of range for instruction
{standard input}:31554: Error: operand value out of range for instruction
{standard input}:31590: Error: operand value out of range for instruction
{standard input}:31626: Error: operand value out of range for instruction
{standard input}:31662: Error: operand value out of range for instruction
{standard input}:31698: Error: operand value out of range for instruction
{standard input}:31734: Error: operand value out of range for instruction
{standard input}:31770: Error: operand value out of range for instruction
{standard input}:31806: Error: operand value out of range for instruction
{standard input}:31842: Error: operand value out of range for instruction
{standard input}:31878: Error: operand value out of range for instruction
{standard input}:31914: Error: operand value out of range for instruction
{standard input}:31950: Error: operand value out of range for instruction
{standard input}:31986: Error: operand value out of range for instruction
{standard input}:32022: Error: operand value out of range for instruction
{standard input}:32058: Error: operand value out of range for instruction
Makefile:540: recipe for target 'libhavege_la-havegecollect.lo' failed
make[6]: *** [libhavege_la-havegecollect.lo] Error 1
make[6]: Leaving directory '/home/matt/openwrt/build_dir/target-mips_24kc_musl/haveged-1.9.2/src'
Makefile:478: recipe for target 'all-recursive' failed
make[5]: *** [all-recursive] Error 1
make[5]: Leaving directory '/home/matt/openwrt/build_dir/target-mips_24kc_musl/haveged-1.9.2'
Makefile:374: recipe for target 'all' failed
make[4]: *** [all] Error 2
make[4]: Leaving directory '/home/matt/openwrt/build_dir/target-mips_24kc_musl/haveged-1.9.2'
Makefile:64: recipe for target '/home/matt/openwrt/build_dir/target-mips_24kc_musl/haveged-1.9.2/.built' failed
make[3]: *** [/home/matt/openwrt/build_dir/target-mips_24kc_musl/haveged-1.9.2/.built] Error 2
make[3]: Leaving directory '/home/matt/openwrt/feeds/packages/utils/haveged'
package/Makefile:107: recipe for target 'package/feeds/packages/haveged/compile' failed
make[2]: *** [package/feeds/packages/haveged/compile] Error 2
make[2]: Leaving directory '/home/matt/openwrt'
package/Makefile:103: recipe for target '/home/matt/openwrt/staging_dir/target-mips_24kc_musl/stamp/.package_compile' failed
make[1]: *** [/home/matt/openwrt/staging_dir/target-mips_24kc_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/matt/openwrt'
/home/matt/openwrt/include/toplevel.mk:216: recipe for target 'world' failed
make: *** [world] Error 2
```

hnyman, 2018-03-12:
haveged builds ok with GCC 7.3 and musl 1.1.19 for some targets like
mvebu (arm_cortex-a9_vfpv3) and
ipq806x (arm_cortex-a15_neon-vfpv4)

But it seems to fail for ar71xx (mips_24kc)

nbd168, 2018-03-12:
DIsabling MIPS16 for this package might help

hnyman, 2018-03-12:
quick googling revealed that this might be related to MIPS16 code generation.
https://gcc.gnu.org/bugzilla/show_bug.cgi?id=77757

Just testing it...
MIPS16 disabling helps

hnyman, 2018-03-12:
Should be fixed with 5182ddc16

closed 2018-03-12 by hnyman

cotequeiroz, 2018-03-21:
If you want to keep MIPS16 code generation, you may just disable it in the havege_gather function with __attribute__((nomips16))
```
--- a/src/havegecollect.c
+++ b/src/havegecollect.c
@@ -140,7 +140,11 @@ static LOOP_BRANCH havege_cp(H_COLLECT *
  * Protect the collection mechanism against ever-increasing gcc optimization
  */
 #if defined (GCC_VERSION) && GCC_VERSION >= 40400
+# if defined (__mips) && GCC_VERSION >= 70000
+static int havege_gather(H_COLLECT * h_ctxt) __attribute__((optimize(1),nomips16));
+# else
 static int havege_gather(H_COLLECT * h_ctxt) __attribute__((optimize(1)));
+# endif
 #else
 static int  havege_gather(H_COLLECT * h_ctxt);
 #endif
@@ -317,6 +321,9 @@ static LOOP_BRANCH havege_cp( /* RETURN:
  * The collection loop is constructed by repetitions of oneinteration.h interleaved
  * with control points generated by the LOOP macro.
  */
+#if defined (__mips) && GCC_VERSION >= 70000
+__attribute__((nomips16))     /* avoid a gcc7 bug  specific to mips16 code */
+#endif
 static int havege_gather(     /* RETURN: 1 if initialized    */
    H_COLLECT * h_ctxt)        /* IN:     collector context  */
 {

```