openwrt/packages #5754: luasocket not available for x86

tekoholic, 2018-03-13:
For quite some time, it seems that luasocket is available for all (or most) architectures except for the 32-bit x86.

Being depended upon by several other things, the particular one I have interest in being cjdns, this is not a good thing...

May I simply ask that it be built for x86, or for a reason why it cannot be?

dibdot, 2018-03-13:
Where did you look fo it? I found the current version of luasocket here:

https://downloads.openwrt.org/snapshots/packages/i386_pentium/packages/

tekoholic, 2018-03-14:
Well, um...

opkg install
https://downloads.openwrt.org/snapshots/packages/i386_pentium/packages/luasocket_3.0-rc1-20130
909-4_i386_pentium.ipk
Collected errors:
 * wfopen:
https://downloads.openwrt.org/snapshots/packages/i386_pentium/packages/luasocket_3.0-rc1-20130909-4_i386_pentium.ipk:
No such file or directory.
 * pkg_init_from_file: Failed to extract control file from
https://downloads.openwrt.org/snapshots/packages/i386_pentium/packages/luasocket_3.0-rc1-20130909-4_i386_pentium.ipk

tekoholic, 2018-03-14:
> Well, um...
>
> opkg install https://downloads.openwrt.org/snapshots/packages/i386_
> pentium/packages/luasocket_3.0-rc1-20130
> 909-4_i386_pentium.ipk
> Collected errors:
>  * wfopen: https://downloads.openwrt.org/snapshots/packages/i386_
> pentium/packages/luasocket_3.0-rc1-20130909-4_i386_pentium.ipk: No such
> file or directory.
>  * pkg_init_from_file: Failed to extract control file from
> https://downloads.openwrt.org/snapshots/packages/i386_
> pentium/packages/luasocket_3.0-rc1-20130909-4_i386_pentium.ipk.

Is this maybe an issue for LibreMesh? They (and I, based upon the
foundation) pawned it off on y'all...

tekoholic, 2018-03-14:
I might add, it installs and runs just fine on another router based upon
their "versions" of things (ar71xx wndr3700v2), but cannot install on the
primary test-rig *x86.32")...

dibdot, 2018-03-14:
Please check the architecture first with: opkg print-architecture (there exist two i386 targets)
After that install the package with "http", e.g.:
`opkg install http://downloads.openwrt.org/snapshots/packages/i386_pentium/packages/luasocket_3.0-rc1-20130909-4_i386_pentium.ipk`

jow-, 2018-03-14:
Also note that opkg does not understand `https://`, only `http://` urls by default

tekoholic, 2018-03-15:
opkg print-architecture gives:
arch all 1
arch noarch 1
arch i386_pentium4 10

To avoid the lack of opkg https support, I used aria2c http://downloads.openwrt.org/snapshots/packages/i386_pentium/packages/luasocket_3.0-rc1-201
30909-4_i386_pentium.ipk, then opkg install from local file. This presented:

Unknown package 'luasocket'.
Collected errors:
 * pkg_hash_fetch_best_installation_candidate: Packages for luasocket found, but incompatible with the architectures configured
 * opkg_install_cmd: Cannot install package luasocket.

This makes sense of course, as ...pentium and ...pentium4 do differ. Once again, am I maybe looking at something LibreMesh specific here? In that they've named their arch slightly differently?

dibdot, 2018-03-15:
You've never mentioned before if you are on release or snapshot level. The build for 17.01.x fails due to a compile error, see here: http://downloads.lede-project.org/releases/faillogs/i386_pentium4/packages/luasocket/compile.txt

tekoholic, 2018-03-15:
You're absolutely correct, I hadn't mentioned that yet. At this point, issue is edited to reflect appropriately.

Still gotta wonder, if it was a known bug, how it is that there was not an issue already...

neheb, 2020-01-12:
LEDE is EOL. I fixed this for 18.06 and above. Please update.

closed 2020-01-12 by neheb