openwrt/packages #5759: mwan3:  lede-17.01.4 x86_64-ext4 does not start

sitcsitc, 2018-03-14:
Using a pcengines board with lede-17.01.4-x86-64-combined-ext4.img
mwan3 installed and configured for failover between two wan interfaces, one on eth1 metric1 and eth2 metric2. 
Error: Missing both of the required interface IP rules
Also, the mwan3track processes are not running.

Everything configured as known to work on v15.05.1, here is the config:
config interface 'wan'
	option enabled '1'
	option reliability '2'
	option count '1'
	option timeout '2'
	option interval '5'
	option down '3'
	option up '8'
	list track_ip '8.8.8.8'
	list track_ip '208.67.220.220'

config interface 'wan2'
	option reliability '1'
	option count '1'
	option timeout '2'
	option interval '5'
	option down '3'
	option up '8'
	list track_ip '8.8.4.4'
	list track_ip '208.67.222.222'
	option enabled '1'

config member 'wan_m1_w3'
	option interface 'wan'
	option metric '1'
	option weight '3'

config member 'wan_m2_w3'
	option interface 'wan'
	option metric '2'
	option weight '3'

config member 'wan2_m1_w2'
	option interface 'wan2'
	option metric '1'
	option weight '2'

config member 'wan2_m2_w2'
	option interface 'wan2'
	option metric '2'
	option weight '2'

config policy 'wan_only'
	list use_member 'wan_m1_w3'

config policy 'wan2_only'
	list use_member 'wan2_m1_w2'

config policy 'balanced'
	list use_member 'wan_m1_w3'
	list use_member 'wan2_m1_w2'

config policy 'wan_wan2'
	list use_member 'wan_m1_w3'
	list use_member 'wan2_m2_w2'

config policy 'wan2_wan'
	list use_member 'wan_m2_w3'
	list use_member 'wan2_m1_w2'

config rule 'failover'
	option dest_ip '0.0.0.0/0'
	option proto 'all'
	option sticky '0'
	option use_policy 'wan_wan2'

feckert, 2018-03-15:
I need more information. To help you. Any debug messages from the system?
Have you installed ip-full ip-tiny?
We have noticed that the dependency was only set to ip.
But we need ip-full or ip-tiny to get the thing working

bluewavenet, 2018-03-27:
@sitcsitc 
Try adding:
config globals 'globals'
	option enabled '1'
to the config file then try restarting

feckert, 2018-03-28:
@sitcsitc @bluewavenet you are working on lede-17.01 so there is no globals section it was added in the master tree

feckert, 2018-04-03:
@sitcsitc ping?

sitcsitc, 2018-04-05:
@feckert 
I already put this hardware into production with v15 so I can't test right now. When I get more hardware in-hand, I will run through this again and report back. I had installed ip separately in this case so it would have been ip-full.

feckert, 2019-03-04:
@hnyman please close this issue

closed 2019-03-04 by hnyman

feckert, 2019-03-04:
@hnyman thanks :+1: