openwrt/packages #5760: mwan3: when using pppoe, while in failover state, local DNS is broken

sitcsitc, 2018-03-14:
Using lede-17.01.4-ar71xx-generic-wndr3800
If either of the failover interfaces are set to pppoe and the mwan3 is in a state of failover, local DNS is broken.

As a test I changed primary failover from wan_wan2 to wan2_wan and in either case, while in failover state, DNS was broken.

You can use external DNS servers to resolve but not dnsmasq (local dns). 

mwan3track processes are running:
 5041 root      1192 S    {mwan3track} /bin/sh /usr/sbin/mwan3track wan2 eth0.2 1 1 2 5 3 8 208.67.220.220 8.8.8.8
13801 root      1204 S    /usr/sbin/pppd nodetach ipparam wan ifname pppoe-wan +ipv6 set AUTOIPV6=1 nodefaultroute usepeerdns maxfail 1 user si
13965 root      1192 S    {mwan3track} /bin/sh /usr/sbin/mwan3track wan pppoe-wan 2 1 2 5 3 8 208.67.222.222 8.8.4.4

This setup is known to work on version 15.05.1. Here is what I am using.
config interface 'wan'
	option enabled '1'
	option reliability '2'
	option count '1'
	option timeout '2'
	option interval '5'
	option down '3'
	option up '8'
	list track_ip '8.8.4.4'
	list track_ip '208.67.222.222'

config interface 'wan2'
	list track_ip '8.8.8.8'
	list track_ip '208.67.220.220'
	option reliability '1'
	option count '1'
	option timeout '2'
	option interval '5'
	option down '3'
	option up '8'
	option enabled '1'

config member 'wan_m1_w3'
	option interface 'wan'
	option metric '1'
	option weight '3'

config member 'wan_m2_w3'
	option interface 'wan'
	option metric '2'
	option weight '3'

config member 'wan2_m1_w2'
	option interface 'wan2'
	option metric '1'
	option weight '2'

config member 'wan2_m2_w2'
	option interface 'wan2'
	option metric '2'
	option weight '2'

config policy 'wan_only'
	list use_member 'wan_m1_w3'

config policy 'wan2_only'
	list use_member 'wan2_m1_w2'

config policy 'balanced'
	list use_member 'wan_m1_w3'
	list use_member 'wan2_m1_w2'

config policy 'wan_wan2'
	list use_member 'wan_m1_w3'
	list use_member 'wan2_m2_w2'

config policy 'wan2_wan'
	list use_member 'wan_m2_w3'
	list use_member 'wan2_m1_w2'

config rule 'failover'
	option dest_ip '0.0.0.0/0'
	option proto 'all'
	option sticky '0'
	option use_policy 'wan_wan2'

feckert, 2018-03-15:
I think It has to do with the self interface.
Have you set this up?
https://wiki.openwrt.org/doc/howto/mwan3#the_routable_loopback_self

feckert, 2018-04-03:
@sitcsitc ping?

lucize, 2018-04-03:
did you set manual dns on each interface and don't let the pppoe connection to override it ?
uncheck `Use DNS servers advertised by peer`

sitcsitc, 2018-04-05:
@feckert Thanks for the reminder and I apologize for not troubleshooting this earlier. I tried many things a few weeks ago and nothing was reproducible. Sometimes it was broken and others not. So I took a break and today started from scratch. I did nothing unusual, just "opkg install luci-app-mwan3", setup a vlan on eth0.2, added interface wan2 on vlan and added to wan zone, set metric 10 for eth1, metric 20 for wan2, enabled wan2 in mwan3 config, set mwan3 rule to failover "wan_wan2 ". That was all I did and it worked flawlessly. The DNS worked from the device and from downstream clients on the lan. 

I did mess around with that routable loopback self thing. It never worked and didn't make any sense to me. It is clear that is not needed in my case (failover).

I'm now wondering if the upgrade path from 15 to 17 was to blame partly. Anyhow, I would consider this closed.

feckert, 2018-07-05:
@sitcsitc so as you have opened the ticket could you please close this. I have not the privilege to do this.

closed 2018-07-05 by sitcsitc

kukulo2011, 2018-10-07:
I had the same issue with broken DNS while using multiple wan, while one connection was in failed state.
In interfaces it helped to **uncheck** the option **Use advertised DNS by peer** and **Use custom DNS servers: 8.8.8.8, 8.8.4.4** for both wan interfaces in the advanced options.

kukulo2011, 2018-10-07:
[https://imgur.com/0xdGmm8](url)