openwrt/packages #5761: mwan3: cannot use with ipv6 only interfaces

luizluca, 2018-03-14:
Maintainer: @feckert , @kubco2 
Environment: LEDE Reboot 17.01.4 r3560-79f57e422d x86

Description:

I have setup a IPv6 only wan. I plan to use multi-homing with mwan3. However, mwan3 does not allow me to use it if wan interface does not have a IPv4 default route. I added a dummy address to it in order to allow mwan3 to bring it up.

feckert, 2018-03-15:
Sorry for lede-17.01 and in the master there is no tracking support implemented for only ipv6.
You have always use an ipv4 tracking ip. But ipv6 routing should work.
This is already on the road map to implement ipv6 tracking parallel to ipv4 as will, but i have no free time to do this :-(

luizluca, 2018-03-19:
@feckert , ipv6 tracking do work. If I specify only ipv6 tracking addresses, it checks correctly the connectivity as ping works both with ipv4 and ipv6 addresses. It's just it simply doesn't launch a tracker if not ipv4 default route is set.

luizluca, 2018-03-19:
I don't know if it is related but when I set last_resort as unreachable for all policies, the tracker will always fail. 
Even specifying the wan interface as source in ping, ip6tables mangle rules still sets mark to 0xfe00. The result is that once failing, the tracker cannot recover the interface.

- Is this expected?
- Shouldn't it always use the main route table?
- Would I need to manually set a rule for tracker that falls back to main route table? (from router wan1 address to any wan1 tracker addresses, proto icmp/echo request, use wan1_only, fall back to main table)

feckert, 2018-03-20:
The ping is always bind to the interface with the -I option.
https://github.com/openwrt/packages/blob/0b748a3ac5960225ae35f24f7035ce64732d0d3f/net/mwan3/files/usr/sbin/mwan3track#L18-L23

> Even specifying the wan interface as source in ping, ip6tables mangle rules still sets mark to 0xfe00. The result is that once failing, the tracker cannot recover the interface.

Yes it is true if all interfaces are down they will **not** recover anymore.

> Is this expected?

This is a bug.
As I know i have fixed this in master and on lede-17.01
https://github.com/openwrt/packages/commit/282e90014e936ed1db527608e6ac4904a895b1ab#diff-34590d11545c84b961d1992ce3aefdeb

> ipv6 tracking do work. If I specify only ipv6 tracking addresses, it checks correctly the connectivity as ping works both with ipv4 and ipv6 addresses.

I have not tested this since i am the maintainer. This was done/implemented by @Adze1502. I have no ipv6 ony from my provider :-(. Only ds-lite. I will have a focus on ipv6 only in the next months

> It's just it simply doesn't launch a tracker if not ipv4 default route is set.

Because you are on lede-17.01 i have to look deeper into this issue i am not familiar with this version.
But I have found the following line in the hutplug scripts. Which check the default route.

https://github.com/openwrt/packages/blob/0b748a3ac5960225ae35f24f7035ce64732d0d3f/net/mwan3/files/etc/hotplug.d/iface/15-mwan3#L28-L40

Have you set the family option to ipv6?

luizluca, 2018-03-27:
Thanks @feckert for the answer.

> This is a bug.
> As I know i have fixed this in master and on lede-17.01
> 282e900#diff-34590d11545c84b961d1992ce3aefdeb
> 

I checked the diff but I'm not able to apply it manually into mwan3 2.0.2-1. The lines are simply too different. Git might have too many changed not released yet (and we may never have a lede17.01.5). I might simply build from git or avoid using blackrole/reject for rules that matches ping tests.

> Have you set the family option to ipv6?

You got it! I didn't know this family option for mwan3. None of wikis mention it:
https://openwrt.org/doc/uci/mwan3
https://wiki.openwrt.org/doc/howto/mwan3

Nor it is listed in example conf:

https://github.com/openwrt/packages/blob/master/net/mwan3/files/etc/config/mwan3 

Anyway, an interface in OpenWrt with static settings can have both IPv4 and IPv6 together. There is no need for a IPv6 dedicated (wan6), even though it is common because IPv4 and IPv6 might use a different proto. If mwan3 do need isolated interfaces, it might warn the user. if not, it might be adapted to work with interfaces that have only IPv4, only IPv6 or both.

And why not also create a wan6 in example conf? wan,wan2 might need to be renamed to something that do not collide with default wan,wan6, like: wan, wan6, wan_2, wan6_2.

feckert, 2018-03-28:
> And why not also create a wan6 in example conf? wan,wan2 might need to be renamed to something that do not collide with default wan,wan6, like: wan, wan6, wan_2, wan6_2.

We could to this. If you have an example config then open a PR.
The tracking implementation is not perfect.
From my point of view we have to change the tracking to force either ipv4 or ipv6.
https://github.com/openwrt/packages/blob/e911c7db9d3e1d5e9334fb93cef6dc9dbe80d14c/net/mwan3/files/usr/sbin/mwan3track#L107-L117

I am not sure what the policy in openwrt is.
A a common interface which is setting up both ipv4/ipv6 connection
or
Separated interfaces one is setting up ipv4 and the other is setting up ipv6

luizluca, 2018-03-29:
AFAIK, OpenWrt does not have a policy about it. Is simply allow both when it is tecnically possible. If mwan3 requires split interfaces, it could warn as it already does when metric is not specified.

feckert, 2018-04-01:
@luizluca what about the issue https://github.com/openwrt/packages/issues/5798 is this related to this?
Are problems solved or are there still problems with NDP if you also set the proto to ipv6?

luizluca, 2018-04-02:
> Are problems solved or are there still problems with NDP if you also set the proto to ipv6?

No, it is not related. I just tested it again. If I remove: 

```
ip6tables --table mangle -I OUTPUT 1 -p ipv6-icmp -m icmp6 --icmpv6-type 135
```

IPv6 interfaces might not work (normally they won't).

lowjoel, 2018-12-24:
I'm not sure if this is the same issue, but this is what came up in a GitHub search.

With respect to https://github.com/openwrt/packages/issues/5761#issuecomment-374507090, I'm finding that for an IPv6-only interface (a 6in4 tunnel), I can only specify a `track_ip` that is on the same link.

For example, in the following configuration in `/etc/config/mwan3`:
```
config interface 'wan6'
	option enabled '1'
	option family 'ipv6'
	option reliability '1'
	option count '1'
	# Google DNS
	list track_ip '2001:4860:4860::8888'
	list track_ip '2001:4860:4860::8844'
	# OpenDNS
	list track_ip '2620:0:ccd::2'
	list track_ip '2620:0:ccc::2'
```

I'll see messages like these in the syslog:
```
Mon Dec 24 19:29:28 2018 user.notice mwan3[5009]: Execute ifup event on interface wan6 (6in4-wan6)
Mon Dec 24 19:29:28 2018 user.notice mwan3[5009]: Starting tracker on interface wan6 (6in4-wan6)
Mon Dec 24 19:29:29 2018 user.info mwan3[5009]: connection tracking not flushed on interface wan6 (6in4-wan6) ifup
Mon Dec 24 19:29:32 2018 user.info mwan3track[5646]: Check (ping) failed for target "2001:4860:4860::8888" on interface wan6 (6in4-wan6)
Mon Dec 24 19:29:36 2018 user.info mwan3track[5646]: Check (ping) failed for target "2001:4860:4860::8844" on interface wan6 (6in4-wan6)
Mon Dec 24 19:29:40 2018 user.info mwan3track[5646]: Check (ping) failed for target "2620:0:ccd::2" on interface wan6 (6in4-wan6)
Mon Dec 24 19:29:44 2018 user.info mwan3track[5646]: Check (ping) failed for target "2620:0:ccc::2" on interface wan6 (6in4-wan6)
```

It turns out that the -I argument for a IPv6-only link with a link-local and a public routable address will not work:

```
~# ping -I 6in4-wan6 2001:4860:4860::8888
PING 2001:4860:4860::8888 (2001:4860:4860::8888): 56 data bytes
^C
--- 2001:4860:4860::8888 ping statistics ---
5 packets transmitted, 0 packets received, 100% packet loss
```

but if I specify the public IPv6 address:

```
~# ping -I 2001:<redacted> 2001:4860:4860::8888
PING 2001:4860:4860::8888 (2001:4860:4860::8888) from 2001:470:18:983::2: 56 data bytes
64 bytes from 2001:4860:4860::8888: seq=0 ttl=61 time=193.777 ms
64 bytes from 2001:4860:4860::8888: seq=1 ttl=61 time=193.686 ms
64 bytes from 2001:4860:4860::8888: seq=2 ttl=61 time=193.700 ms
64 bytes from 2001:4860:4860::8888: seq=3 ttl=61 time=201.931 ms
^C
--- 2001:4860:4860::8888 ping statistics ---
4 packets transmitted, 4 packets received, 0% packet loss
round-trip min/avg/max = 193.686/195.773/201.931 ms
```

I'm running OpenWrt 18.06.1 with `mwan3 - 2.6.18-1`. If this deserves to be in a separate ticket, I'd be happy to open one.

Thanks!

feckert, 2019-05-25:
No further suggestions discussions for this issue

closed 2019-05-25 by feckert