openwrt/packages #5769: libuhttpd-2.0.2: failed to build with x86_64 target

KazukiShimada, 2018-03-16:
Maintainer: @zhaojh329
Environment: x86_64, OpenWrt version 5862f01ef71e7ec75394e877e74e781b7d08a4d3

Description:
libuhttpd-2.0.2 failed to build with x86_64 target again.
x86 is OK this time.
```
make[6]: Entering directory '/home/zukky/openwrt/openwrt/build_dir/target-x86_64_musl/libuhttpd-nossl/libuhttpd-2.0.2'
make[6]: Leaving directory '/home/zukky/openwrt/openwrt/build_dir/target-x86_64_musl/libuhttpd-nossl/libuhttpd-2.0.2'
make[6]: Entering directory '/home/zukky/openwrt/openwrt/build_dir/target-x86_64_musl/libuhttpd-nossl/libuhttpd-2.0.2'
[  8%] Building C object src/CMakeFiles/uhttpd.dir/lua_template.c.o
/home/zukky/openwrt/openwrt/build_dir/target-x86_64_musl/libuhttpd-nossl/libuhttpd-2.0.2/src/lua_template.c: In function 'uh_template':
/home/zukky/openwrt/openwrt/build_dir/target-x86_64_musl/libuhttpd-nossl/libuhttpd-2.0.2/src/lua_template.c:599:24: error: cast from pointer to integer of different size [-Werror=pointer-to-int-cast]
     lua_pushinteger(L, (lua_Integer)(lua_Integer *)cl);
                        ^
cc1: all warnings being treated as errors
src/CMakeFiles/uhttpd.dir/build.make:206: recipe for target 'src/CMakeFiles/uhttpd.dir/lua_template.c.o' failed
make[6]: *** [src/CMakeFiles/uhttpd.dir/lua_template.c.o] Error 1
make[6]: Leaving directory '/home/zukky/openwrt/openwrt/build_dir/target-x86_64_musl/libuhttpd-nossl/libuhttpd-2.0.2'
CMakeFiles/Makefile2:85: recipe for target 'src/CMakeFiles/uhttpd.dir/all' failed
make[5]: *** [src/CMakeFiles/uhttpd.dir/all] Error 2
make[5]: Leaving directory '/home/zukky/openwrt/openwrt/build_dir/target-x86_64_musl/libuhttpd-nossl/libuhttpd-2.0.2'
Makefile:129: recipe for target 'all' failed
make[4]: *** [all] Error 2
make[4]: Leaving directory '/home/zukky/openwrt/openwrt/build_dir/target-x86_64_musl/libuhttpd-nossl/libuhttpd-2.0.2'
Makefile:79: recipe for target '/home/zukky/openwrt/openwrt/build_dir/target-x86_64_musl/libuhttpd-nossl/libuhttpd-2.0.2/.built' failed
make[3]: *** [/home/zukky/openwrt/openwrt/build_dir/target-x86_64_musl/libuhttpd-nossl/libuhttpd-2.0.2/.built] Error 2
make[3]: Leaving directory '/home/zukky/openwrt/openwrt/feeds/packages/libs/libuhttpd'
package/Makefile:107: recipe for target 'package/feeds/packages/libuhttpd/compile' failed
make[2]: *** [package/feeds/packages/libuhttpd/compile] Error 2
make[2]: Leaving directory '/home/zukky/openwrt/openwrt'
package/Makefile:103: recipe for target '/home/zukky/openwrt/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile' failed
make[1]: *** [/home/zukky/openwrt/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/zukky/openwrt/openwrt'
/home/zukky/openwrt/openwrt/include/toplevel.mk:216: ターゲット 'world' のレシピで失敗しました
make: *** [world] エラー 2
```

zhaojh329, 2018-03-16:
I found that Lua does not actually support 64 bit integers, and I have to change the way I do it.

zhaojh329, 2018-03-16:
I had test it.
The size of LUA_INTEGER is 4 in openwrt-x64, so this is a bug of OpenWRT.
See https://github.com/openwrt/openwrt/blob/master/package/utils/lua/patches/010-lua-5.1.3-lnum-full-260308.patch

jow-, 2018-03-16:
Yes, OpenWrt uses 32bit integers for Lua. I don't see why this is a bug.

zhaojh329, 2018-03-16:
But some body will compile for x64.

jow-, 2018-03-16:
You problem is that you store pointers as Lua numbers, this is asking for trouble. Use proper userdata or light userdata to store native C pointers in the Lua VM.

zhaojh329, 2018-03-16:
In this case, why do you use LNUM_INT32 and LNUM_INT64.

jow-, 2018-03-16:
Because we do not want to maintain different settings on different platforms.

zhaojh329, 2018-03-18:
Fixed. #5783

KazukiShimada, 2018-03-19:
I update to latest and build successfully.
Thanks to all.

closed 2018-03-19 by KazukiShimada