openwrt/packages #5771: OpenConnect: Can't Access the Internet After Assigning IP

tierra-bronza, 2018-03-16:
I setup Openwrt as client. The client was assigned IP. I also checked logs and I saw this:
`daemon.info openconnect[30178]: Established DTLS connection (using GnuTLS). Ciphersuite (DTLS1.2)-(RSA)-(AES-128-GCM).  `
  But I couldn't connect to the Internet. I failed to ping any public IP address like 8.8.8.8 but the remote VPN LAN segment. Then I added a static route:
`Interface: openconnection 
Target: 8.8.8.8/24 
Gateway: 192.168.1.1 (which is my VPN gateway)`
  After doing that I tried to trace 8.8.8.8 and the traffic did pass through VPN, however, everything got worse when I turned the target into 0.0.0.0/0. I couldn't even ping 192.168.1.1. That was weird.
  Here's the configuration:
`config interface 'vpn'
	option proto 'openconnect'
	option port '23333'
	option username 'xx'
	option password 'xx'
	option auto '0'
	option serverhash 'xxxxxxx'
	option interface 'WAN'
	option defaultroute '0'
	option server 'xxx.xxx'
	option authgroup 'DEFAULT'`
  I guess there are no mistakes with the configuration above as well as the server for it just worked perfectly on both my PC and Android clients. My firmware is CC 15.05 and all packages I used are up to date, and I don't think that the issue has anything to do with routers because someone else who have connected to other servers via his Openwrt sucessfully experienced the same issue when trying to connect to my VPN server. By the way, it seemed OpenConnect-GUI for Windows too had a problem similar to this one. What I used are AnyConnect clients, though the two protocols are compatible.

nmav, 2018-05-12:
You seem to have a setup which all internet traffic goes into vpn server but the VPN server isn't fully functional (e.g., doesn't provide DNS).

nmav, 2019-01-10:
Please re-open and add further information if this issue still holds.

closed 2019-01-10 by nmav