openwrt/packages #5778: transmission: rpc throws error 421

poranje, 2018-03-16:
Maintainer: @neheb
Environment: tplink_c2600, OpenWrt r6417-b7b14fd (SNAPSHOT), transmission 2.93-6

Description:
When using transmission-web interface on http://<host>:9091/transmission/web/ the following error message is shown:
```
Connection Failed
Could not connect to the server. You may need to reload the page to reconnect.
```
A click on the "Details" button gives:
```
421: Misdirected RequestTransmission received your request, but the hostname was unrecognized.
To fix this, choose one of the following options:
Enable password authentication, then any hostname is allowed.
Add the hostname you want to use to the whitelist in settings.
If you're editing settings.json, see the 'rpc-host-whitelist' and 'rpc-host-whitelist-enabled' entries.
This requirement has been added to help prevent DNS Rebinding attacks.
```
The log (Message level = DEBUG) does not show any info on the error event.

The setting of 'rpc-host-whitelist' that is mentioned in the error messag is not handled by the init script (although 'rpc-host-whitelist-enabled' is). Adding handling of 'rpc_host_whitelist' to the init script does helps. Both 'rpc-whitelist' and 'rpc-host-whitelist' seem to be needed (I do not know what the differences in the semantics of those two options are).

neheb, 2018-03-16:
This is an error with the RPC whitelist. By default, it's set to 192.168.1.*, meaning the IP address of your router is different. I think I will remove this default as iptables is more than competent to block bad traffic.

poranje, 2018-03-16:
It seems @neheb misread the issue.
As the rpc_whitelist option was explcitly set to a specific value (10.*.*.*) no default (192.168.1.*) is involved in the issue; nor did the json config file show that default. Only after adding handling of the 'rpc_host_whitelist' option to the init script the error vanishes.

neheb, 2018-03-16:
Oh I see. The only thing I can think of is you did an upgrade install which did not add rpc_host_whitelist_enabled to the config. I disabled the whole thing by default since it's quite useless for OpenWrt (dnsmasq deals with DNS Rebinding).

Check to make sure rpc_host_whitelist_enabled is set to 0 in /etc/config/transmission

Now that I think about it, I should probably just disable it in the init script and remove configuration.

poranje, 2018-03-16:
When removal of handling of the rpc-host-whitelist and rpc-host-whitelist-enabled options by the init script would imply that users of the RPC API cannot be whitelisted anymore and need to be authenticated by name/password, then that would be a loss of functionality for some use cases.

Regards,
Paul

poranje, 2018-04-05:
Missing handling in the init script of rpc_host_whitelist is the cause of the error. Please add it.

```--- a/etc/init.d/transmission	2018-04-05 15:40:39.786134022 +0200
+++ b/etc/init.d/transmission	2018-04-05 15:35:05.000000000 +0200
@@ -99,7 +99,7 @@
 		append_params_quotes "$cfg" \
 			blocklist_url bind_address_ipv4 bind_address_ipv6 download_dir incomplete_dir \
 			peer_congestion_algorithm peer_socket_tos rpc_bind_address rpc_password rpc_url \
-			rpc_username rpc_whitelist script_torrent_done_filename watch_dir
+			rpc_username rpc_whitelist rpc_host_whitelist script_torrent_done_filename watch_dir
 
 		echo "\"invalid-key\": false" >> $config_file
 		echo "}" >> $config_file
```

neheb, 2019-03-30:
Added. closing.

closed 2019-03-30 by neheb