openwrt/packages #5798: mwan3: randomly blocks outgoing traffic on reboot (messing with IPv6 NDP)

luizluca, 2018-03-22:
Maintainer: @feckert , @kubco2
Environment: LEDE Reboot 17.01.4 r3560-79f57e422d x86

I'm using IPv6 only wan interfaces. All settings are static. Currently, I have only one wan configured.
In order to allow mwan3 to start, I added a stub IPv4 address to them (172.16.255.2/30).
Tracker is using only IPv6 addresses. I have a single rule that uses main routing table for everything (for now):

```
config rule 'ipv6'
        option sticky '0'
        option use_policy 'default'
        option src_ip '::/0'
        option dest_ip '::/0'
        option proto 'all'
```

About 50% of time, when I reboot, ping did not work when using wan interfaces. I cannot even ping directly connected IPv6 hosts, like my default gateway.

I compared "ip -6 rule" and "ip6tables -t mangle -L -n -v" and both are identical both time (when it works and when it does not). If I simply bring mwan3 interface down and up, it works again.

I did step-by-step what mwan3 would do to turn it off. Ping starts to work after:

```
ip6tables -t mangle -D OUTPUT -j mwan3_hook 
```

So, if I run:

```
ip6tables -t mangle -D OUTPUT -j mwan3_hook
ping6 -c1 2001:4860:4860::8888
ip6tables -t mangle -A OUTPUT -j mwan3_hook
```

My guess is that mwan3 is messing with ICMPv6 stuff like Neighbor Discovery. This is what my router knows about its default gateway:

```
# ip -6 nei show 2001:1291:1002:110::a
2001:1291:1002:110::a dev eth2  FAILED
```

And after the ip6tables/ping/ip6tables trick:

```
# ip -6 nei show 2001:1291:1002:110::a
2001:1291:1002:110::a dev eth2 lladdr 38:4f:49:23:18:46 router REACHABLE
```

So, instead of ip6tables/ping/ip6tables trick, I can also simply do:

```
ip -6 nei replace 2001:1291:1002:110::a dev eth2 lladdr 38:4f:49:23:18:46
```

The problem lies with mwan3 messing with NDP for wan interfaces (configured with static addresses?). If we skip mwan3 for ICMPv6 Neighbor Solicitation (135):

```
ip6tables --table mangle -I OUTPUT 1 -p ipv6-icmp -m icmp6 --icmpv6-type 135 -m limit --limit 1000/sec -j ACCEPT
```

It simply works. Maybe it should not mess with those IPv6 stuff. OpenWrt firewall, by default, allow a bunch of IPv6 stuff that might also be interesting to skip mwan3.

Now, back to the observed behavior, when the router tried to detect its default router before mwan3 was up, NDP worked and everything was fine. If the router did not send a packet before mwan3, it could not find its own default gateway. The rest is consequence. For those interfaces that uses DHCPv6, the discovery happens very soon and it might not manifest the problem.

I workaround my problem adding the cited rule to firewall.user. However, I guess mwan3 might deserve a fix.
Or did I get or do something wrong from the start?

feckert, 2018-09-22:
I think this was solved with commit https://github.com/openwrt/packages/commit/b6249f1781efc4fefbdf87b661d53c0923ec7438#diff-34590d11545c84b961d1992ce3aefdeb

feckert, 2018-11-21:
@hnyman No reaction from issue creator and I think this is solved. So please close this issue :-)

closed 2018-11-21 by hnyman

luizluca, 2018-11-25:
Yes, solved now with the 18.06 backport. Thanks @feckert