openwrt/packages #5809: transmission-daemon: init script assumes that user = group

luizluca, 2018-03-24:
Maintainer: @neheb
Environment: OpenWrt SNAPSHOT r6514-ed369e0 ar71xx/generic
Description:

transmission init script assumes that user and group will be always the same. It must be a group option in uci transmission conf, which can use user conf as default value.

neheb, 2018-03-24:
Is this not true in your case? Stock configuration on most desktop OSes is "transmission:transmission". It's "transmission:root" currently. I don't believe this line has any effect: https://github.com/openwrt/packages/blob/master/net/transmission/files/transmission.init#L114

luizluca, 2018-03-25:
When you allow the user to set which user to run transmission in UCI configuration, you cannot assume that. The user can create a new user for transmission or even user nobody, which group is nogroup.

If user must be transmission, the option should be removed from configuration.

neheb, 2018-04-06:
This needs LuCI support ideally.

The initial reason why I changed it to user:user was because it was being created as user:root, which is bogus. Not sure if it has security implications.

In any case, I'll add UCI support.

neheb, 2020-01-12:
I added this in UCI. Closing.

closed 2020-01-12 by neheb