openwrt/packages #5811: erlang: upgrade to latest release

leoliu, 2018-03-24:
Erlang 17.5 is seriously out of date and may contain unfixed security issues. The latest stable release is `20.3.2` from https://github.com/erlang/otp/releases. Note `20.3` contains two severe bugs that `20.3.2` fixes.

diizzyy, 2018-04-16:
@leoliu 
Can you make PR for it?

leoliu, 2018-04-18:
@diizzyy Does this look alright? https://github.com/leoliu/packages/commits/master

diizzyy, 2018-04-18:
I've made a few comments, as it doesn't even compile or start to compile (patches fail).

In general I think you'll find these helpful if you want to bump erlang.

https://git.alpinelinux.org/cgit/aports/tree/community/erlang?h=master
https://git.archlinux.org/svntogit/community.git/tree/trunk?h=packages/erlang-nox
https://git.busybox.net/buildroot/tree/package/erlang

leoliu, 2018-04-18:
Thanks for your comments. I hope I have addressed (some of) them. Is there a simple way to test if my changes are buildable? Thanks.

diizzyy, 2018-04-18:
```make menuconfig``` --> select all packages related to erlang in lang
Save and exit
```make -j1 package/erlang/compile V=s```

leoliu, 2018-04-19:
Looks like setting a build environment may take a while. I wonder if someone could take over https://github.com/leoliu/packages/commit/a1ba0f2b22098f37c40835c9b2a302a54ddb1424?

diizzyy, 2018-04-19:
It'll take a few hours tops depending on your computer for the first time as you need to compile a toolchain. Next time you run it'll only compile the packages which is much faster.
You can speed it up by using a higher job number however it'll also make output "garbled" as you usually will have several processes running in parallel making it hard to read logs in case of failure.

leoliu, 2018-08-16:
Excellent. 19.x is receiving updates which means more secure. Unfortunately it might sort of end-of-life soon. I think there needs a way to keep the package reasonably up-to-date. Sorry I didn't contribute anything significant.

closed 2018-08-17 by dibdot