openwrt/packages #5835: zabbix-extra-wifi: lua error in discovery item

adriankirchner, 2018-03-27:
Maintainer: @champtar 
Environment: armv7l, Cisco Meraki MR33, OpenWrt SNAPSHOT r6168+17-4979a72e37
Description:
After installing `zabbix-extra-wifi` via `opkg` and restarting `zabbix-agentd`, the key `wifi.ifdiscovery` for wifi discovery is not working due to a lua error:

```bash
$ zabbix_get -s my-mr33.local -p 10062 -k 'wifi.ifdiscovery'
lua: (command line):1: bad argument #1 to 'pairs' (table expected, got nil)
stack traceback:
	[C]: in function 'pairs'
	(command line):1: in main chunk
	[C]: ?
```

list of installed packages:

```bash
$ opkg list-installed
ath10k-firmware-qca4019 - 2018-02-09-8f4bafdd-1
ath10k-firmware-qca9887 - 2018-02-09-8f4bafdd-1
base-files - 184-r6168+17-4979a72e37
busybox - 1.27.2-3
dnsmasq - 2.79-2
dropbear - 2017.75-5
firewall - 2018-03-20-5cdf15ee-2
fstools - 2018-02-11-3d239815-1
fwtool - 1
hostapd-common - 2017-08-24-c2d4f2eb-6
ip6tables - 1.6.2-1
ipq-wifi-meraki_mr33 - 1
iptables - 1.6.2-1
iw - 4.14-1
iwinfo - 2018-02-15-223e09bf-1
jshn - 2018-03-21-eebe3fcd-1
jsonfilter - 2018-02-04-c7e938d6-1
kernel - 4.14.18-1-5ef1d5c2a42f41e69a71621d8a054ed4
kmod-ath - 4.14.18+2017-11-01-4
kmod-ath10k - 4.14.18+2017-11-01-4
kmod-cfg80211 - 4.14.18+2017-11-01-4
kmod-gpio-button-hotplug - 4.14.18-2
kmod-ip6tables - 4.14.18-1
kmod-ipt-conntrack - 4.14.18-1
kmod-ipt-core - 4.14.18-1
kmod-ipt-nat - 4.14.18-1
kmod-leds-gpio - 4.14.18-1
kmod-lib-crc-ccitt - 4.14.18-1
kmod-mac80211 - 4.14.18+2017-11-01-4
kmod-nf-conntrack - 4.14.18-1
kmod-nf-conntrack6 - 4.14.18-1
kmod-nf-ipt - 4.14.18-1
kmod-nf-ipt6 - 4.14.18-1
kmod-nf-nat - 4.14.18-1
kmod-nf-reject - 4.14.18-1
kmod-nf-reject6 - 4.14.18-1
kmod-nls-base - 4.14.18-1
kmod-ppp - 4.14.18-1
kmod-pppoe - 4.14.18-1
kmod-pppox - 4.14.18-1
kmod-slhc - 4.14.18-1
kmod-usb-core - 4.14.18-1
kmod-usb-dwc3 - 4.14.18-1
kmod-usb-dwc3-of-simple - 4.14.18-1
kmod-usb-phy-qcom-dwc3 - 4.14.18-1
kmod-usb3 - 4.14.18-1
lede-keyring - 2017-01-20-a50b7529-1
libblobmsg-json - 2018-03-21-eebe3fcd-1
libc - 1.1.18-1
libgcc - 5.5.0-1
libip4tc - 1.6.2-1
libip6tc - 1.6.2-1
libiwinfo - 2018-02-15-223e09bf-1
libiwinfo-lua - 2018-02-15-223e09bf-1
libjson-c - 0.12.1-1
libjson-script - 2018-03-21-eebe3fcd-1
liblua - 5.1.5-1
libmbedtls - 2.7.0-1
libncurses - 6.0-1
libnl-tiny - 0.1-5
libpthread - 1.1.18-1
libubox - 2018-03-21-eebe3fcd-1
libubus - 2018-01-16-5bae22eb-1
libubus-lua - 2018-01-16-5bae22eb-1
libuci - 2018-01-01-5beb95da-1
libuci-lua - 2018-01-01-5beb95da-1
libuclient - 2017-11-02-4b87d831-1
libustream-mbedtls - 2016-07-02-ec80adaa-2
libxtables - 1.6.2-1
logd - 2018-02-14-128bc35f-1
lua - 5.1.5-1
luci - git-18.082.53982-92408aa-1
luci-app-firewall - git-18.082.53982-92408aa-1
luci-base - git-18.082.53982-92408aa-1
luci-lib-ip - git-18.082.53982-92408aa-1
luci-lib-jsonc - git-18.082.53982-92408aa-1
luci-lib-nixio - git-18.082.53982-92408aa-1
luci-mod-admin-full - git-18.082.53982-92408aa-1
luci-proto-ipv6 - git-18.082.53982-92408aa-1
luci-proto-ppp - git-18.082.53982-92408aa-1
luci-ssl - git-18.082.53982-92408aa-1
luci-theme-bootstrap - git-18.082.53982-92408aa-1
mtd - 21
nand-utils - 1.5.2-1
nano - 2.9.4-1
netifd - 2018-03-13-69aeaabc-5
odhcp6c - 2017-09-05-1f93bd4c-10
odhcpd-ipv6only - 1.4-1
opkg - 2017-12-07-3b417b9f-2
ppp - 2.4.7-12
ppp-mod-pppoe - 2.4.7-12
procd - 2018-01-23-653629f1-2
px5g-mbedtls - 4
rpcd - 2017-12-07-cfe1e75c-1
rpcd-mod-rrdns - 20170710
terminfo - 6.0-1
ubi-utils - 1.5.2-1
ubox - 2018-02-14-128bc35f-1
ubus - 2018-01-16-5bae22eb-1
ubusd - 2018-01-16-5bae22eb-1
uci - 2018-01-01-5beb95da-1
uclient-fetch - 2017-11-02-4b87d831-1
uhttpd - 2017-11-04-a235636a-1
uhttpd-mod-ubus - 2017-11-04-a235636a-1
usign - 2015-07-04-ef641914-1
wireless-regdb - 2017-10-20-4343d359
wpad-mini - 2017-08-24-c2d4f2eb-6
zabbix-agentd - 3.2.7-2
zabbix-extra-mac80211 - 3.2.7-2
zabbix-extra-network - 3.2.7-2
zabbix-extra-wifi - 3.2.7-2
```

champtar, 2018-03-27:
Hi @adriankirchner 
Have you tried to run the code manually ?
https://github.com/openwrt/packages/blob/master/admin/zabbix/files/wifi#L6

Also is zabbix running as root?

adriankirchner, 2018-03-27:
Hi @champtar,

running the code as root:

```bash
lua: (command line):1: bad argument #1 to 'type' (string expected, got nil)
stack traceback:
	[C]: in function 'type'
	(command line):1: in main chunk
	[C]: ?
```

Zabbix Agent is running as user `zabbix`.

Output of `iwinfo`:
```
wlan0     ESSID: unknown
          Access Point: 00:00:00:00:00:00
          Mode: Client  Channel: unknown (unknown)
          Tx-Power: 0 dBm  Link Quality: unknown/70
          Signal: unknown  Noise: unknown
          Bit Rate: unknown
          Encryption: unknown
          Type: nl80211  HW Mode(s): 802.11bgnac
          Hardware: 168C:0050 0000:0000 [Generic MAC80211]
          TX power offset: unknown
          Frequency offset: unknown
          Supports VAPs: yes  PHY name: phy0

wlan1     ESSID: unknown
          Access Point: 00:00:00:00:00:00
          Mode: Client  Channel: unknown (unknown)
          Tx-Power: 0 dBm  Link Quality: unknown/70
          Signal: unknown  Noise: unknown
          Bit Rate: unknown
          Encryption: unknown
          Type: nl80211  HW Mode(s): 802.11bgn
          Hardware: unknown [Generic MAC80211]
          TX power offset: unknown
          Frequency offset: unknown
          Supports VAPs: yes  PHY name: phy1

wlan2     ESSID: unknown
          Access Point: 00:00:00:00:00:00
          Mode: Client  Channel: unknown (unknown)
          Tx-Power: 0 dBm  Link Quality: unknown/70
          Signal: unknown  Noise: unknown
          Bit Rate: unknown
          Encryption: unknown
          Type: nl80211  HW Mode(s): 802.11nac
          Hardware: unknown [Generic MAC80211]
          TX power offset: unknown
          Frequency offset: unknown
          Supports VAPs: yes  PHY name: phy2
```

adriankirchner, 2018-10-20:
Since Zabbix Agent is running as user `zabbix` it need an ubus ACL (based on [this post](https://forum.openwrt.org/t/solved-zabbix-server-cant-discover-lede-interfaces/10454/12)):

File: `/usr/share/acl.d/zabbix-agent.json`
```
{
	"user": "zabbix",
	"access": {
		"network.wireless": {
			"methods": [ "status" ]
		}
	}
}
```

Reload ubus with `killall -HUP ubusd`

closed 2018-10-20 by adriankirchner

champtar, 2018-10-20:
@adriankirchner do you want to try to create a PR to add this ACL ?

adriankirchner, 2018-10-20:
Would love to but I'm not sure if I'll find time in the near future to learn stuff like building stuff locally on my openwrt device and understanding how opkg works (eg. for ubus reloading). Maybe you can get me some hints.

champtar, 2018-10-23:
fixed with https://github.com/openwrt/packages/commit/1afcc61601cb1e4c6238525f5d07c20e66e32ad9
Thanks @adriankirchner

champtar, 2018-10-23:
@adriankirchner i've also reimplemented network discovery using ubus, see https://github.com/openwrt/packages/commit/5d1aa99648bc2e22d7db7cc12926ea79e130bb04

adriankirchner, 2018-10-23:
Nice one!

troffasky, 2024-06-17:
So did this fix actually work for you @adriankirchner? I am still having this issue today on a fresh install of 23.05.

adriankirchner, 2024-06-17:
@troffasky I'm not using the devices anymore, but if I recall correctly, the fix worked flawlessly.