openwrt/packages #5840: can not block specific porn site after adding the domain into blacklist

tacuar, 2018-03-27:
for some normal websites, after adding them into blacklist, it works fine.

however i tried to add a port site 91porn.com into the blacklist, no matter how i try, it can not block it.

could someone tell what would be the correct way of using it , in order to make it work?

luizluca, 2018-03-27:
@tacuar , which package are you talking about?

tacuar, 2018-04-10:
@dibdot  sorry for the late reply .

I was talking about the package adblock. I thought the issue would be reported to adblock maintenance owner only and did not notice it is reported to the general package.

dibdot, 2018-04-10:
@tacuar can't reproduce your issue - works for me. Anyway if the problem still persist, enable debug, rerun adblock and send me the full logs (logread -e "adblock" to dev@brenken.org). Furthermore send me your blacklist, too.