openwrt/packages #5853: Problem with compile PHP 7 with ICU61.1

steelrat2000, 2018-04-01:
223/5000
Hello.
After updating the ICU package to 61_1, the PHP package is not compiled. I have such errors:
`/home/jo/lede/build_dir/target-arm_cortex-a9+vfpv3_musl_eabi/php-7.2.4/ext/intl/intl_convertcpp.h:29:25: error: 'UnicodeString' was not declared in this scope
 int intl_stringFromChar(UnicodeString &ret, char *str, size_t str_len, UErrorCode *status);
                         ^~~~~~~~~~~~~
/home/jo/lede/build_dir/target-arm_cortex-a9+vfpv3_musl_eabi/php-7.2.4/ext/intl/intl_convertcpp.h:29:25: note: suggested alternative:
In file included from /home/jo/lede/build_dir/target-arm_cortex-a9+vfpv3_musl_eabi/php-7.2.4/ext/intl/intl_convertcpp.h:26:0,
                 from /home/jo/lede/build_dir/target-arm_cortex-a9+vfpv3_musl_eabi/php-7.2.4/ext/intl/intl_convertcpp.cpp:21:
/home/jo/lede/staging_dir/target-arm_cortex-a9+vfpv3_musl_eabi/usr/include/unicode/unistr.h:286:20: note:   'icu_61::UnicodeString'
 class U_COMMON_API UnicodeString : public Replaceable
                    ^~~~~~~~~~~~~
In file included from /home/jo/lede/build_dir/target-arm_cortex-a9+vfpv3_musl_eabi/php-7.2.4/ext/intl/intl_convertcpp.cpp:21:0:
/home/jo/lede/build_dir/target-arm_cortex-a9+vfpv3_musl_eabi/php-7.2.4/ext/intl/intl_convertcpp.h:29:40: error: 'ret' was not declared in this scope
 int intl_stringFromChar(UnicodeString &ret, char *str, size_t str_len, UErrorCode *status);
                                        ^~~
/home/jo/lede/build_dir/target-arm_cortex-a9+vfpv3_musl_eabi/php-7.2.4/ext/intl/intl_convertcpp.h:29:40: note: suggested alternative: 'drem'
 int intl_stringFromChar(UnicodeString &ret, char *str, size_t str_len, UErrorCode *status);
                                        ^~~
                                        drem
/home/jo/lede/build_dir/target-arm_cortex-a9+vfpv3_musl_eabi/php-7.2.4/ext/intl/intl_convertcpp.h:29:45: error: expected primary-expression before 'char'
 int intl_stringFromChar(UnicodeString &ret, char *str, size_t str_len, UErrorCode *status);
                                             ^~~~
/home/jo/lede/build_dir/target-arm_cortex-a9+vfpv3_musl_eabi/php-7.2.4/ext/intl/intl_convertcpp.h:29:63: error: expected primary-expression before 'str_len'
 int intl_stringFromChar(UnicodeString &ret, char *str, size_t str_len, UErrorCode *status);
                                                               ^~~~~~~
/home/jo/lede/build_dir/target-arm_cortex-a9+vfpv3_musl_eabi/php-7.2.4/ext/intl/intl_convertcpp.h:29:83: error: expected primary-expression before '*' token
 int intl_stringFromChar(UnicodeString &ret, char *str, size_t str_len, UErrorCode *status);
                                                                                   ^
/home/jo/lede/build_dir/target-arm_cortex-a9+vfpv3_musl_eabi/php-7.2.4/ext/intl/intl_convertcpp.h:29:84: error: 'status' was not declared in this scope
 int intl_stringFromChar(UnicodeString &ret, char *str, size_t str_len, UErrorCode *status);
                                                                                    ^~~~~~
/home/jo/lede/build_dir/target-arm_cortex-a9+vfpv3_musl_eabi/php-7.2.4/ext/intl/intl_convertcpp.h:29:84: note: suggested alternative: 'stat64'
 int intl_stringFromChar(UnicodeString &ret, char *str, size_t str_len, UErrorCode *status);
                                                                                    ^~~~~~
                                                                                    stat64
/home/jo/lede/build_dir/target-arm_cortex-a9+vfpv3_musl_eabi/php-7.2.4/ext/intl/intl_convertcpp.h:29:90: error: expression list treated as compound expression in initializer [-fpermissive]
 int intl_stringFromChar(UnicodeString &ret, char *str, size_t str_len, UErrorCode *status);
                                                                                          ^
/home/jo/lede/build_dir/target-arm_cortex-a9+vfpv3_musl_eabi/php-7.2.4/ext/intl/intl_convertcpp.h:31:40: error: 'UnicodeString' does not name a type; did you mean 'gdImageString'?
 zend_string* intl_charFromString(const UnicodeString &from, UErrorCode *status);
                                        ^~~~~~~~~~~~~
                                        gdImageString
/home/jo/lede/build_dir/target-arm_cortex-a9+vfpv3_musl_eabi/php-7.2.4/ext/intl/intl_convertcpp.cpp:28:25: error: redefinition of 'int intl_stringFromChar'
 int intl_stringFromChar(UnicodeString &ret, char *str, size_t str_len, UErrorCode *status)
                         ^~~~~~~~~~~~~
In file included from /home/jo/lede/build_dir/target-arm_cortex-a9+vfpv3_musl_eabi/php-7.2.4/ext/intl/intl_convertcpp.cpp:21:0:
/home/jo/lede/build_dir/target-arm_cortex-a9+vfpv3_musl_eabi/php-7.2.4/ext/intl/intl_convertcpp.h:29:5: note: 'int intl_stringFromChar' previously defined here
 int intl_stringFromChar(UnicodeString &ret, char *str, size_t str_len, UErrorCode *status);
     ^~~~~~~~~~~~~~~~~~~
/home/jo/lede/build_dir/target-arm_cortex-a9+vfpv3_musl_eabi/php-7.2.4/ext/intl/intl_convertcpp.cpp:28:25: error: 'UnicodeString' was not declared in this scope
 int intl_stringFromChar(UnicodeString &ret, char *str, size_t str_len, UErrorCode *status)
                         ^~~~~~~~~~~~~
/home/jo/lede/build_dir/target-arm_cortex-a9+vfpv3_musl_eabi/php-7.2.4/ext/intl/intl_convertcpp.cpp:28:25: note: suggested alternative:
In file included from /home/jo/lede/build_dir/target-arm_cortex-a9+vfpv3_musl_eabi/php-7.2.4/ext/intl/intl_convertcpp.h:26:0,
                 from /home/jo/lede/build_dir/target-arm_cortex-a9+vfpv3_musl_eabi/php-7.2.4/ext/intl/intl_convertcpp.cpp:21:
/home/jo/lede/staging_dir/target-arm_cortex-a9+vfpv3_musl_eabi/usr/include/unicode/unistr.h:286:20: note:   'icu_61::UnicodeString'
 class U_COMMON_API UnicodeString : public Replaceable
                    ^~~~~~~~~~~~~
/home/jo/lede/build_dir/target-arm_cortex-a9+vfpv3_musl_eabi/php-7.2.4/ext/intl/intl_convertcpp.cpp:28:40: error: 'ret' was not declared in this scope
 int intl_stringFromChar(UnicodeString &ret, char *str, size_t str_len, UErrorCode *status)
                                        ^~~
/home/jo/lede/build_dir/target-arm_cortex-a9+vfpv3_musl_eabi/php-7.2.4/ext/intl/intl_convertcpp.cpp:28:40: note: suggested alternative: 'recv'
 int intl_stringFromChar(UnicodeString &ret, char *str, size_t str_len, UErrorCode *status)
                                        ^~~
                                        recv
/home/jo/lede/build_dir/target-arm_cortex-a9+vfpv3_musl_eabi/php-7.2.4/ext/intl/intl_convertcpp.cpp:28:45: error: expected primary-expression before 'char'
 int intl_stringFromChar(UnicodeString &ret, char *str, size_t str_len, UErrorCode *status)
                                             ^~~~
/home/jo/lede/build_dir/target-arm_cortex-a9+vfpv3_musl_eabi/php-7.2.4/ext/intl/intl_convertcpp.cpp:28:63: error: expected primary-expression before 'str_len'
[... 21 lines cut ...]
  int32_t capacity = from.length() * 3;
                          ^~~~~~
/home/jo/lede/build_dir/target-arm_cortex-a9+vfpv3_musl_eabi/php-7.2.4/ext/intl/intl_convertcpp.cpp:71:11: error: request for member 'isEmpty' in 'from', which is of non-class type 'const int'
  if (from.isEmpty()) {
           ^~~~~~~
/home/jo/lede/build_dir/target-arm_cortex-a9+vfpv3_musl_eabi/php-7.2.4/ext/intl/intl_convertcpp.cpp:77:31: error: request for member 'getBuffer' in 'from', which is of non-class type 'const int'
  const UChar *utf16buf = from.getBuffer();
                               ^~~~~~~~~
/home/jo/lede/build_dir/target-arm_cortex-a9+vfpv3_musl_eabi/php-7.2.4/ext/intl/intl_convertcpp.cpp:79:76: error: request for member 'length' in 'from', which is of non-class type 'const int'
  u_strToUTF8WithSub(ZSTR_VAL(u8res), capacity, &actual_len, utf16buf, from.length(),
                                                                            ^~~~~~
Makefile:962: recipe for target 'ext/intl/intl_convertcpp.lo' failed
make[3]: *** [ext/intl/intl_convertcpp.lo] Error 1
make[3]: Leaving directory '/home/jo/lede/build_dir/target-arm_cortex-a9+vfpv3_musl_eabi/php-7.2.4'
Makefile:595: recipe for target '/home/jo/lede/build_dir/target-arm_cortex-a9+vfpv3_musl_eabi/php-7.2.4/.built' failed
make[2]: *** [/home/jo/lede/build_dir/target-arm_cortex-a9+vfpv3_musl_eabi/php-7.2.4/.built] Error 2
make[2]: Leaving directory '/home/jo/lede/feeds/packages/lang/php7'
package/Makefile:107: recipe for target 'package/feeds/packages/php7/compile' failed
make[1]: *** [package/feeds/packages/php7/compile] Error 2
make[1]: Leaving directory '/home/jo/lede'
/home/jo/lede/include/toplevel.mk:216: recipe for target 'package/php7/compile' failed
make: *** [package/php7/compile] Error 2`

I found 'makefile' on the ICU 59_1 disk and compiles everything fine.
Can you adapt ICU 61 or PHP to make everything work?

nxhack, 2018-04-02:
https://bugs.archlinux.org/task/58061
http://source.icu-project.org/repos/icu/trunk/icu4c/readme.html#RecBuild

steelrat2000, 2018-04-02:
Thanks for the help.
Created a fix and now compiles everything on ICU61.

mhei, 2018-04-02:
I've added two upstream patches and at least it compiles now. But due to Easter holiday travelling, I had/have little time and could only compile-test so far. I will see to verify deeper next evenings... or maybe you could check, whether it works for you.

steelrat2000, 2018-04-02:
Z Twoimi poprawkami PHP7 kompiluje się również.

Gdy kompilowałem PHP zauważyłem jeszcze takie ostrzeżenie 
`configure: WARNING: ========================================================
configure: WARNING: Use of bundled libzip is deprecated and will be removed.
configure: WARNING: Some features such as encryption and bzip2 are not available.
configure: WARNING: Use system library and --with-libzip is recommended.
configure: WARNING: ========================================================`

O co im chodzi?

mhei, 2018-04-03:
This is an unrelated hint, that we need to factor out libzip someday.

I run-time tested intl extension a little bit without recognizable problems, so I'm closing this issue.

closed 2018-04-03 by mhei