openwrt/packages #5868: installing asterisk13: Error relocating /usr/sbin/asterisk: fopencookie: symbol not found

paolomeraviglia, 2018-04-04:
Please make sure that the issue subject starts with `asterisk13: `

Maintainer: @\samm-git 
Environment: arm_cortex-a9, R8000, 17.01.4)

Description:

```
I did

opkg update
opkg install asterisk13

after that install finish with error

Error relocating /usr/sbin/asterisk: fopencookie: symbol not found

```

gemini666, 2018-04-12:
Hi. I have the same problem. No solution yet.

jslachta, 2018-04-14:
Hi,

I replicated this issue in telephony repository, thanks for letting me know. I am investigating what has happened.

https://github.com/openwrt/telephony/issues/319

Jiri

jslachta, 2018-04-14:
I am closing this issue here. Please follow up the issue in telephony repository. Thank you!

closed 2018-04-14 by jslachta

jslachta, 2018-04-14:
Hi @gemini666 and @paolomeraviglia,

you seem to have used another repository than you should use. In december 2017 fopencookie() was [implemeted in musl](https://git.musl-libc.org/cgit/musl/commit/src/stdio/fopencookie.c?id=061843340fbf2493bb615e20e66f60c5d1ef0455). This applies to current snapshots, not for the 17.* releases.

Please check your opkg.conf - the telephony feed should point on:

https://downloads.lede-project.org/releases/packages-17.01/arm_cortex-a9/telephony/

or

https://downloads.openwrt.org/releases/packages-17.01/arm_cortex-a9/telephony/

gemini666, 2018-04-14:
Hi.
I have mips_24 architecture.