openwrt/packages #5871: libmbedtls - 2.7.2-1: Error loading shared library libmbedcrypto.so.1

gerardr1, 2018-04-04:
Maintainer: @\<hauke> 
Environment: (Linux router 4.4.92 #0 Tue Oct 17 14:59:45 2017 mips GNU/Linux)

Description:
After updating the libmbedtls package this morning (libmbedtls - 2.7.2-1) the log contains this error and uhttpd does not start:

Wed Apr  4 08:54:01 2018 daemon.err uhttpd[1481]: Failed to load ustream-ssl library: Error loading shared library libmbedcrypto.so.1: No such file or directory (needed by /lib/libustream-ssl.so)

A workaround was effected by logging in via SSH and creating the symlink:

lrwxrwxrwx    1 root     root            22 Apr  4 09:26 libmbedcrypto.so.1 -> libmbedcrypto.so.2.7.2

and then starting uhttpd. Maybe it's really a libustream-ssl issue, but it occurred after the libmbedtls update.

p-wassi, 2018-04-07:
@hauke 

In mbedtls 2.7.1 (which was never built in OpenWrt), the library/Makefile was changed
which renamed libmbedcrypto.so.1 to libmbedcrypto.so.2 [1]

[1]:

diizzyy, 2018-07-20:
@gerardr1 @p-wassi @oilervoss 
I guess we can close this now?

gerardr1, 2018-07-20:
Oh, yes, easily.  Sorry,
      if I missed some procedure.
      
      Thanks ... Gerard Robinson
      
    
    On 07/20/18 15:00, Daniel Engberg
      wrote:
    
    
      
      @gerardr1
        @p-wassi
        @oilervoss
        I guess we can close this now?
      —
      [
{
"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/openwrt/packages/issues/5871#issuecomment-406697697",
"url": "https://github.com/openwrt/packages/issues/5871#issuecomment-406697697",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
},
{
"@type": "MessageCard",
"@context": "http://schema.org/extensions",
"hideOriginalBody": "false",
"originator": "AF6C5A86-E920-430C-9C59-A73278B5EFEB",
"title": "Re: [openwrt/packages] libmbedtls - 2.7.2-1: Error loading shared library libmbedcrypto.so.1 (#5871)",
"sections": [
{
"text": "",
"activityTitle": "**Daniel Engberg**",
"activityImage": "https://assets-cdn.github.com/images/email/message_cards/avatar.png",
"activitySubtitle": "@diizzyy",
"facts": [

]
}
],
{
"name": "Add a comment",
"@type": "ActionCard",
"inputs": [
{
"isMultiLine": true,
"@type": "TextInput",
"id": "IssueComment",
"isRequired": false
}
],
"actions": [
{
"name": "Comment",
"@type": "HttpPOST",
"target": "https://api.github.com",
"body": "{\n\"commandName\": \"IssueComment\",\n\"repositoryFullName\": \"openwrt/packages\",\n\"issueId\": 5871,\n\"IssueComment\": \"{{IssueComment.value}}\"\n}"
}
]
},
{
"name": "Close issue",
"@type": "HttpPOST",
"target": "https://api.github.com",
"body": "{\n\"commandName\": \"IssueClose\",\n\"repositoryFullName\": \"openwrt/packages\",\n\"issueId\": 5871\n}"
},
{
"targets": [
{
"os": "default",
"uri": "https://github.com/openwrt/packages/issues/5871#issuecomment-406697697"
}
],
"@type": "OpenUri",
"name": "View on GitHub"
},
{
"name": "Unsubscribe",
"@type": "HttpPOST",
"target": "https://api.github.com",
"body": "{\n\"commandName\": \"MuteNotification\",\n\"threadId\": 320566340\n}"
}
],
"themeColor": "26292E"
}
]
    
    
When I find myself in times of trouble,|All unattributed content herein:
Then I'll know just where I'll be,     |Copyright(C)2018 Gerard Robinson
Swearing at the console, let me be ... |All Rights Reserved; gerardr@...

closed 2018-08-05 by hnyman