openwrt/packages #5906: mwan3 breaks odhcpd's ipv6 ra service

wackejohn, 2018-04-12:
Maintainer: @feckert
Environment:
OpenWrt SNAPSHOT r6666-f6e6eadc99 / LuCI Master (git-18.101.21196-d488681)
Running on turris omnia with two pppoe wan interface,four tinc vpn connection(two for ipv4,two for ipv6).

Description:
I've configured the two ipv6 interface using tinc,and the ipv6 connection is working properly in router itself.But the odhcpd's ipv6 ra service only works when doing mwan3 restart or mwan3 stop,this caused other devices won't get the default route of ipv6.

The odhcpd ra log:
```
Thu Apr 12 19:53:17 2018 daemon.info odhcpd[5691]: Using a RA lifetime of 1800 seconds on br-lan
```

My PC connected to the router,the route only works 1800s,when doing a mwan3 restart.
```
HOME-Server:~ # ip -6 route
2001:470:*:2::/64 dev eth1  proto kernel  metric 256  pref medium
2001:470:*:2::/64 dev eth1  proto kernel  metric 256  pref medium
fd5d:189b:b5f9::/64 dev eth1  proto kernel  metric 256  pref medium
fdc4:31b5:d5a5::/64 dev eth1  proto kernel  metric 256  pref medium
fe80::/64 dev eth1  proto kernel  metric 256  pref medium
default via fe80::290:bff:fe1d:3844 dev eth1  proto ra  metric 1024  expires 24sec hoplimit 64 pref medium
```
My network config:
```
root@HOME-Router:~# cat /etc/config/network 

config interface 'loopback'
        option ifname 'lo'
        option proto 'static'
        option ipaddr '127.0.0.1'
        option netmask '255.0.0.0'

config globals 'globals'
        option ula_prefix 'fd5d:189b:b5f9::/64'

config interface 'lan'
        option type 'bridge'
        option proto 'static'
        option ipaddr '192.168.1.1'
        option netmask '255.255.255.0'
        option macaddr '00:90:0B:1D:38:44'
        option stp '1'
        option _orig_ifname 'eth0.1 wlan0 wlan1'
        option _orig_bridge 'true'
        option ifname 'lan0 lan1 lan2 lan3'
        option ip6assign '64'

config interface 'wan1'
        option proto 'pppoe'
        option username ''
        option password ''
        option peerdns '0'
        option dns '114.114.114.114 114.114.115.115'
        option metric '10'
        option ifname 'eth2'
        option force_link '1'
        option ipv6 '1'

config interface 'wan2'
        option proto 'pppoe'
        option peerdns '0'
        option dns '114.114.114.114 114.114.115.115'
        option metric '20'
        option username ''
        option password ''
        option ipv6 'auto'
        option force_link '1'
        option _orig_ifname 'eth1.2'
        option _orig_bridge 'false'
[... 355 lines cut ...]
        option sticky '0'
        option dest_ip '172.247.39.94'
        option use_policy 'wan1_wan2'

config rule 'WEB'
        option dest_port '22,25,80,110,143,443,465,587,993,995,8000'
        option family 'ipv4'
        option proto 'tcp'
        option sticky '0'
        option use_policy 'wan1_wan2'

config rule 'loadbalance'
        option proto 'all'
        option family 'ipv4'
        option equalize '1'
        option sticky '0'
        option use_policy 'loadbalanced'

config rule 'ipv6'
        option proto 'all'
        option family 'ipv6'
        option equalize '1'
        option sticky '0'
        option use_policy 'wanc_wand'
```

luizluca, 2018-04-12:
It is probably related to issue #5798. It is probably an outgoing packet from router.

If you can sniff which packet is in trouble, you could exempt them from mwan3 with something like I did for NDP:

```
ip6tables --table mangle -I OUTPUT 1 -p ipv6-icmp -m icmp6 --icmpv6-type 135 -j ACCEPT
```
You can also try to detect it simply adding a log rule in the same position of the previous rule. It will catch all packets that originate from the router itself, which is not too much traffic.

wackejohn, 2018-04-13:
Thank you for reply,I'm not good at iptables,I don't know which packet type for ipv6 ra service...But I'll try to search for it.Thanks.

feckert, 2018-04-13:
@luizluca should we add this to mwan3 `mwan3_hook `table?

> ip6tables --table mangle -I OUTPUT 1 -p ipv6-icmp -m icmp6 --icmpv6-type 135 -j ACCEPT

wackejohn, 2018-04-13:
@feckert Found some info form:
```
https://www.cisco.com/c/en/us/td/docs/ios-xml/ios/ipv6/configuration/15-2mt/ip6-15-2mt-book/ip6-neighb-disc.html
```
And I did the below command:
```
ip6tables --table mangle -I OUTPUT 1 -p ipv6-icmp -m icmp6 --icmpv6-type 134 -j ACCEPT
ip6tables --table mangle -I OUTPUT 1 -p ipv6-icmp -m icmp6 --icmpv6-type 135 -j ACCEPT
ip6tables --table mangle -I OUTPUT 1 -p ipv6-icmp -m icmp6 --icmpv6-type 137 -j ACCEPT
```
Seems that the PC got the ra message sussessfully from the router.And will do a further test to see if it working properly.Thanks.

luizluca, 2018-04-13:
> @luizluca should we add this to mwan3 mwan3_hooktable?
> 
> ip6tables --table mangle -I OUTPUT 1 -p ipv6-icmp -m icmp6 --icmpv6-type 135 -j ACCEPT

Yes, at least for my case. But we might also need to add more stuff there for other cases.

luizluca, 2018-04-13:
> And I did the command below:
> 
> ip6tables --table mangle -I OUTPUT 1 -p ipv6-icmp -m icmp6 --icmpv6-type 134 -j ACCEPT
> ip6tables --table mangle -I OUTPUT 1 -p ipv6-icmp -m icmp6 --icmpv6-type 135 -j ACCEPT
> ip6tables --table mangle -I OUTPUT 1 -p ipv6-icmp -m icmp6 --icmpv6-type 137 -j ACCEPT

@wackejohn , did you try with only ICMP type 135?

wackejohn, 2018-04-14:
@luizluca ,actually I've tried only ICMP type 135 at first,but nothing happend...

wackejohn, 2018-04-15:
@feckert 
I've found another problem when using two tinc ipv6 connections(working propery when there was only one ipv6 interface).
The mwan3 would sent the ipv6 package to wrong interface then made the ipv6 connection failed.
I have two tinc ipv6 connection(connected to two vps and the vps connected to a 6in4 tunnel each):

```
2001:470:f442::/48
2001:470:f2d2::/48
```
 Ping failed from my windows PC:
```
C:\Users\Wacke>ping 2607:f8b0:4007:803::200e -t

正在 Ping 2607:f8b0:4007:803::200e 具有 32 字节的数据:
请求超时。
请求超时。
请求超时。
请求超时。
请求超时。
请求超时。
请求超时。

2607:f8b0:4007:803::200e 的 Ping 统计信息:
    数据包: 已发送 = 7，已接收 = 0，丢失 = 7 (100% 丢失)，
Control-C
```
And the tinc host vps tcpdump:
```
[root@50KVM-2017114448 vpn6]# tcpdump -i vpn6 -vv host 2607:f8b0:4007:803::200e
tcpdump: listening on vpn6, link-type RAW (Raw IP), capture size 262144 bytes
10:44:15.029520 IP6 (hlim 127, next-header ICMPv6 (58) payload length: 40) 2001:470:f2d2:3:4827:ea57:46f4:8dff > lax17s02-in-x0e.1e100.net: [icmp6 sum ok] ICMP6, echo request, seq 1107
10:44:19.548711 IP6 (hlim 127, next-header ICMPv6 (58) payload length: 40) 2001:470:f2d2:3:4827:ea57:46f4:8dff > lax17s02-in-x0e.1e100.net: [icmp6 sum ok] ICMP6, echo request, seq 1108
10:44:24.568162 IP6 (hlim 127, next-header ICMPv6 (58) payload length: 40) 2001:470:f2d2:3:4827:ea57:46f4:8dff > lax17s02-in-x0e.1e100.net: [icmp6 sum ok] ICMP6, echo request, seq 1109
10:44:29.560691 IP6 (hlim 127, next-header ICMPv6 (58) payload length: 40) 2001:470:f2d2:3:4827:ea57:46f4:8dff > lax17s02-in-x0e.1e100.net: [icmp6 sum ok] ICMP6, echo request, seq 1110
10:44:34.564403 IP6 (hlim 127, next-header ICMPv6 (58) payload length: 40) 2001:470:f2d2:3:4827:ea57:46f4:8dff > lax17s02-in-x0e.1e100.net: [icmp6 sum ok] ICMP6, echo request, seq 1111
10:44:39.567532 IP6 (hlim 127, next-header ICMPv6 (58) payload length: 40) 2001:470:f2d2:3:4827:ea57:46f4:8dff > lax17s02-in-x0e.1e100.net: [icmp6 sum ok] ICMP6, echo request, seq 1112
10:44:44.571740 IP6 (hlim 127, next-header ICMPv6 (58) payload length: 40) 2001:470:f2d2:3:4827:ea57:46f4:8dff > lax17s02-in-x0e.1e100.net: [icmp6 sum ok] ICMP6, echo request, seq 1113
^C
7 packets captured
7 packets received by filter
0 packets dropped by kernel
[root@50KVM-2017114448 vpn6]# ifconfig -a vpn6
vpn6: flags=4305<UP,POINTOPOINT,RUNNING,NOARP,MULTICAST>  mtu 1500
        inet6 fe80::7ed7:7067:faa7:2f33  prefixlen 64  scopeid 0x20<link>
        inet6 2001:470:f442::1  prefixlen 64  scopeid 0x0<global>
        unspec 00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00  txqueuelen 1000  (UNSPEC)
        RX packets 11072  bytes 1024865 (1000.8 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 2575  bytes 601153 (587.0 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
```
Another tinc host vps tcpdump:
```
[root@vps_server vpn6]# tcpdump -i vpn6 -vv host 2607:f8b0:4007:803::200e
tcpdump: listening on vpn6, link-type RAW (Raw IP), capture size 262144 bytes
^C
0 packets captured
0 packets received by filter
0 packets dropped by kernel
[root@vps_server vpn6]# ifconfig -a vpn6
vpn6: flags=4305<UP,POINTOPOINT,RUNNING,NOARP,MULTICAST>  mtu 1500
        inet6 fe80::73c:776e:62e0:980d  prefixlen 64  scopeid 0x20<link>
        inet6 2001:470:f2d2::1  prefixlen 64  scopeid 0x0<global>
        unspec 00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00  txqueuelen 1000  (UNSPEC)
        RX packets 2213  bytes 262642 (256.4 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 2448  bytes 782314 (763.9 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

[root@vps_server vpn6]# 
[... 28 lines cut ...]

[root@vps_server vpn6]# tcpdump -i vpn6 -vv host 2607:f8b0:4007:803::200e
tcpdump: listening on vpn6, link-type RAW (Raw IP), capture size 262144 bytes
02:53:43.590206 IP6 (hlim 127, next-header ICMPv6 (58) payload length: 40) 2001:470:f2d2:3:4827:ea57:46f4:8dff > lax17s38-in-x0e.1e100.net: [icmp6 sum ok] ICMP6, echo request, seq 1118
02:53:43.591261 IP6 (hlim 56, next-header ICMPv6 (58) payload length: 40) lax17s38-in-x0e.1e100.net > 2001:470:f2d2:3:4827:ea57:46f4:8dff: [icmp6 sum ok] ICMP6, echo reply, seq 1118
02:53:44.607075 IP6 (hlim 127, next-header ICMPv6 (58) payload length: 40) 2001:470:f2d2:3:4827:ea57:46f4:8dff > lax17s38-in-x0e.1e100.net: [icmp6 sum ok] ICMP6, echo request, seq 1119
02:53:44.608284 IP6 (hlim 56, next-header ICMPv6 (58) payload length: 40) lax17s38-in-x0e.1e100.net > 2001:470:f2d2:3:4827:ea57:46f4:8dff: [icmp6 sum ok] ICMP6, echo reply, seq 1119
02:53:45.624576 IP6 (hlim 127, next-header ICMPv6 (58) payload length: 40) 2001:470:f2d2:3:4827:ea57:46f4:8dff > lax17s38-in-x0e.1e100.net: [icmp6 sum ok] ICMP6, echo request, seq 1120
02:53:45.625796 IP6 (hlim 56, next-header ICMPv6 (58) payload length: 40) lax17s38-in-x0e.1e100.net > 2001:470:f2d2:3:4827:ea57:46f4:8dff: [icmp6 sum ok] ICMP6, echo reply, seq 1120
02:53:46.641406 IP6 (hlim 127, next-header ICMPv6 (58) payload length: 40) 2001:470:f2d2:3:4827:ea57:46f4:8dff > lax17s38-in-x0e.1e100.net: [icmp6 sum ok] ICMP6, echo request, seq 1121
02:53:46.642594 IP6 (hlim 56, next-header ICMPv6 (58) payload length: 40) lax17s38-in-x0e.1e100.net > 2001:470:f2d2:3:4827:ea57:46f4:8dff: [icmp6 sum ok] ICMP6, echo reply, seq 1121
02:53:47.648487 IP6 (hlim 127, next-header ICMPv6 (58) payload length: 40) 2001:470:f2d2:3:4827:ea57:46f4:8dff > lax17s38-in-x0e.1e100.net: [icmp6 sum ok] ICMP6, echo request, seq 1122
[... cut ...]
02:53:49.676500 IP6 (hlim 56, next-header ICMPv6 (58) payload length: 40) lax17s38-in-x0e.1e100.net > 2001:470:f2d2:3:4827:ea57:46f4:8dff: [icmp6 sum ok] ICMP6, echo reply, seq 1124
02:53:50.692162 IP6 (hlim 127, next-header ICMPv6 (58) payload length: 40) 2001:470:f2d2:3:4827:ea57:46f4:8dff > lax17s38-in-x0e.1e100.net: [icmp6 sum ok] ICMP6, echo request, seq 1125
02:53:50.693418 IP6 (hlim 56, next-header ICMPv6 (58) payload length: 40) lax17s38-in-x0e.1e100.net > 2001:470:f2d2:3:4827:ea57:46f4:8dff: [icmp6 sum ok] ICMP6, echo reply, seq 1125
^C
16 packets captured
16 packets received by filter
0 packets dropped by kernel
```
Or did I missconfigured?

feckert, 2018-05-04:
@luizluca

> Yes, at least for my case. But we might also need to add more stuff there for other cases.

what else stuff should we add to the mwan3_hook?

the following stuff 

> ip6tables --table mangle -I OUTPUT 1 -p ipv6-icmp -m icmp6 --icmpv6-type 134 -j ACCEPT
> ip6tables --table mangle -I OUTPUT 1 -p ipv6-icmp -m icmp6 --icmpv6-type 135 -j ACCEPT
> ip6tables --table mangle -I OUTPUT 1 -p ipv6-icmp -m icmp6 --icmpv6-type 137 -j ACCEPT

could be added to this function in the source
https://github.com/openwrt/packages/blob/a40234905eef5af99cc593ac36637f39d816dd1c/net/mwan3/files/lib/mwan3/mwan3.sh#L175-L218
I think we should add also a uci mwa3.globals option to enable/disable adding this rules.

luizluca, 2018-05-05:
@feckert , I don't know what else. I know that mwan3 is breaking NDP (icmp 135). However, my internet uses fixed IPv6 addresses. I would add the icmpv6-type 135 rule as a first step that fixes (or workaround) a known problem.

Maybe a mwan3.global is not needed for it. It really simply avoid mwan3 to classify a ICMPv6 package that makes sense only on a "link scope". I don't expect bad side effects.

I would suggest to add (not tested)

`
[ "$IPT" = "$IPT6" ] && $IPT -A mwan3_hook -p ipv6-icmp -m icmp6 --icmpv6-type 135 -j ACCEPT
`

here:

https://github.com/openwrt/packages/blob/a40234905eef5af99cc593ac36637f39d816dd1c/net/mwan3/files/lib/mwan3/mwan3.sh#L200

luizluca, 2018-05-07:
@feckert , I just confirmed that besides ICMPv6 135, we do need to avoid ICMPv6 136 as well.

```
if [ "$IPT" = "$IPT6" ]; then
    $IPT6 -A mwan3_hook -p ipv6-icmp -m icmp6 --icmpv6-type 135 -j ACCEPT
    $IPT6 -A mwan3_hook -p ipv6-icmp -m icmp6 --icmpv6-type 136 -j ACCEPT
fi
```

wackejohn, 2018-05-19:
@feckert 
How to fix the problem that mwan3 won't work properly with two ipv6 connections?

feckert, 2018-05-22:
@wackejohn  i though @lucize would open a PR to fix this but i could to this also. I will open a PR with the fix and then you could test this.

wackejohn, 2018-05-22:
@feckert 
Thank you for your reply.
> I will open a PR with the fix and then you could test this.
Can this PR fix send ipv6 package to wrong interface issue?

luizluca, 2018-05-22:
@feckert , I thought you would do it ;-) Just a little comm problem.
However, I will not be able to do it in the following weeks (I'm out of office).

BTW, I guess you mistyped my account.

feckert, 2018-05-23:
@wackejohn 
> Can this PR fix send ipv6 package to wrong interface issue?

You have to test this.

feckert, 2018-05-23:
@wackejohn i have prepared a PR with the source suggestion from @luizluca 
https://github.com/TDT-AG/packages/tree/pr/20180523-net-mwan3-fix-ndp
please test the changes

feckert, 2018-05-23:
I have found the following ICMPv6 types which are used for RA

> Summary of ICMPv6 messages type 133 through 137 are used for IPv6 Neighbor Discovery
> 
> 133 : Router Solicitation
> 134 : Router Advertisement
> 135 : Neighbor Solicitation
> 136 : Neighbor Advertisement
> 137 : Redirect

i could add them all ?

wackejohn, 2018-05-23:
@feckert 
Maybe you can add them all,in my case,the 134 type must be added.
And I was trying to fix send ipv6 package to wrong interface issue with config bellow:
`
config rule 'ipv6_wanc'
        option proto 'all'
        option family 'ipv6'
        option equalize '1'
        option use_policy 'wanc_only
        option sticky '0'
        option src_ip '2001:470:f442:2::/64'

config rule 'ipv6_wand'
        option proto 'all'
        option family 'ipv6'
        option equalize '1'
        option use_policy 'wand_only'
        option sticky '0'
        option src_ip '2001:470:f2d2:2::/64'

config rule 'ipv6_wanc_wand'
        option proto 'all'
        option family 'ipv6'
        option equalize '1'
        option use_policy 'wanc_wand'
        option sticky '0'
`
After some test,I found that the windows was working,but linux wasn't working... Any idea?

feckert, 2018-05-30:
@wackejohn @luizluca I have update my upcoming PR for ipv6 NDP problem
https://github.com/TDT-AG/packages/commits/pr/20180523-net-mwan3-fix-ndp
maybe you could test this? I have change the iptables target from ACCEPT to RETURN if some else is interested in the mangle table on the package type.

I have no ipv6 in use :-1: but i have the focus on this in the next mouths.
So from my side thank you for reporting/testing :smile:

wackejohn, 2018-05-30:
@feckert 
I've tested your latest PR,the RA service working properly now.
But for two ipv6 connections failover issue,do you have any idea to fix it?

feckert, 2018-05-30:
@wackejohn 
> But for two ipv6 connections failover issue,do you have any idea to fix it?

For clarification a ping from a lan (Windows/Linux) device did not leave the router over the right wan interface?
Or do we talk about the mwan3tracker on the router it self?

wackejohn, 2018-05-30:
@feckert 
We are talking about a ipv6 ping from lan (windows/linux) device didn't leave the router over the right wan interface.You can see the commit on 15 Apr.

feckert, 2018-05-30:
@wackejohn 
on the first view i could not see a miss configuration.

what is the option equalize ?
I dont know them.

luizluca, 2018-05-30:
@feckert , sorry but I'll be away from office for two weeks. I'll test it ASAP.

wackejohn, 2018-05-31:
@feckert
Maybe it's about the IPv6 source address selection problem in linux.My linux PC won't select the right IPv6 source address when one of IPv6 wan down.(eg:wanc is down,but my linux PC still use wanc's subnet address as source address,then make the connection failed)

wackejohn, 2018-05-31:
@feckert 
I've fixed the issue by tuned off the "force_link" option in wanc and wand.Both linux and windows are working properly with the config bellow:
```
config rule 'ipv6_wanc'
        option proto 'all'
        option family 'ipv6'
        option use_policy 'wanc_only
        option sticky '0'
        option src_ip '2001:470:f442:2::/64'

config rule 'ipv6_wand'
        option proto 'all'
        option family 'ipv6'
        option use_policy 'wand_only'
        option sticky '0'
        option src_ip '2001:470:f2d2:2::/64'
``` 
And I've deleted the config bellow:
```
config rule 'ipv6_wanc_wand'
        option proto 'all'
        option family 'ipv6'
        option equalize '1'
        option use_policy 'wanc_wand'
        option sticky '0'
```
For now,the ipv6 failover or loadbalance won't work with mwan3 because of the source address....

luizluca, 2018-06-01:
@wackejohn , aren't you using NAT66?

wackejohn, 2018-06-01:
@luizluca 
I'm using the ipv6 tunnel with tinc to my vps,and my vps is configued with HE's 6in4 tunnel.

feckert, 2018-06-28:
@wackejohn if the RA service is working as expected could close this issue then?

luizluca, 2018-06-29:
@wackejohn , load balancing and failover only works if you use NAT66 (I read that somewhere). mwan3 could configure it for you, but it is a little outside of the scope of mwan3.

wackejohn, 2018-06-29:
@luizluca @feckert 
Maybe I should try the NAT66 to see if it will work or not, thanks for all your works.

closed 2018-06-29 by wackejohn