openwrt/packages #5930: LMDB: not found in packages

LLE8, 2018-04-18:
Maintainer:  unknown
Environment: LEDE 17.01.4

Tested in LEDE 17.01.4 on TP-Link TL-WDR4300. This is beta, full of bugs!

Makefile:
```
# Copyright (C) 2006-2010 OpenWrt.org
#
# This is free software, licensed under the GNU General Public License v2.
# See /LICENSE for more information.
#

##LMDB for LEDE 17.01.4 LLE8

include $(TOPDIR)/rules.mk
PKG_NAME:=lmdb
PKG_VERSION:=0.9.21
PKG_RELEASE:=1

PKG_SOURCE:=LMDB_$(PKG_VERSION).tar.gz
PKG_SOURCE_URL:=https://github.com/LMDB/lmdb/archive/
PKG_HASH:=1187b635a4cc415bb6972bba346121f81edd996e99b8f0816151d4090f90b559
PKG_SOURCE_SUBDIR:=lmdb-LMDB_$(PKG_VERSION)
PKG_BUILD_DIR:=$(BUILD_DIR)/lmdb-LMDB_$(PKG_VERSION)
MAKE_PATH:=libraries/liblmdb

include $(INCLUDE_DIR)/package.mk

define Package/liblmdb0
        SECTION:=libs
        CATEGORY:=Libraries
        TITLE:=Lightning Memory-Mapped Database
        URL:=https://symas.com/lmdb/
        MAINTAINER:=X3_nobodY
endef

define Package/liblmdb0/description
        Lightning Memory-Mapped Database.
endef

define Package/lmdb-utils
        SECTION:=utils
        CATEGORY:=Utilities
        TITLE:=Lightning Memory-Mapped Database
        URL:=https://symas.com/lmdb/
        MAINTAINER:=X3_nobodY
endef

define Package/lmdb-utils/description
        Lightning Memory-Mapped Database.
endef

define Package/lmdb-test
        SECTION:=utils
        CATEGORY:=Utilities
        TITLE:=Lightning Memory-Mapped Database
        URL:=https://symas.com/lmdb/
        MAINTAINER:=X3_nobodY
endef

define Package/lmdb-test/description
        Lightning Memory-Mapped Database.
endef

define Build/Compile
        $(MAKE) -C $(PKG_BUILD_DIR)/$(MAKE_PATH) \
                CC="$(TARGET_CC)" \
                CFLAGS+="$(TARGET_CFLAGS)" \
                LDFLAGS+="$(TARGET_LDFLAGS)"
endef

define Build/InstallDev
        $(INSTALL_DIR) $(1)/usr/include
        $(INSTALL_DIR) $(1)/usr/lib
        $(CP) $(PKG_BUILD_DIR)/libraries/liblmdb/lmdb.h $(1)/usr/include
        $(CP) $(PKG_BUILD_DIR)/libraries/liblmdb/liblmdb.a $(1)/usr/lib
endef

define Package/liblmdb0/install
        $(INSTALL_DIR) $(1)/usr/lib
        $(CP) $(PKG_BUILD_DIR)/libraries/liblmdb/*.so*  $(1)/usr/lib
endef

define Package/lmdb-utils/install
        $(INSTALL_DIR) $(1)/usr/bin
        $(INSTALL_BIN) $(PKG_BUILD_DIR)/libraries/liblmdb/mdb_stat $(1)/usr/bin
        $(INSTALL_BIN) $(PKG_BUILD_DIR)/libraries/liblmdb/mdb_copy $(1)/usr/bin
        $(INSTALL_BIN) $(PKG_BUILD_DIR)/libraries/liblmdb/mdb_dump $(1)/usr/bin
        $(INSTALL_BIN) $(PKG_BUILD_DIR)/libraries/liblmdb/mdb_load $(1)/usr/bin
endef

define Package/lmdb-test/install
        $(INSTALL_DIR) $(1)/usr/bin
        $(INSTALL_BIN) $(PKG_BUILD_DIR)/libraries/liblmdb/mtest $(1)/usr/bin
endef

$(eval $(call BuildPackage,liblmdb0))
$(eval $(call BuildPackage,lmdb-utils))
$(eval $(call BuildPackage,lmdb-test))
```

Patch "0001-Debian-quirks-to-make-shared-library-versioned.patch" 
copied from Debian source package, may be not required:
```
From: Debian Berkeley DB Group <pkg-db-devel@lists.alioth.debian.org>
Date: Tue, 22 Dec 2015 09:36:38 +0100
Subject: Debian quirks to make shared library versioned

---
 libraries/liblmdb/Makefile | 21 ++++++++++++---------
 1 file changed, 12 insertions(+), 9 deletions(-)

diff --git a/libraries/liblmdb/Makefile b/libraries/liblmdb/Makefile
index f254511..ddb6d91 100644
--- a/libraries/liblmdb/Makefile
+++ b/libraries/liblmdb/Makefile
@@ -27,10 +27,11 @@ CFLAGS      = $(THREADS) $(OPT) $(W) $(XCFLAGS)
 LDLIBS =
 SOLIBS =
 SOEXT  = .so
-prefix = /usr/local
+SOVER   = 0
+prefix = /usr
 exec_prefix = $(prefix)
 bindir = $(exec_prefix)/bin
-libdir = $(exec_prefix)/lib
+libdir = $(exec_prefix)/lib/$(DEB_HOST_MULTIARCH)
 includedir = $(prefix)/include
 datarootdir = $(prefix)/share
 mandir = $(datarootdir)/man
@@ -38,7 +39,7 @@ mandir = $(datarootdir)/man
 ########################################################################

 IHDRS  = lmdb.h
-ILIBS  = liblmdb.a liblmdb$(SOEXT)
+ILIBS  = liblmdb.a liblmdb$(SOEXT) liblmdb$(SOEXT).$(SOVER) liblmdb$(SOEXT).$(SOVER).0.0
 IPROGS = mdb_stat mdb_copy mdb_dump mdb_load
 IDOCS  = mdb_stat.1 mdb_copy.1 mdb_dump.1 mdb_load.1
 PROGS  = $(IPROGS) mtest mtest2 mtest3 mtest4 mtest5
@@ -49,13 +50,13 @@ install: $(ILIBS) $(IPROGS) $(IHDRS)
        mkdir -p $(DESTDIR)$(libdir)
        mkdir -p $(DESTDIR)$(includedir)
        mkdir -p $(DESTDIR)$(mandir)/man1
-       for f in $(IPROGS); do cp $$f $(DESTDIR)$(bindir); done
-       for f in $(ILIBS); do cp $$f $(DESTDIR)$(libdir); done
-       for f in $(IHDRS); do cp $$f $(DESTDIR)$(includedir); done
-       for f in $(IDOCS); do cp $$f $(DESTDIR)$(mandir)/man1; done
+       for f in $(IPROGS); do cp -pP $$f $(DESTDIR)$(bindir); done
+       for f in $(ILIBS); do cp -pP $$f $(DESTDIR)$(libdir); done
+       for f in $(IHDRS); do cp -pP $$f $(DESTDIR)$(includedir); done
+       for f in $(IDOCS); do cp -pP $$f $(DESTDIR)$(mandir)/man1; done

 clean:
-       rm -rf $(PROGS) *.[ao] *.[ls]o *~ testdb
+       rm -rf $(PROGS) *.[ao] *.[ls]o *.[ls]o.* *~ testdb

 test:  all
        rm -rf testdb && mkdir testdb
@@ -66,7 +67,9 @@ liblmdb.a:    mdb.o midl.o

 liblmdb$(SOEXT):       mdb.lo midl.lo
 #      $(CC) $(LDFLAGS) -pthread -shared -Wl,-Bsymbolic -o $@ mdb.o midl.o $(SOLIBS)
-       $(CC) $(LDFLAGS) -pthread -shared -o $@ mdb.lo midl.lo $(SOLIBS)
+       $(CC) $(LDFLAGS) -fPIC -Wl,-soname,liblmdb$(SOEXT).$(SOVER) -pthread -Wl,-Bsymbolic -shared -o $@.$(SOVER).0.0 mdb.lo midl.lo $(SOLIBS)
+       ln -s $@.$(SOVER).0.0 liblmdb$(SOEXT).$(SOVER)
+       ln -s liblmdb$(SOEXT).$(SOVER) liblmdb$(SOEXT)

 mdb_stat: mdb_stat.o liblmdb.a
 mdb_copy: mdb_copy.o liblmdb.a
```

diizzyy, 2018-07-05:
Please close this, do a proper pull request instead
@hnyman @champtar

diizzyy, 2018-07-20:
Ping

closed 2018-08-17 by dibdot