openwrt/packages #5932: mwan3: VMs fail to get address from router DHCP when "Local source interface" is set to "lan"

celticnestor, 2018-04-18:
Maintainer: @feckert 

Environment: (put here arch, model, OpenWrt version)
Linksys WRT1900ACS
FW Lede r6565-fd588dbf6b
LuCI Master (git-18.088.49307-90ed423)
Kernel Version	4.14.29
mwan3 2.6.12-1

Description:
My setup includes in addition to the router a laptop with the following config:

1. Host runs Win10 and Hyper-V

2. Guest VM runs Ubuntu 17.10

Router DHCP server is able to serve an address to the Ubuntu machine when mwan3 "Local source interface" is set to "none". **When I set "Local source interface" to "lan", Ubuntu VM can not get an address from the router any longer.**

Happy to submit any additional info as need

feckert, 2018-05-30:
@celticnestor 
if you set the Local source interface to lan then only router initiated traffic should be affected. The lan network for example 192.168.0.0/24 is a know network and should not be touched.
What is the output of *mwan3 status* and what is your uci *network* an *mwan3* config.

celticnestor, 2018-06-12:
Hi @feckert 

I did additional research and problem only presents itself when machine running Hyper-V is connected to the network over WiFi. In this config host machine will load "Microsoft MAC bridge" to enable guest OS have access to the internet, when this driver is enabled, network problems start to happen.

Problem does not occur when host connects over wired Ethernet.

As I am not having enough time to debug this I am going wired... thx for your help anyway!

closed 2018-06-12 by celticnestor

luizluca, 2018-06-12:
@celticnestor. It look like the same problem of https://www.virtualbox.org/ticket/17186 . Maybe it's a hyper-v bug or a configuration problem.

If a pkg arrives at the AP using a different mac (guest mac) address used to connect to it (host mac), it will never return. That's expected. It is a similar problem that WDS tries to solve. You can try another vm software. Vbox always uses host mac when bridging on wireless, with some extra magic hacks to reroute vm pkgs to the guest. Hyper-v might not use it or it needs a special config.