openwrt/packages #5942: Share tvheadend 4.1

9000000, 2018-04-19:
Please make sure that the issue subject starts with `<package-name>: `

This repo here is only for packages maintained in this repo.  For base packages residing in the same repo as the build system and maintained by core devs, please consider opening tickets there for more timely responses

 - OpenWrt base system: https://bugs.openwrt.org
 - Most LuCI packages: https://github.com/openwrt/luci/issues

# Issue template (remove lines from top till here)

Maintainer: @\<github-user> (find it by checking history of the package Makefile)
Environment: (put here arch, model, OpenWrt version)

Description:

```
Formating code blocks by wrapping them with pairs of ```
```

hnyman, 2018-04-22:
not a bug, bit an issue. 
Looks like stuff for the forum.

Like said, jump in to be the maintainer

closed 2018-04-22 by hnyman

ajsomething, 2018-04-22:
These packages will not install on ddwrt router. incorrect arch. need armv7 like the current repo