openwrt/packages #5943: mwan3: flow_offloading

lucize, 2018-04-19:
mwan3: Hawing "Local source interface" is set to "lan" would break routing if flow_offloading is enabled, also I encountered the following oops when enabled flow_offloading_hw

setting it to wan, wanb or none seems to work properly

```
[  425.886392] ------------[ cut here ]------------
[  425.886402] WARNING: CPU: 2 PID: 0 at net/netfilter/nf_conntrack_rtcache.c:197 0x8f0463d8
[  425.886404] Modules linked in: pppoe ppp_async pppox ppp_generic nf_conntrack_ipv6 iptable_nat ipt_REJECT ipt_MASQUERADE xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_recent xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_helper xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_TCPMSS xt_REDIRECT xt_NETMAP xt_LOG xt_HL xt_FLOWOFFLOAD xt_DSCP xt_CLASSIFY wireguard slhc nf_reject_ipv4 nf_nat_redirect nf_nat_masquerade_ipv4 nf_conntrack_ipv4 nf_nat_ipv4 nf_nat nf_log_ipv4 nf_flow_table_hw nf_flow_table nf_defrag_ipv6 nf_defrag_ipv4 nf_conntrack_rtcache nf_conntrack iptable_mangle iptable_filter ipt_ECN ip_tables crc_ccitt xt_set ip_set_list_set ip_set_hash_netiface ip_set_hash_netport ip_set_hash_netnet ip_set_hash_net ip_set_hash_netportnet ip_set_hash_mac
[  425.886531]  ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink ip6t_REJECT nf_reject_ipv6 nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables x_tables ip6_udp_tunnel udp_tunnel leds_gpio gpio_button_hotplug
[  425.886579] CPU: 2 PID: 0 Comm: swapper/2 Tainted: G        W       4.14.34 #0
[  425.886581] Stack : 00000000 00000000 00000000 00000000 805b7ada 00000042 00000000 00000000
[  425.886599]         8fc44834 805508a7 804e1b0c 00000002 00000000 00000001 8fc11c40 2c45ddf5
[  425.886617]         00000000 00000000 805b0000 e3ddf769 00000004 80440fec b2a193a6 00089e70
[  425.886636]         00000000 80550000 0000c195 805e0000 00000000 00000000 80570000 8f0463d8
[  425.886654]         00000009 000000c5 00000001 00000003 00000002 80550000 00000008 805b0008
[  425.886672]         ...
[  425.886677] Call Trace:
[  425.886685] [<80010498>] show_stack+0x58/0x100
[  425.886693] [<8042a1dc>] dump_stack+0x9c/0xe0
[  425.886700] [<8002e1d8>] __warn+0xe0/0x114
[  425.886708] [<8002e29c>] warn_slowpath_null+0x1c/0x30
[  425.886720] [<8f0463d8>] 0x8f0463d8
[  425.886728] ---[ end trace 7c39a3f673569d93 ]---
[  425.886808] ------------[ cut here ]------------
```

luizluca, 2018-04-23:
For reference, also in https://forum.lede-project.org/t/netfilter-flow-offload-hw-nat/10237/38

nos1609, 2018-05-13:
https://bugs.openwrt.org/index.php?do=details&task_id=1539

lucize, 2018-05-28:
I also saw that nf_conntrack_max will be full very quickly (some hours) with just 3/4 devices, so maybe changing the defaults  to something much higher would help in mitigating the hangs and timeouts with page loading

lucize, 2018-06-03:
Is not crashing anymore and browsing si smooth, maybe it was wireguard fault, anyhow 6 active devices are using 61387 / 90345 (67%) from conntrack_table so that was a real problem

feckert, 2018-09-22:
@lucize I think the local_source option was a wrong design decision.
Could you set this to none?
There are a lot of problems with this option on VPN (IPsec and OpenVPN)
The problem is that if you set this option to none no load balancing and no fail-over is working for router initiated traffic.

- Load balancing could not be done with router initiated traffic because we have no firewall mangle table before routing decision.

- Failover maybe could be solved with this proposal https://github.com/TDT-AG/packages/commit/eddc8dc19a9e561d842ea9ec8d1f0693f7824f23 please have a try

feckert, 2019-03-28:
No reaction so I will close this PR.
I think the issue was fixed, because local source was removed.
So it works for me.

closed 2019-03-28 by feckert