openwrt/packages #5952: mwan3: can't configure with dual-stack IPV6 over PPPoE

cwbsw, 2018-04-22:
Maintainer: @Florian Eckert 
Environment: x86 model, mwan3 2.6.14

Description:

My ISP provides dual stack IPV6 over PPPoE, I'm assigned a private IPV4 address and a dhcpv6-pd delegated ::/56 prefix after PPPoE Authentication, so the wan interface have both IPV4 and IPV6 address, but mwan3 can only treat it as IPV4 family or IPV6 family, if one works, the another one doesn't. 

Is it desired or a bug?

luizluca, 2018-04-22:
You have to create an interface in mwan3 for each protocol family.

cwbsw, 2018-04-23:
@luizluca It means that one interface's IPV4 and IPV6 address can not work at the same time? because one interface in /etc/config/work can only be added to mwan3 once.

luizluca, 2018-04-23:
@cwbsw , the little I know about mwan3 internals tells that it will use a interface either as IPv4 or a IPv6 interface.

Check this:
https://github.com/openwrt/packages/blob/master/net/mwan3/files/lib/mwan3/mwan3.sh#L224

mwan3 reads family option and decide whether to use ipv4 or ipv6. If you do not have family defined, it defaults to ipv4. It might not be the only place it uses a logic like that.

As mwan3 simply uses the same OpenWrt interfaces, you must split them in something like wan, wan6 does. I also have a router with pppoe and dhcpv6 (but I do not use mwan3 there) and I use something like this:

```
config interface 'my_dsl'
	option proto 'pppoe'
	option ifname 'eth1'
	option username 'xxx'
	option password 'yyy'
	option delegate '0'

config interface 'my_dsl6'
	option ifname '@my_dsl'
	option proto 'dhcpv6'
	option reqaddress 'try'
	option reqprefix '56'
```

So, AFAIK, you must do the same. In mwan3, configure both my_dsl and my_dsl6 as independent interfaces and do specify family. You'll also have to duplicate members. In policy, you'll be able to add the two members (v4 and v6).

cwbsw, 2018-04-23:
@luizluca  I understand, it's resolved. Thanks very much.

closed 2018-04-23 by cwbsw

minhng99, 2019-01-18:
@cwbsw could you post your /etc/config/network and /etc/config/mwan3 here? I'm having the same issue but I couldn't fix it even when i've changed the `option ifname '@my_dsl'`

sjz123321, 2025-03-02:
@luizluca Yes it works. I am using openwrt-23.05.
In new openwrt version. instead `option ifname '@my_dsl'` with `option device '@my_dsl'`
or fixed it automaticly via click luci -> network -> interface