openwrt/packages #5965: ddns-scripts: extra options for wget

vortigont, 2018-04-25:
Maintainer: @chris5560
Environment: ramips/mt7621 snapshot 'r6643+29-08ccfde'

Description:
There is no way to specify additional options for wget, or am I missing something?
The issue with "now-dns.com" service is that it's API uses basic auth via https, but Wget >1.10.2 doesn't send it by default. It's required to pass ‘--auth-no-challenge’ option to enable it.

I've hacked 'dynamic_dns_functions.sh' script as a WA.
```
-__PROG="$WGET_SSL -nv -t 1 -O $DATFILE -o $ERRFILE"
+__PROG="$WGET_SSL -nv -t 1 -O $DATFILE -o $ERRFILE --auth-no-challenge"
```
But it should be resolved with some tunable knob.
I can make a PR with a patch, something like adding $wget_flags config option or $auth_basic boolean that could be passed to wget. What do you think?

vortigont, 2018-05-11:
@chris5560 
now-dns is included in services list but it is not working out of the box for this reasons:
 - by default ddns-script prefers wget if installed both with curl
 - ddns-script tries to update IP with wget but unable to catch the error (it reports OK but IP is not changed)
Debugging this could be quite annoying for users. Pls, see my PR #6039 for more details.
Yes, switching to curl could help in this case, but:
 - curl silently leaks plain-text password to the wire if not forced to https with config option
 - you need to know that you must use curl and https with now-dns and which config option to use

Actually curl just does that thing that was disabled in wget by default - sends clear-text password before trying challenge-responce. But since now-ip does not allow updates via plain-text http there is no risk in using --auth-no-challenge with wget, this option just makes wget behave like curl by default.
I realize that there is no clean solution for that - we could either use curl but we need to know how to setup ddns-script config for now-dns service or we could implement additional on-demand options for wget but anyway we need to know that it is required for specific service. Maybe extra options in service file for each service (like force_https, curl/wget args etc...) could do, but this is a topic for further discussion.
Thanks.

chris5560, 2018-05-13:
I agree to merge #6039 for short term.
Force https can be solved inside service/service_ipv6 by modifing the url like you did.

Because "now-dns.com" is currently the only service producing problems the best sollution will be to write as separate script like already exist for No-IP, Amazon, Cloudflair, and others.
Here all the service specific settings can be done.

vortigont, 2018-05-14:
Thanks @chris5560, I'll try to look into solution with a separate script to make both wget and curl work with now-dns.

closed 2018-05-14 by vortigont

stokito, 2022-12-05:
So the fix for the `now-dns.com` was only about switching to https but not related to the auth challenge.

But we still should add the `--auth-no-challenge` option simply because it makes unnecessary round trip with two requests instead of one.
This makes an additional load to a router and the ddns provider and should be avoided.
It was implemented in the wget because it imitates a human in browser which first sees a login page and then types credentials.
It doesn't adds any additional security. Especially if the service url is preconfigred and always valid.
Even if some ddns provider changed their domain and now on that place is some another site they still can answer 401 and then receive the password. Which still useless because the service gone.

Please just add the `--auth-no-challenge` option to wget.
It's even supported by uclient-fetch.
I can send a PR if you don't have a time for this.