openwrt/packages #5971: ddns-scripts: can not detect local IPv6

EnergyMouse, 2018-04-26:
Maintainer: @chris5560
Environment: mipsel_24kc, ramips/mt7620, r6703-bb46520159.

```
root@hostname:~# logread -e cloudflare_ipv6
Thu Apr 26 11:54:28 2018 user.notice ddns-scripts[2488]: cloudflare_ipv6: PID '2488' started at 2018-04-26 11:54
Thu Apr 26 11:54:30 2018 user.err ddns-scripts[2488]: cloudflare_ipv6: Can not detect local IP using network_get_ipaddr6 'wan6' - Error: '1' - TERMINATE
Thu Apr 26 11:54:30 2018 user.warn ddns-scripts[2488]: cloudflare_ipv6: PID '2488' exit WITH ERROR '1' at 2018-04-26 11:54
root@hostname:~# cat /var/log/ddns/cloudflare_ipv6.log 
 115428       : ************ ************** ************** **************
 115428  note : PID '2488' started at 2018-04-26 11:54
 115428       : ddns version  : 2.7.7-4
 115428       : uci configuration:\nddns.cloudflare_ipv6.domain='hostname@domain.gltd'
ddns.cloudflare_ipv6.enabled='1'
ddns.cloudflare_ipv6.interface='wan6'
ddns.cloudflare_ipv6.ip_network='wan6'
ddns.cloudflare_ipv6.ip_source='network'
ddns.cloudflare_ipv6.lookup_host='hostname.domain.gltd'
ddns.cloudflare_ipv6.param_opt='proxied=false'
ddns.cloudflare_ipv6.password='*password*'
ddns.cloudflare_ipv6.service_name='cloudflare.com-v4'
ddns.cloudflare_ipv6.use_ipv6='1'
ddns.cloudflare_ipv6.username='email@email.com'
ddns.cloudflare_ipv6=service
 115428       : verbose mode  : 0 - run normal, NO console output
 115429       : check interval: 600 seconds
 115429       : force interval: 259200 seconds
 115429       : retry interval: 60 seconds
 115429       : retry counter : 0 times
 115429       : No old process
 115429       : last update: never
 115429       : Detect registered/public IP
 115429       : #> /usr/bin/nslookup hostname.domain.gltd  >/var/run/ddns/cloudflare_ipv6.dat 2>/var/run/ddns/cloudflare_ipv6.err
 115430       : Registered IP '2a02:2698:9025:e8f8::1' detected
 115430  info : Starting main loop at 2018-04-26 11:54
 115430       : Detect local IP on 'network'
 115430 ERROR : Can not detect local IP using network_get_ipaddr6 'wan6' - Error: '1' - TERMINATE
 115430  WARN : PID '2488' exit WITH ERROR '1' at 2018-04-26 11:54\n
root@hostname:~# ip -6 a
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 state UNKNOWN qlen 1000
    inet6 ::1/128 scope host 
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 state UNKNOWN qlen 1000
    inet6 fe80::f2b4:29ff:fe5e:571d/64 scope link 
       valid_lft forever preferred_lft forever
10: br-lan: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 state UP qlen 1000
    inet6 2a02:2698:9025:ea0f::1/64 scope global dynamic 
       valid_lft 85641sec preferred_lft 2841sec
    inet6 fd22:2715:93c1::1/60 scope global 
       valid_lft forever preferred_lft forever
    inet6 fe80::f0b4:29ff:fe5e:571d/64 scope link 
       valid_lft forever preferred_lft forever
12: eth0.2@eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 state UP qlen 1000
    inet6 fe80::f2b4:29ff:fe5e:571d/64 scope link 
       valid_lft forever preferred_lft forever
13: pppoe-wan: <POINTOPOINT,MULTICAST,NOARP,UP,LOWER_UP> mtu 1492 state UNKNOWN qlen 3
    inet6 fe80::a1d8:7f31:e5a6:bf7e/10 scope link 
       valid_lft forever preferred_lft forever
```

chris5560, 2018-04-29:
As you see in your "ip -6" command your "global" IPv6 is on br-lan.
Please try to set "option ip_network 'br-lan'" or try to detect via url

dibdot, 2018-09-29:
let's close dated ddns tickets without further input.

closed 2018-09-29 by dibdot