openwrt/packages #5978: <package-name>: mwan3 2.6.14-1

Dimit77, 2018-04-29:
When I've tried to ping 'tracking iP' or 'default gateway' I got error:

/usr/lib/lua/luci/dispatcher.lua:460: Failed to execute call dispatcher target for entry '/admin/status/mwan/diagnostics_display/wan1/ping_trackips'.
The called action terminated with an exception:
/usr/lib/lua/luci/controller/mwan3.lua:102: attempt to call field 'shellquote' (a nil value)
stack traceback:
	[C]: in function 'assert'
	/usr/lib/lua/luci/dispatcher.lua:460: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:141: in function </usr/lib/lua/luci/dispatcher.lua:140>

Environment: WRT3200, 17.01.4

feckert, 2018-05-17:
Do you use a vanilla 17.01 or did you patched something? I asked because the shellquote funktion was introduced in luci master and the application luci-app-mwan3 package was moved from the package feed to the luci feed.
https://github.com/openwrt/luci/blob/1e39fef1474b9f80f1a774b4faca1c0c8a9c34f3/applications/luci-app-mwan3/luasrc/controller/mwan3.lua#L102

Dimit77, 2018-05-17:
I installed v.17.01.4 and mwan3 2.0.2
but then I updated mwan3 to 2.6.14 and got that error

feckert, 2018-05-18:
`I installed v.17.01.4 and mwan3 2.0.2
but then I updated mwan3 to 2.6.14 and got that error`

This was a error from the luci? But said you only updated the mwan3 package?

`could I install luci master from snapshots to my 17.01.4 ?`

@Dimit77 good question the **shellquote** function was add in the luci-base package.
I think it is not a good idea.

Dimit77, 2018-05-18:
Yes I only update mwan3
I got error when I try to ping IP from luci
Thanks

feckert, 2018-06-08:
See the following script
https://github.com/openwrt/packages/blob/master/net/mwan3/files/etc/hotplug.d/iface/14-mwan3
The interesting lines are
Copy line
https://github.com/openwrt/packages/blob/f93d518c2c61a03bcef200f64d57b933b42aae6e/net/mwan3/files/etc/hotplug.d/iface/14-mwan3#L1-L12

After this lines you could do what you want
These scripts will be called from netifd with the following parameters.

- ACTION=(ifup|ifdown|ifup-failed| ...)
- INTERFACE=(from etc/config/network)
- DEVICE=(physical device from this INTERFACE).

Dimit77, 2018-06-22:
I tried to write script:)
Am I right? 
if [ "$INTERFACE" = "wan1" || "$INTERFACE" = "wan2" ]; then
if [ "$ACTION = "ifdown" ]; then
sleep 70 && touch /etc/banner && reboot
fi
fi
exit 0

feckert, 2018-06-28:
@Dimit77 I do not think this is good to wait in this script for 70 seconds. The **netifd** will kill this service due timeout. I think it will be better to call a helper script witch could be detouched as a service and then could sleep and wait.

feckert, 2018-09-22:
@Dimit77 could we close this issue?

feckert, 2018-11-21:
@hnyman No reaction from issue creator. Could you please close this issue?

closed 2018-11-21 by hnyman