openwrt/packages #6002: instagram, tumblr, and the other built-in ads

AddaxSoft, 2018-05-03:
Description:

The ads that are built-in apps like instagram and tumblr do not get blocked.
The rest are blocked effectively.

dibdot, 2018-05-03:
Trace the dns queries of those (so called) social media apps and add them to you blacklist. To get all dns requests of your dnsmasq instance set the option "logqueries" and check your logfile afterwards ... 

Please remember that you block always on (sub-)domain level (and not on page element level!) - the same applies to other projects like pihole. For remaining ads on advertising platforms like youtube, facebook etc. it’s likely that you need additional (browser) plugins which are hiding ads on DOM level … anyway you’ll never get 100%.

dibdot, 2018-05-09:
no package issue, close this for now.

closed 2018-05-09 by dibdot