openwrt/packages #6004: Squid: Unable squid with error with -O3

frymezim, 2018-05-04:
I'm unable to compile squid with -O3 -pipe -mcpu=cortex-a15 -mfpu=neon-vfpv4 -mfloat-abi=hard but works fine with -O2 @ratkaj 
```
make[3]: Entering directory '/home/dkmori/openwrt/feeds/packages/net/squid'
rm -f /home/dkmori/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/squid-3.5.27/.built
touch /home/dkmori/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/squid-3.5.27/.built_check
make  -C /home/dkmori/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/squid-3.5.27/lib all
make[4]: Entering directory '/home/dkmori/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/squid-3.5.27/lib'
Making all in rfcnb
make[5]: Entering directory '/home/dkmori/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/squid-3.5.27/lib/rfcnb'
make[5]: Nothing to be done for 'all'.
make[5]: Leaving directory '/home/dkmori/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/squid-3.5.27/lib/rfcnb'
Making all in smblib
make[5]: Entering directory '/home/dkmori/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/squid-3.5.27/lib/smblib'
make[5]: Nothing to be done for 'all'.
make[5]: Leaving directory '/home/dkmori/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/squid-3.5.27/lib/smblib'
make[5]: Entering directory '/home/dkmori/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/squid-3.5.27/lib'
/bin/bash ../libtool  --tag=CC   --mode=link arm-openwrt-linux-muslgnueabi-gcc -Wall -Wpointer-arith -Wwrite-strings -Wmissing-prototypes -Wmissing-declarations -Wcomments -Wshadow -Werror -pipe -D_REENTRANT -O3 -pipe -mcpu=cortex-a15 -mfpu=neon-vfpv4 -mfloat-abi=hard -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -mfloat-abi=hard -iremap/home/dkmori/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/squid-3.5.27:squid-3.5.27 -Wformat -Werror=format-security -fPIC -fstack-protector-strong -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro   -L/home/dkmori/openwrt/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/usr/lib -L/home/dkmori/openwrt/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/lib -L/home/dkmori/openwrt/staging_dir/toolchain-arm_cortex-a15+neon-vfpv4_gcc-7.3.0_musl_eabi/usr/lib -L/home/dkmori/openwrt/staging_dir/toolchain-arm_cortex-a15+neon-vfpv4_gcc-7.3.0_musl_eabi/lib -specs=/home/dkmori/openwrt/include/hardened-ld-pie.specs -znow -zrelro  -o libmiscencoding.la  base64.lo charset.lo html_quote.lo md5.lo rfc1738.lo rfc2617.lo uudecode.lo  
libtool: link: arm-openwrt-linux-muslgnueabi-gcc-ar cru .libs/libmiscencoding.a .libs/base64.o .libs/charset.o .libs/html_quote.o .libs/md5.o .libs/rfc1738.o .libs/rfc2617.o .libs/uudecode.o 
/home/dkmori/openwrt/staging_dir/toolchain-arm_cortex-a15+neon-vfpv4_gcc-7.3.0_musl_eabi/lib/gcc/arm-openwrt-linux-muslgnueabi/7.3.0/../../../../arm-openwrt-linux-muslgnueabi/bin/ar: `u' modifier ignored since `D' is the default (see `U')
libtool: link: arm-openwrt-linux-muslgnueabi-gcc-ranlib .libs/libmiscencoding.a
libtool: link: ( cd ".libs" && rm -f "libmiscencoding.la" && ln -s "../libmiscencoding.la" "libmiscencoding.la" )
/bin/bash ../libtool  --tag=CXX   --mode=link arm-openwrt-linux-muslgnueabi-g++ -Wall -Wpointer-arith -Wwrite-strings -Wcomments -Wshadow -Woverloaded-virtual -Werror -pipe -D_REENTRANT -I/home/dkmori/openwrt/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/usr/include -O3 -pipe -mcpu=cortex-a15 -mfpu=neon-vfpv4 -mfloat-abi=hard -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -mfloat-abi=hard -iremap/home/dkmori/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/squid-3.5.27:squid-3.5.27 -Wformat -Werror=format-security -fPIC -fstack-protector-strong -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro   -L/home/dkmori/openwrt/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/usr/lib -L/home/dkmori/openwrt/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/lib -L/home/dkmori/openwrt/staging_dir/toolchain-arm_cortex-a15+neon-vfpv4_gcc-7.3.0_musl_eabi/usr/lib -L/home/dkmori/openwrt/staging_dir/toolchain-arm_cortex-a15+neon-vfpv4_gcc-7.3.0_musl_eabi/lib -specs=/home/dkmori/openwrt/include/hardened-ld-pie.specs -znow -zrelro  -o libmisccontainers.la  hash.lo  
libtool: link: arm-openwrt-linux-muslgnueabi-gcc-ar cru .libs/libmisccontainers.a .libs/hash.o 
/home/dkmori/openwrt/staging_dir/toolchain-arm_cortex-a15+neon-vfpv4_gcc-7.3.0_musl_eabi/lib/gcc/arm-openwrt-linux-muslgnueabi/7.3.0/../../../../arm-openwrt-linux-muslgnueabi/bin/ar: `u' modifier ignored since `D' is the default (see `U')
libtool: link: arm-openwrt-linux-muslgnueabi-gcc-ranlib .libs/libmisccontainers.a
libtool: link: ( cd ".libs" && rm -f "libmisccontainers.la" && ln -s "../libmisccontainers.la" "libmisccontainers.la" )
depbase=`echo rfc1035.lo | sed 's|[^/]*$|.deps/&|;s|\.lo$||'`;\
/bin/bash ../libtool  --tag=CC   --mode=compile arm-openwrt-linux-muslgnueabi-gcc -DHAVE_CONFIG_H   -I.. -I../include -I../lib -I../src -I../include    -I/home/dkmori/openwrt/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/usr/include -I/home/dkmori/openwrt/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/usr/include -I/home/dkmori/openwrt/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/include -I/home/dkmori/openwrt/staging_dir/toolchain-arm_cortex-a15+neon-vfpv4_gcc-7.3.0_musl_eabi/usr/include -I/home/dkmori/openwrt/staging_dir/toolchain-arm_cortex-a15+neon-vfpv4_gcc-7.3.0_musl_eabi/include/fortify -I/home/dkmori/openwrt/staging_dir/toolchain-arm_cortex-a15+neon-vfpv4_gcc-7.3.0_musl_eabi/include  -Wall -Wpointer-arith -Wwrite-strings -Wmissing-prototypes -Wmissing-declarations -Wcomments -Wshadow -Werror -pipe -D_REENTRANT -O3 -pipe -mcpu=cortex-a15 -mfpu=neon-vfpv4 -mfloat-abi=hard -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -mfloat-abi=hard -iremap/home/dkmori/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/squid-3.5.27:squid-3.5.27 -Wformat -Werror=format-security -fPIC -fstack-protector-strong -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro  -MT rfc1035.lo -MD -MP -MF $depbase.Tpo -c -o rfc1035.lo rfc1035.c &&\
mv -f $depbase.Tpo $depbase.Plo
libtool: compile:  arm-openwrt-linux-muslgnueabi-gcc -DHAVE_CONFIG_H -I.. -I../include -I../lib -I../src -I../include -I/home/dkmori/openwrt/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/usr/include -I/home/dkmori/openwrt/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/usr/include -I/home/dkmori/openwrt/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/include -I/home/dkmori/openwrt/staging_dir/toolchain-arm_cortex-a15+neon-vfpv4_gcc-7.3.0_musl_eabi/usr/include -I/home/dkmori/openwrt/staging_dir/toolchain-arm_cortex-a15+neon-vfpv4_gcc-7.3.0_musl_eabi/include/fortify -I/home/dkmori/openwrt/staging_dir/toolchain-arm_cortex-a15+neon-vfpv4_gcc-7.3.0_musl_eabi/include -Wall -Wpointer-arith -Wwrite-strings -Wmissing-prototypes -Wmissing-declarations -Wcomments -Wshadow -Werror -pipe -D_REENTRANT -O3 -pipe -mcpu=cortex-a15 -mfpu=neon-vfpv4 -mfloat-abi=hard -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -mfloat-abi=hard -iremap/home/dkmori/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/squid-3.5.27:squid-3.5.27 -Wformat -Werror=format-security -fPIC -fstack-protector-strong -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -MT rfc1035.lo -MD -MP -MF .deps/rfc1035.Tpo -c rfc1035.c  -fPIC -DPIC -o .libs/rfc1035.o
In file included from ../compat/xstring.h:13:0,
                 from ../compat/compat_shared.h:225,
                 from ../compat/compat.h:100,
                 from ../include/squid.h:43,
                 from rfc1035.c:17:
rfc1035.c: In function 'rfc1035NamePack':
/home/dkmori/openwrt/staging_dir/toolchain-arm_cortex-a15+neon-vfpv4_gcc-7.3.0_musl_eabi/include/fortify/string.h:51:9: error: argument 2 null where non-null expected [-Werror=nonnull]
  return __orig_memcpy(__od, __os, __n);
         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/dkmori/openwrt/staging_dir/toolchain-arm_cortex-a15+neon-vfpv4_gcc-7.3.0_musl_eabi/include/fortify/stdlib.h:28:0,
                 from ../compat/types.h:29,
                 from ../compat/compat.h:70,
                 from ../include/squid.h:43,
                 from rfc1035.c:17:
[... cut ...]
make[2]: *** [package/feeds/packages/squid/compile] Error 2
make[2]: Leaving directory '/home/dkmori/openwrt'
package/Makefile:103: recipe for target '/home/dkmori/openwrt/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/stamp/.package_compile' failed
make[1]: *** [/home/dkmori/openwrt/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/dkmori/openwrt'
/home/dkmori/openwrt/include/toplevel.mk:216: recipe for target 'world' failed
make: *** [world] Error 2
```

jow-, 2018-05-04:
Don't do it then? :) Not all software is supposed to be buildable / runnable with -O3.

frymezim, 2018-05-04:
I see thanks for the info.

closed 2018-05-04 by frymezim

jow-, 2018-05-04:
Note that my comment was only partly serious ;) A possible fix might be adding `TARGET_CFLAGS += -Wno-error=nonnull`. Allegedly this has been fixed in later gcc versions but I couldn't test myself.

frymezim, 2018-05-04:
I've been using GCC7 for my builds still got that error, I added `TARGET_CFLAGS := $(filter-out -O%,$(TARGET_CFLAGS)) -O2` on squid makefile and works fine. I'll try that flag later. Thanks again.

danielg4, 2018-07-11:
In the 18.06 branch, this identical issue occurs in the lxc package on x86, so it is neither package-specific nor target-specific, and should perhaps be moved to the OpenWrt bug tracker.

"Allegedly this has been fixed in later gcc versions…" Which versions? Alleged where? @jow-