openwrt/packages #6006: tinyproxy: Support for multiple instances

mikma, 2018-05-05:
Maintainer: Jo-Philipp Wich <jo@mein.io>
Environment: x86_64, LEDE Reboot 17.01.4 r3560-79f57e422d

Description:
Did tinyproxy in openwrt support a config file with multiple sections at one point? It doesn't to work in the current version for me, but the following patch seems to fix it.

```
--- tinyproxy.orig	2017-12-16 00:34:30.000000000 +0000
+++ tinyproxy	2018-05-05 10:25:17.252167961 +0000
@@ -3,7 +3,8 @@
 
 START=50
 
-CFGFILE=/var/etc/tinyproxy.conf
+CFGFILE_BASE=/var/etc/tinyproxy.conf
+SERVICE_PID_FILE_BASE=/var/run/tinyproxy.pid
 
 section_enabled() {
 	config_get_bool enabled "$1" 'enabled' 0
@@ -20,6 +21,7 @@
 }
 
 start_proxy() {
+	CFGFILE="${CFGFILE_BASE}.${section}"
 	section_enabled "$1" || return 1
 
 	mkdir -m0755 -p /var/etc
@@ -72,6 +74,7 @@
 
 	config_foreach write_upstream upstream
 
+	local SERVICE_PID_FILE="SERVICE_PID_FILE_BASE.$section"
 	service_start /usr/sbin/tinyproxy -c "$CFGFILE"
 }

```

diizzyy, 2019-01-31:
@jow- 
Ping

jow-, 2019-02-01:
No, it didn‘t. If you think it is useful, please open a PR to implement the desired behaviour.

closed 2019-02-01 by jow-