openwrt/packages #6012: lighttpd: 1.4.48 - missing openssl module in 17.01.04 release

bluewavenet, 2018-05-06:
Reference to #5343
I am, after 25th April 2018, getting this problem (lighttpd 1.4.48-1)
Before this date, lighttpd 1.4.45-3 was the release and did not have the issue.
Lighttpd in 17.01.04 is currently broken for SSL.

pkopa, 2018-05-06:
I can confirm that version 1.4.48-2 that fixed the issue doesn't compile on mips_24kc architecture:
http://downloads.lede-project.org/snapshots/faillogs/mips_24kc/packages/lighttpd/compile.txt

bluewavenet, 2018-05-06:
What happened to 1.4.45-3? 
**I didn't think new package releases were pushed into the current release of LEDE/OpenWrt.**
This will effect anyone installing or image building from 17.01.04.
I have hacked the components of 1.4.45-3 out of one of my previous images and built it into my latest so I have a work around.

bluewavenet, 2018-05-06:
@pkopa 
I have not checked them all but it looks like it does not compile in any of the mips or arm architectures.

avijc, 2018-05-09:
I must be blind, but pkopa's log does not seem like a build failure to me. Can you please point me where the failure is? If it does not compile, any ideas why?

avijc, 2018-05-09:
Looking at the end of that compile.txt it looks like mod_openssl.so got built, but it was not included in the package. There are various "Packaged contents of .../ipkg-mips_24kc/lighttpd-mod-accesslog" lines for the other modules, but not for mod_openssl. Perhaps the end of https://github.com/openwrt/packages/blob/master/net/lighttpd/Makefile needs to be updated to include the appropriate references to mod_openssl to get it included.

Disclaimer: I haven't used OpenWRT at all. I'm only here because someone filed an upstream bug in lighttpd's bug tracker for this OpenWRT packaging issue. I hope you can fix this by yourself.

bluewavenet, 2018-05-09:
@avijc thanks for looking. I came to the same conclusion. I am not a maintainer on here but will raise a PR
Thanks :-)

bluewavenet, 2018-05-09:
see PR #6029

pprindeville, 2018-05-14:
Isn't this a duplicate of issue #5343?

pprindeville, 2018-05-14:
@champtar How do we get a cherry-pick of PR #5344 into 17.0.4?  Thanks

bluewavenet, 2018-05-14:
@pprindeville , no not a duplicate of #5343, but pointing out a problem with the FIX for #5343
The module ipk does not get built in 1.4.48-2

bluewavenet, 2018-05-14:
The REALLY big issue here is the major change in **stable 17.01.4 packages** from lighttpd version 1.4.45-3 to the **known broken** version 1.4.48-1 on the 25th April 2018.

I was under the impression that packages in the OpenWrt/LEDE release were not to be changed apart from in exceptional circumstances.

I have a custom ImageBuilder process for production updates of custom code and this change caused considerable problems.

Version 1.4.48-2 in snapshots also has the problem that the mod_openssl ipk is not built.
My PR #6038 is to fix this issue

@champtar @jow- @pprindeville @MikePetullo

MikePetullo, 2018-05-21:
OpenWrt just merged #6077. The package at https://downloads.openwrt.org/releases/17.01.4/packages/x86_64/packages/ includes mod_openssl.so as of Mon May 21 10:18:12 2018. I think this issue can close.

bluewavenet, 2018-05-24:
Fixed! :-)

closed 2018-05-24 by bluewavenet