openwrt/packages #6024: lua-mosquitto: calling tls_set function fails

robinforge, 2018-05-09:
# Issue template (remove lines from top till here)

Maintainer: @karlp 
Environment: ar71xx, Chaos Calmer 15.05.1

Description:

tls_set function fails with following output:

lua: mqtt-session.lua:36: attempt to call method 'tls_set' (a nil value)
stack traceback:
	mqtt-session.lua:36: in main chunk
	[C]: ?

It looks like the function isn't present (I also ran a 'does function exist' piece of Lua code and this indicated the function didn't exist, whereas 'connect' function for example does exist).

If I run my Lua code on my Mac with lua-mosquitto compiled from [here](https://github.com/flukso/lua-mosquitto) then tls_set works great.

My Lua file looks like this:

```
--[[ Connection variables --]]
broker_url = "XXX"
port = "XXX"
keepalive = "60"
cafile = "XXX/CA.crt"
certfile = "/XXX/client.crt"
keyfile = "/XXX/private.key"

--[[ Load in the mosquitto module --]]
mqtt = require("mosquitto") 

--[[ Make client a new MQTT instance --]]
client = mqtt.new()

client.ON_CONNECT = function()
  print("connected")
  client:subscribe("calibration")
  local mid = client:subscribe("calibration", 0)
end

client.ON_MESSAGE = function(mid, topic, payload)
  print(payload)
end

--[[ tls_set with certificate and key data passed to it --]]
client:tls_set(cafile, nil , certfile, keyfile)

-- [[ connect object with broker and port values passed to it--]]
client:connect(broker_url, port, keepalive)
client:loop_forever()

```

karlp, 2018-05-09:
CC branch doesn't get updates.  Use 17.01 or master for up to date packages.

closed 2018-05-09 by karlp

robinforge, 2018-05-10:
Upgraded to LEDE 17.01 and it works, thanks!