openwrt/packages #6035: <ddns-scripts> ddns-scripts_cloudflare.com-v4 SSL certificate type error

pycvalade, 2018-05-11:
Maintainer: @chris5560 
Environment: `TP-Link Archer C7 v4 - LEDE 17.01 (OpenWrt SNAPSHOT r6738-a4322eb)`

Description:
I followed this: https://wiki.openwrt.org/doc/howto/ddns.client
Installed the SSL certificate manually and also tried the `ca-certificates` package

Output of enabling worker for ipv4 outputs in log file:
```
 203044       : ************ ************** ************** **************
 203044  note : PID '10503' started at 2018-05-10 20:30
 203044       : ddns version  : 2.7.7-5
 203044       : uci configuration:\nddns.cloudflare_ipv4.domain='sip@domain.com'
ddns.cloudflare_ipv4.enabled='1'
ddns.cloudflare_ipv4.interface='wan'
ddns.cloudflare_ipv4.ip_network='wan'
ddns.cloudflare_ipv4.ip_source='network'
ddns.cloudflare_ipv4.lookup_host='sip.domain.com'
ddns.cloudflare_ipv4.param_opt='true'
ddns.cloudflare_ipv4.password='*password*'
ddns.cloudflare_ipv4.service_name='cloudflare.com-v4'
ddns.cloudflare_ipv4.use_https='1'
ddns.cloudflare_ipv4.username='email@gmail.com'
ddns.cloudflare_ipv4=service
 203044       : verbose mode  : 0 - run normal, NO console output
 203045       : check interval: 600 seconds
 203045       : force interval: 259200 seconds
 203045       : retry interval: 60 seconds
 203045       : retry counter : 0 times
 203045       : No old process
 203046       : last update: never
 203046       : Detect registered/public IP
 203046       : #> /usr/bin/host -t A sip.domain.com  >/var/run/ddns/cloudflare_ipv4.dat 2>/var/run/ddns/cloudflare_ipv4.err
 203046       : Registered IP '104.27.179.44' detected
 203046  info : Starting main loop at 2018-05-10 20:30
 203046       : Detect local IP on 'network'
 203046       : Local IP '192.222.194.44' detected on network 'wan'
 203046       : Update needed - L: '192.222.194.44' <> R: '104.27.179.44'
 203047       : parsing script '/usr/lib/ddns/update_cloudflare_com_v4.sh'
 203047       : #> /usr/bin/curl -RsS -o /var/run/ddns/cloudflare_ipv4.dat --stderr /var/run/ddns/cloudflare_ipv4.err --noproxy '*' --header 'X-Auth-Email: email@gmail.com'  --header 'X-Auth-Key: *password*'  --header 'Content-Type: application/json'  --request GET 'https://api.cloudflare.com/client/v4/zones?name=domain.com'
 203047 ERROR : cURL Error: '77'
 203047       : curl: (77) Error reading ca cert file /etc/ssl/certs/ca-certificates.crt - mbedTLS: (-0x2180) X509 - The CRT/CRL/CSR format is invalid, e.g. different type expected
 203047  WARN : Transfer failed - retry 1/0 in 60 seconds
```

Problem is this:
```
 203047       : parsing script '/usr/lib/ddns/update_cloudflare_com_v4.sh'
 203047       : #> /usr/bin/curl -RsS -o /var/run/ddns/cloudflare_ipv4.dat --stderr /var/run/ddns/cloudflare_ipv4.err --noproxy '*' --header 'X-Auth-Email: email@gmail.com'  --header 'X-Auth-Key: *password*'  --header 'Content-Type: application/json'  --request GET 'https://api.cloudflare.com/client/v4/zones?name=domain.com'
 203047 ERROR : cURL Error: '77'
 203047       : curl: (77) Error reading ca cert file /etc/ssl/certs/ca-certificates.crt - mbedTLS: (-0x2180) X509 - The CRT/CRL/CSR format is invalid, e.g. different type expected
 203047  WARN : Transfer failed - retry 1/0 in 60 seconds
```

Certificate is present:
```
root@py-net:~# ls -lh /etc/ssl/certs/
-rw-r--r--    1 root     root           0 May  8 16:58 ca-certificates.crt
```

Looked up everywhere, didn't find anything about this..

dibdot, 2018-05-11:
Please doublecheck your certificates ...you've listed a 0 byte file for ca-certificates.crt

pycvalade, 2018-07-05:
For an unknown reason, following the wiki ended up with a 0 byte certificate file. The download would just never retrieve the file.

I ended up downloading it from my computer, scp'ed it to the router and everything worked properly afterwards.

closed 2018-07-05 by pycvalade