openwrt/packages #6045: CI: Travis makes use of a version of intltool which is too old to build libsoup

MikePetullo, 2018-05-13:
As seen with pull request #5371, Travis appears to make use of a version of intltool which is too old to build the latest libsoup package.

The following is the error which appears when attempting a CI build of libsoup:

```
./configure: line 12636: intltool-update: command not found
checking for intltool >= 0.35.0...  found
configure: error: Your intltool is too old.  You need intltool 0.35.0 or later.
```

diizzyy, 2018-05-13:
It should be added as a hosttool like buildroot does
https://git.busybox.net/buildroot/tree/package/intltool/

diizzyy, 2018-05-13:
I wonder if this would be worth integrating in the end
https://github.com/sabotage-linux/gettext-tiny

champtar, 2018-05-13:
I've started to work on using CircleCI instead of Travis, one of the advantages is that we use any docker image we want to build, see #5993

diizzyy, 2018-05-13:
@champtar 
Before you go ahead and mask build/compilation issues someone needs to check with the main project regarding their stance on host target. Given that 4.8 is still considered acceptable (https://github.com/lede-project/source/blob/master/include/prereq-build.mk#L33 (?)) you can most likely assume that everything else is going to be ancient. I don't think it's a good idea to have different requirements than LEDE/OpenWrt but again, no guidelines etc....

thess, 2018-11-13:
Should be fixed by #7413

closed 2018-11-14 by thess