openwrt/packages #6065: php7-mod-curl missing

B0F1B0, 2018-05-17:
System: Lede 17.01.4
Target: mvebu

I want install Owncloud / Nextcloud, but php7-mod-curl packages is not avaliable. 

link: 
http://downloads.lede-project.org/releases/17.01.4/packages/arm_cortex-a9_vfpv3/packages/

I tried to compile it from souce on my own, but it doesnt work. 

by the way, php7-mod-curl is not avaliable for any MACHINE/SYSTEM/ROUTER on LEDE

checkout here. 
http://downloads.lede-project.org/releases/17.01.4/packages/

mhei, 2018-05-17:
You are right, for some reason php7-mod-curl is missing. But I don't know why it failed to build.
But I tried to compile it locally from scratch and this worked for me. Can you provide me a log and config or your failed build?
On the other hand, if it is not urgent, you could also wait a few weeks for upcoming 18.06 release...

B0F1B0, 2018-05-17:
when i try to compile php7-mod-curl with 

> 
http://downloads.lede-project.org/releases/17.01.4/targets/mvebu/generic/lede-sdk-17.01.4-mvebu_gcc-5.4.0_musl-1.1.16_eabi.Linux-x86_64.tar.xz
> 
i get a lot of compile errors. most of them i was able to manage on my own. my last error message is: 

> configure: error: Sorry, I was not able to diagnose which libmcrypt version you have installed

LOG:

`checking for libmcrypt version... >= 2.5.6
checking for mcrypt_module_open in -lmcrypt... no
checking for mcrypt_module_open in -lmcrypt... no
configure: error: Sorry, I was not able to diagnose which libmcrypt version you have installed.
Makefile:547: recipe for target '/home/bofibo/Downloads/sdk/build_dir/target-arm_cortex-a9+vfpv3_musl-1.1.16_eabi/php-7.1.1/.configured_nnnnnnnnnnnnnnnnnnnnnnnnnnnnnnnnnnnnnnnnnnnnyynynn' failed
make[3]: *** [/home/bofibo/Downloads/sdk/build_dir/target-arm_cortex-a9+vfpv3_musl-1.1.16_eabi/php-7.1.1/.configured_nnnnnnnnnnnnnnnnnnnnnnnnnnnnnnnnnnnnnnnnnnnnyynynn] Error 1
make[3]: Leaving directory '/home/bofibo/Downloads/sdk/feeds/packages/lang/php7'
package/Makefile:105: recipe for target 'package/feeds/packages/php7/compile' failed
make[2]: *** [package/feeds/packages/php7/compile] Error 2
make[2]: Leaving directory '/home/bofibo/Downloads/sdk'
package/Makefile:101: recipe for target '/home/bofibo/Downloads/sdk/staging_dir/target-arm_cortex-a9+vfpv3_musl-1.1.16_eabi/stamp/.package_compile' failed
make[1]: *** [/home/bofibo/Downloads/sdk/staging_dir/target-arm_cortex-a9+vfpv3_musl-1.1.16_eabi/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/bofibo/Downloads/sdk'
/home/bofibo/Downloads/sdk/include/toplevel.mk:191: recipe for target 'world' failed
make: *** [world] Error 2
bofibo@debian8:~/Downloads/sdk$ 
`

it is possible that you can compile it for "MVEBU" and send me the ipk? or tell me what i make wrong?

mhei, 2018-05-18:
Ah ok - I didn't used the SDK but compiled within usual build system from scratch. This might also be the reason for missing packages for download. However, please have a look whether this package attached here works for you:
[php7-mod-curl_7.1.1-1_arm_cortex-a9_vfpv3.ipk.zip](https://github.com/openwrt/packages/files/2017560/php7-mod-curl_7.1.1-1_arm_cortex-a9_vfpv3.ipk.zip)

B0F1B0, 2018-05-18:
thank you very much.

i thought I need a cross-compiler?! 

can you maybe tell me how i can compile it without the sdk? how do i tell Linux for wich "CPU" it is?

mhei, 2018-05-18:
Yes, you need a cross-compiler. I tried the SDK meanwhile, but failed, too. I think you approach is the right one and this should definitively work. Could you try to edit `package/feeds/packages/php7/Makefile` and delete all occurences of `$(SDK)` , i.e. replace this pattern with an empty string? Then run `make` again. My guess is, that dependencies are not correctly pulled and the build tries to access libraries which are not needed and/or are not built yet.

As an alternative way, you can simply checkout OpenWrt/LEDE sources, install package feed and then run `make menuconfig` where you select your platform and php packages. Then running `make` will build not only php packages, but also other packages and complete image etc.

B0F1B0, 2018-05-18:
sorry for my lot of questions, but i really want to figure out how I can do it on my own.
you was able to compile php7-mod-curl so please tell me how do you do it? 

what i have done :
extract the sdk in debian9
./scripts/feeds update -a
./scripts/feeds install php7-mod-curl
make menuconfig
        - >select php7 and mod-curl under       
           language
make

but as i told you, i got a lot of compile errors. i Google it and was able to handle the most missing library's but in the end it does not work.

you said: I didn't used the SDK but compiled within usual build system from scratch.
please tell me your steps. 
thank you

mhei, 2018-05-18:
No problem :smile:  Here are my steps:

```
git clone git://git.openwrt.org/openwrt/openwrt.git lede-v17.01.4.git
cd lede-v17.01.4.git/
git checkout -b v17.01.4 v17.01.4
scripts/feeds update
scripts/feeds install php7
cat <<EOF > .config
CONFIG_TARGET_mvebu=y
CONFIG_TARGET_mvebu_Default=y
CONFIG_TARGET_BOARD="mvebu"
CONFIG_DEVEL=y
# CONFIG_DRIVER_11AC_SUPPORT is not set
# CONFIG_DRIVER_11N_SUPPORT is not set
# CONFIG_DRIVER_11W_SUPPORT is not set
CONFIG_LIBCURL_COOKIES=y
CONFIG_LIBCURL_FILE=y
CONFIG_LIBCURL_FTP=y
CONFIG_LIBCURL_HTTP=y
CONFIG_LIBCURL_MBEDTLS=y
CONFIG_LIBCURL_NO_SMB="!"
CONFIG_LIBCURL_PROXY=y
# CONFIG_PACKAGE_base-files is not set
# CONFIG_PACKAGE_busybox is not set
# CONFIG_PACKAGE_dnsmasq is not set
# CONFIG_PACKAGE_dropbear is not set
# CONFIG_PACKAGE_e2fsprogs is not set
# CONFIG_PACKAGE_firewall is not set
# CONFIG_PACKAGE_fstools is not set
# CONFIG_PACKAGE_fwtool is not set
# CONFIG_PACKAGE_hostapd-common is not set
# CONFIG_PACKAGE_ip6tables is not set
# CONFIG_PACKAGE_iptables is not set
# CONFIG_PACKAGE_iw is not set
# CONFIG_PACKAGE_iwinfo is not set
# CONFIG_PACKAGE_jshn is not set
# CONFIG_PACKAGE_jsonfilter is not set
# CONFIG_PACKAGE_kmod-cfg80211 is not set
# CONFIG_PACKAGE_kmod-gpio-button-hotplug is not set
# CONFIG_PACKAGE_kmod-ip6tables is not set
# CONFIG_PACKAGE_kmod-ipt-conntrack is not set
# CONFIG_PACKAGE_kmod-ipt-core is not set
# CONFIG_PACKAGE_kmod-ipt-nat is not set
# CONFIG_PACKAGE_kmod-lib-crc-ccitt is not set
# CONFIG_PACKAGE_kmod-mac80211 is not set
# CONFIG_PACKAGE_kmod-mwlwifi is not set
# CONFIG_PACKAGE_kmod-nf-conntrack is not set
# CONFIG_PACKAGE_kmod-nf-conntrack6 is not set
# CONFIG_PACKAGE_kmod-nf-ipt is not set
# CONFIG_PACKAGE_kmod-nf-ipt6 is not set
# CONFIG_PACKAGE_kmod-nf-nat is not set
# CONFIG_PACKAGE_kmod-ppp is not set
# CONFIG_PACKAGE_lede-keyring is not set
# CONFIG_PACKAGE_libblobmsg-json is not set
CONFIG_PACKAGE_libcurl=y
# CONFIG_PACKAGE_libext2fs is not set
# CONFIG_PACKAGE_libf2fs is not set
# CONFIG_PACKAGE_libip4tc is not set
# CONFIG_PACKAGE_libip6tc is not set
# CONFIG_PACKAGE_libiwinfo is not set
# CONFIG_PACKAGE_libjson-c is not set
# CONFIG_PACKAGE_libjson-script is not set
CONFIG_PACKAGE_libmbedtls=y
# CONFIG_PACKAGE_libnl-tiny is not set
CONFIG_PACKAGE_libpcre=y
# CONFIG_PACKAGE_libpthread is not set
# CONFIG_PACKAGE_librt is not set
# CONFIG_PACKAGE_libubox is not set
# CONFIG_PACKAGE_libubus is not set
# CONFIG_PACKAGE_libuci is not set
# CONFIG_PACKAGE_libuclient is not set
# CONFIG_PACKAGE_libuuid is not set
# CONFIG_PACKAGE_libxtables is not set
# CONFIG_PACKAGE_logd is not set
# CONFIG_PACKAGE_mkf2fs is not set
# CONFIG_PACKAGE_mtd is not set
# CONFIG_PACKAGE_netifd is not set
# CONFIG_PACKAGE_odhcp6c is not set
# CONFIG_PACKAGE_odhcpd is not set
# CONFIG_PACKAGE_opkg is not set
CONFIG_PACKAGE_php7=y
CONFIG_PACKAGE_php7-cli=y
CONFIG_PACKAGE_php7-mod-curl=y
# CONFIG_PACKAGE_ppp is not set
# CONFIG_PACKAGE_procd is not set
# CONFIG_PACKAGE_procd-nand is not set
# CONFIG_PACKAGE_swconfig is not set
# CONFIG_PACKAGE_ubi-utils is not set
# CONFIG_PACKAGE_uboot-envtools is not set
# CONFIG_PACKAGE_ubox is not set
# CONFIG_PACKAGE_ubus is not set
# CONFIG_PACKAGE_ubusd is not set
# CONFIG_PACKAGE_uci is not set
# CONFIG_PACKAGE_uclient-fetch is not set
# CONFIG_PACKAGE_usign is not set
# CONFIG_PACKAGE_wpad-mini is not set
CONFIG_PACKAGE_zlib=y
CONFIG_PHP7_SYSTEMTZDATA=y
# CONFIG_TARGET_ROOTFS_SQUASHFS is not set
EOF

make -j8
```

B0F1B0, 2018-05-18:
thank you very much. at the moment i am not at home so i can't try it out.
when i think about it, there come some questions in my mind.
1) where do you get the .config from? 
2) where do you know what has to be "comment  -> # <-  " and "uncomment" in the .config file?
3) what about the "dependencies"

thanks for your help and time

mhei, 2018-05-21:
Not sure whether I understood your question right... I generated this config by using `make menuconfig`, then selecting/deselecting packages and then - to reduce the amount of lines here - I used `scripts/diffconfig.sh`. This way, I could provide your some shell script lines and it was not necessary to upload a "full" .config.

diizzyy, 2018-07-20:
@mhei 
Fixed?

B0F1B0, 2018-07-20:
yes. thanks for your help

diizzyy, 2018-07-20:
@B0F1B0 
Great, can you please close this issue then? :-)

B0F1B0, 2018-07-20:
done

closed 2018-07-20 by B0F1B0