openwrt/packages #6067: openssh: failed to run if 'Strip unnecessary functions from libraries' was selected in 'make menuconfig'

MiaoYuFei, 2018-05-17:
Maintainer: @\tripolar
Environment: 
OS: Ubuntu Server 18.04 LTS
Arch: x86_64
OpenWrt version: branch openwrt-18.06
OpenSSH version: 7.7p1
GCC version: 7.3.0
Binutils version: 2.29.1
C Library: musl
Description:
The package 'openssh' can be compiled successfully but failed to run if the option 'Strip unnecessary functions from libraries' was selected in 'make menuconfig'.
It seems that some functions are stripped by mistake. Here is the output for running 'sshd':

Error relocation /usr/sbin/sshd: __subvsi3: symbol not found
Error relocation /usr/sbin/sshd: __negvsi2: symbol not found
Error relocation /usr/sbin/sshd: __mulvdi3: symbol not found
Error relocation /usr/sbin/sshd: __addvsi3: symbol not found
Error relocation /usr/sbin/sshd: __mulvsi3: symbol not found
Error relocation /usr/sbin/sshd: __subvdi3: symbol not found
Error relocation /usr/sbin/sshd: __addvdi3: symbol not found

After the 'Strip unnecessary functions from libraries' was unselected, 'sshd' runs well.
I think a patch may be needed for this package to prevent the build system from stripping necessary functions.

diizzyy, 2018-05-18:
Don't enable it? I'm sure it'll break more packages.

MiaoYuFei, 2018-05-18:
@diizzyy Thanks for your reply. Do you mean it'll break more packages if the option 'Strip unnecessary functions from libraries' was disabled? And do you have any idea why this is happening, or are there any ways to fix this issue? Thanks.

NeoRaider, 2018-05-18:
mklibs ("Strip unnecessary functions from libraries") is known to be broken with musl libc.

In any case, this is not a bug in the openssh package, so I'm closing this issue.

closed 2018-05-18 by NeoRaider