openwrt/packages #6070: <mjpg-streamer - r182-10> Ignoring Empty Buffer

cjp160, 2018-05-18:
When running mjpg-streamer for a long time (24-48 hours), it will occasionally print out "Ignoring empty buffer...". If it does that four times without restarting the stream, the stream will hang.

The issue was brought up in this thread: https://github.com/jacksonliam/mjpg-streamer/issues/100 and fixed in that forked version of mjpg-streamer. Is there a way to fix it in the main OpenWRT package?

thess, 2018-05-18:
Thanks for the note - It's has been a while since I sync'd up with any dev work on mjpg-streamer. Perhaps I can get some time to switch over to the other repo (or at least co-opt some patches).

cjp160, 2018-05-21:
That would be awesome!

thess, 2018-05-22:
If you could give PR #6092 a test - let me know how you make out.
After I do a bit more local testing, I will merge this version into master in a day or 2.

cjp160, 2018-05-22:
Is there a way to install this using opkg on my device?

Collin

thess, 2018-05-23:
If you cannot build it yourself, you'll have to wait until this is merged and the buildbots produce a package for your architecture. Hmmm... you never mentioned what platform / release you are using it on.

thess, 2018-05-24:
PR #6092 merged - please test after buildbots catch up.

cjp160, 2018-05-24:
Will do. So this will no longer hang after the stream outputs “Ignoring empty buffer…” 4 times?

Collin

diizzyy, 2018-07-05:
Can we close this?

closed 2018-07-05 by thess