openwrt/packages #6079: dnsmasq: DHCP options 66, 150 are ignored

VersuchUndIrrtum, 2018-05-21:
dnsmasq:

DHCP options 66, 150 are ignored 

[LEDE Reboot 17.01.4 r3560-79f57e422d / LuCI lede-17.01 branch (git-17.290.79498-d3f0685)]

Steps to reproduce:

1. navigate to http://<IP>/cgi-bin/luci/admin/network/network/lan, DHCP Server, Advanced Settings, DHCP-Options
2. add `6,IP` or/and `150,IP`
3. boot a machine with PXE activated

diizzyy, 2018-07-05:
@VersuchUndIrrtum 
Hi, dnsmasq is included in the OpenWrt distribution not in the packages repo so please make a bug report over at http://bugs.openwrt.org instead.
Please close this one

@hnyman

closed 2018-07-05 by hnyman

LipkeGu, 2019-05-22:
@VersuchUndIrrtum  Option 150 is not (officialy) used and supported in PXE boot (PXE 2.1).
You can use also ip addresses in option 66. also when using option 66 for external tftp-server, you have to set NextServer-IP to 0.0.0.0 and clear filename and enter it in option 67.