openwrt/packages #6084: ddns-scripts: Cloudflare API v1 End of Life on June 4, 2018

phipolis, 2018-05-21:
Maintainer: @chris5560
Environment: N/A

Description:

I received the following message due to using ddns-scripts_cloudflare: 

> You are receiving this message because we determined your api key was used to make v1 Client API calls in the last 30 days. On June 4, 2018, the v1 API will be retired, meaning all calls to the API endpoint (https://www.cloudflare.com/api_json/) will be rejected from that point forward.
>
> In 2016, Cloudflare announced the [deprecation of our v1 API](https://blog.cloudflare.com/sunsetting-api-v1-in-favor-of-cloudflares-current-client-api-api-v4/) in favor of the v4 API. You can find instructions on migrating from v1 to v4 in our migration guide.

So, this sub-package will be nonfunctional in two weeks. Probably time to remove?

diizzyy, 2018-07-05:
@phipolis 
Can you provide a pull request?

phipolis, 2018-07-20:
How best to handle `sysupgrade` for users who've configured v1? To my mind the options are either to stop working until they notice and update their config, or perhaps v1 becomes an alias for v4, perhaps with a log message.

dibdot, 2018-08-08:
fixed in latest snapshot and 18.06 tree with referenced update commit.

closed 2018-08-08 by dibdot