openwrt/packages #6107: python3:lib2to3 failed to convert due to refactor's fixes not searched correctly

photof, 2018-05-24:
python3's lib2to3 would fail in silence if python3 and its packages are installed as compiled .pyc files.

Root cause is in Lib/lib2to3/refactor.py, the function get_all_fix_names only searches '.py' fix names.

=========below is workaround=========

--- a/Lib/lib2to3/refactor.py
+++ b/Lib/lib2to3/refactor.py
@@ -37,6 +37,12 @@
             if remove_prefix:
                 name = name[4:]
             fix_names.append(name[:-3])
+        if name.startswith("fix_") and name.endswith(".pyc"):
+            if remove_prefix:
+                name = name[4:]
+            name = name[:-4]
+            if name not in fix_names:
+                fix_names.append(name)
     return fix_names

jefferyto, 2019-01-08:
Would you be able to submit a pull request with this fix? That way you get the credit :wink:

cshoredaniel, 2019-01-09:
Hey noticed this in CI.

Author name (noj) need to be your real name 'firstname lastname'
Commit subject line MUST start with '<package name>: ' (Fix issue #6107 python3's lib2to3 would fail in silence if python3 and its packages are installed as compiled .pyc files. Root cause is, in Lib/lib2to3/refactor.py, the function get_all_fix_names only searches '.py' fix names.)

Also I'd recommend a shorter commit title (you can expand on it in the description).  See https://github.com/openwrt/packages/blob/master/CONTRIBUTING.md#commits-in-your-pull-requests-should and https://github.com/openwrt/packages/blob/master/CONTRIBUTING.md#advice-on-pull-requests

jefferyto, 2019-01-27:
I believe this can be closed :smile:

closed 2019-01-28 by kidome