openwrt/packages #6113: Shadowsocks-Libev: Not Working in SNAPSHOT and 18.06-SNAPSHOT

TingyiChen, 2018-05-24:
Maintainer: @yousong 
Environment: (put here arch, model, OpenWrt version)
ARCH: MIPS
Model: Xiaomi Mi Router 3G
OpenWRT Version: 2018-05-23 Snapshot or 18.06-SNAPSHOT built at commit a88ff7bf5b77fbde16c3c8d701170f111a5e984f

Description:
When configured shadowsocks server config in luci app,in LOCAL Instances menu no server valid to choose for any instance.

Log:
```
Thu May 24 14:58:24 2018 daemon.err sslocal: Missing parameter(s): -s -p -u -b -k -m
Thu May 24 14:58:24 2018 daemon.err sslocal: Usage: /usr/bin/sslocal [options] Options: 	-s,--server_addr server address 	-p,--server_port server port 	-u,--local_addr	 local Used address 	-b,--local_port	 local Binding port 	-k,--password	 your password 	-m,--method	 encryption algorithm(aes-*-cfb, bf-cfb, cast5-cfb, des-cfb, rc2-cfb, rc4, seed-cfb) 	-d,--daemon	 run as daemon 	-l,--log_level	 log level(0-7), default is LOG_NOTICE 	-h,--help	 print this help
Thu May 24 14:58:24 2018 daemon.info procd: Instance sslocal::instance1 s in a crash loop 6 crashes, 0 seconds since last crash
![_20180524231247](https://user-images.githubusercontent.com/14954514/40494730-2369ea82-5fa8-11e8-835f-66b9095bd781.png)

```

yousong, 2018-05-25:
@TingyiChen , any updates?

TingyiChen, 2018-05-25:
Seems can run now but i can't select backend server for ss-redir in luci config...

TingyiChen, 2018-05-26:
[image]

TingyiChen, 2018-06-01:
Confirmed works now.Thanks

yousong, 2018-07-15:
Please open a new issue.  With additional issue posted above.