openwrt/packages #6142: ddns-scripts: add support for desec.io

doe1097, 2018-05-28:
Hello @vortigont,

Could you please add support for desec.io. It's a free german DynDNS Provider. I'm currently using it as a custom Provider, but it doesn't seem to work properly (forced update wipes the IP). Maybe this can be resolved with a proper template.

Here is the documentation: https://desec.io/#!/en/docs/update-api-details

Best,
doe1097

vortigont, 2018-05-29:
I'll try to check it. Pls, tell more about the issues you are facing with this service.

doe1097, 2018-05-29:
Seems like it was not a problem with the service itself. The ddns-script wasn't able to read the modems public IP, when I selected 'interface' for IP address source [IPv4] and ppoe-wan. I could resolve this by choosing 'URL' instead of interface and now it works fine even though the update method is not recommended by the service:
[quote]Set username and password query string parameters [in the URL](GET ?username=...&password=...). We strongly discourage using this method, but provide it as an emergency solution for situations where folks need to deal with old and/or crappy clients.[/code]

chris5560, 2018-06-02:
desec.io is supported by current version of ddns-scripts.

The entry inside /etc/ddns/services is:
"desec.io" "http://update.dedyn.io/?username=[USERNAME]&password=[PASSWORD]&hostname=[DOMAIN]&myipv4=[IP]" "good|nochg"
and inside /etc/ddns/services_ipv6:
"desec.io" "http://update.dedyn.io/?username=[USERNAME]&password=[PASSWORD]&hostname=[DOMAIN]&myipv6=[IP]" "good|nochg"

to remove the error-message about GET ?username=...&password=... please try to use the following lines:
"desec.io" "http://[USERNAME]:[PASSWORD]@update.dedyn.io/?hostname=[DOMAIN]&myipv4=[IP]" "good|nochg"
and
"desec.io" "http://[USERNAME]:[PASSWORD]@update.dedyn.io/?hostname=[DOMAIN]&myipv6=[IP]" "good|nochg"

Thanks for your support

doe1097, 2018-06-03:
Hello @chris5560,
I just tried the new configuration, but the Logs throw me an error:

224338  WARN : Transfer failed - retry 1/0 in 60 seconds
 224438       : #> /usr/bin/wget-ssl -nv -t 1 -O /var/run/ddns/grnt_dedyn_io.dat -o /var/run/ddns/grnt_dedyn_io.err --no-proxy 'http://MYUSER:MYPASSWD@update.dedyn.io/?hostname=MYDDNS&myipv4=MYIP'
 224438 ERROR : GNU Wget Error: '6'
 224438       : Unknown authentication scheme.

closed 2018-06-03 by doe1097