openwrt/packages #6148: pptp client

m0rDev, 2018-05-29:
I'm sorry if I write wrong.

LEDE Reboot 17.01.4 r3560-79f57e422d

kmod-pptp | 4.4.92-1
ppp-mod-pptp | 2.4.7-12

If you set the parameter **ipparam** in file /etc/ppp/options.pptp 

If you set the parameter ** ipparam ** in file /etc/ppp/options.pptp
then in Web lucy -> network -> Interfaces - the status **pptp** of the connection is not displayed

RX: 0 B (0 Pkts.)
TX: 0 B (0 Pkts.)

This connection exists. It can be seen through the terminal - **ifconfig**

pptp-usd  Link encap:Point-to-Point Protocol
          inet addr:192.168.100.52  P-t-P:192.168.100.240  Mask:255.255.255.255
          UP POINTOPOINT RUNNING NOARP MULTICAST  MTU:1500  Metric:1
          RX packets:3073 errors:0 dropped:0 overruns:0 frame:0
          TX packets:5025 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:3
          RX bytes:314513 (307.1 KiB)  TX bytes:525333 (513.0 KiB)

If you remove or comment out the parameter  ** ipparam ** in file /etc/ppp/options.pptp. Then in the Web Lucy -> network -> Interfaces **pptp** status is working

Uptime: 0h 0m 13s
MAC-Address: 00:00:00:00:00:00
RX: 64 B (4 Pkts.)
TX: 96 B (5 Pkts.)
IPv4: 192.168.100.58/32

dibdot, 2018-06-04:
Already answered here: https://github.com/openwrt/luci/issues/1852

closed 2018-06-04 by dibdot

m0rDev, 2018-06-04:
You did not answer anything