openwrt/packages #6175: <adblock> blocklist with overall 0 domains loaded successfully

icf20, 2018-06-04:
LEDE Reboot 17.01.4 r3560-79f57e422d
adblock - 3.4.3-1

for some reason is not loading any domain to block
```

root@OpenWrt:~# wget https://adaway.org/hosts.txt
Downloading 'https://adaway.org/hosts.txt'
Connecting to 104.24.104.89:443
Writing to 'hosts.txt'

Download completed (13722 bytes)
```
wget is working correctly 

any ideas ?

dibdot, 2018-06-04:
Please enable debug in adblock, make a re-run and finally provide the output of `logread -e "adblock"`.

icf20, 2018-06-04:
```
Mon Jun  4 09:58:34 2018 user.debug adblock-[3.4.3]: f_main ::: url: https://adaway.org/hosts.txt, rc: 1, src_log: /bin/wget: unrecognized option: no-cache
Usage: /bin/wget [options] <URL>
Options:
	-4				Use IPv4 only
	-6				Use IPv6 only
	-q				Turn off status messages
	-O <file>			Redirect output to file (use "-" for stdout)
	-P <dir>			Set directory for output files
	--user=<user>			HTTP authentication username
	--password=<password>		HTTP authentication password
	--user-agent|-U <str>		Set HTTP user aMon Jun  4 09:58:34 2018 user.debug adblock-[3.4.3]: f_list ::: mode: remove, cnt: 0, in_rc: 1, out_rc: 0
```

closed 2018-06-04 by icf20