openwrt/packages #6184: <adblock> WindowsSpyBlocker url have changed

icf20, 2018-06-04:
old url https://raw.githubusercontent.com/crazy-max/WindowsSpyBlocker/master/data/hosts/win10/spy.txt  is 404

new url ? https://raw.githubusercontent.com/crazy-max/WindowsSpyBlocker/master/data/hosts/spy.txt

dibdot, 2018-06-04:
Good catch, will be fixed with #6186

closed 2018-06-04 by dibdot

tehfink, 2019-06-09:
@dibdot Just installed adblock, and seeing this same issue:
```bash
# uname -a
Linux AP-01 4.14.95 #0 Mon Jan 28 08:54:32 2019 armv5tel GNU/Linux
# cat /etc/openwrt_version 
r7676-cddd7b4c77
root@AP-01:/opt# cat /etc/openwrt_release 
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='18.06.2'
DISTRIB_REVISION='r7676-cddd7b4c77'
DISTRIB_TARGET='kirkwood/generic'
DISTRIB_ARCH='arm_xscale'
DISTRIB_DESCRIPTION='OpenWrt 18.06.2 r7676-cddd7b4c77'
DISTRIB_TAINTS=''
# opkg list-installed | grep adblock
adblock - 3.5.5-3
luci-app-adblock - git-19.127.24963-84692e5-1
# logread -e "adblock"
…
Sun Jun  9 09:33:47 2019 user.debug adblock-3.5.5-2[3935]: f_main ::: name: winspy, url: https://raw.githubusercontent.com/crazy-max/WindowsSpyBlocker/master/data/hosts/win10/spy.txt, rc: 8, log: Downloading 'https://raw.githubusercontent.com/crazy-max/WindowsSpyBlocker/master/data/hosts/win10/spy.txt' Connecting to 151.101.244.133:443 HTTP error 404
```

dibdot, 2019-06-09:
Just use the "--force-maintainer" opkg option during package installation to overwrite your old config with the default package config (with the correct URL).