openwrt/packages #6188: <adblock> blocklist with overall 375 domains loaded successfully

icf20, 2018-06-04:
`user.info adblock-[3.4.3]: blocklist with overall 375 domains loaded successfully`

no matter how many lists i enable i only get that number, why ?

dibdot, 2018-06-04:
See my answer to #6175 ... provide valuable logs, please

icf20, 2018-06-04:
this issue is unrelated to that one, this time there is downloading 

```
root@OpenWrt:~# logread -e "adblock"
Mon Jun  4 19:17:50 2018 user.debug adblock-[3.4.3]: f_uci  ::: config: dhcp, change: 
Mon Jun  4 19:17:50 2018 user.debug adblock-[3.4.3]: f_uci  ::: config: firewall, change: 
Mon Jun  4 19:17:50 2018 user.debug adblock-[3.4.3]: f_jsnup::: status: running, mode: normal mode, cnt: 0, notify: 0, notify_cnt: 0, notify_pid: -
Mon Jun  4 19:17:50 2018 user.info adblock-[3.4.3]: start adblock processing (start)
Mon Jun  4 19:17:50 2018 user.debug adblock-[3.4.3]: f_main ::: dns: dnsmasq, fetch_util: /bin/uclient-fetch (libustream-ssl), backup: 0, backup_mode: 0, whitelist_mode: 0, force_srt: 1, force_dns: 1, mem_total: 28
Mon Jun  4 19:17:50 2018 user.debug adblock-[3.4.3]: f_main ::: name: adaway, enabled: 1
Mon Jun  4 19:17:51 2018 user.debug adblock-[3.4.3]: f_list ::: mode: merge, cnt: 409, in_rc: 0, out_rc: 0
Mon Jun  4 19:17:51 2018 user.debug adblock-[3.4.3]: f_main ::: name: adguard, enabled: 1
Mon Jun  4 19:17:56 2018 user.debug adblock-[3.4.3]: f_list ::: mode: merge, cnt: 19626, in_rc: 0, out_rc: 0
Mon Jun  4 19:17:56 2018 user.debug adblock-[3.4.3]: f_main ::: name: bitcoin, enabled: 1
Mon Jun  4 19:17:57 2018 user.debug adblock-[3.4.3]: f_list ::: mode: merge, cnt: 282, in_rc: 0, out_rc: 0
Mon Jun  4 19:17:57 2018 user.debug adblock-[3.4.3]: f_main ::: name: blacklist, enabled: 1
Mon Jun  4 19:17:57 2018 user.debug adblock-[3.4.3]: f_list ::: mode: remove, cnt: 0, in_rc: 1, out_rc: 0
Mon Jun  4 19:17:57 2018 user.debug adblock-[3.4.3]: f_main ::: name: disconnect, enabled: 1
Mon Jun  4 19:25:04 2018 user.debug adblock-[3.4.3]: f_list ::: mode: merge, cnt: 3797, in_rc: 0, out_rc: 0
Mon Jun  4 19:25:04 2018 user.debug adblock-[3.4.3]: f_main ::: name: dshield, enabled: 0
Mon Jun  4 19:25:04 2018 user.debug adblock-[3.4.3]: f_list ::: mode: remove, cnt: 0, in_rc: 1, out_rc: 0
Mon Jun  4 19:25:04 2018 user.debug adblock-[3.4.3]: f_main ::: name: feodo, enabled: 1
Mon Jun  4 19:25:04 2018 user.debug adblock-[3.4.3]: f_list ::: mode: remove, cnt: 0, in_rc: 0, out_rc: 0
Mon Jun  4 19:25:04 2018 user.debug adblock-[3.4.3]: f_main ::: name: hphosts, enabled: 0
Mon Jun  4 19:25:04 2018 user.debug adblock-[3.4.3]: f_list ::: mode: remove, cnt: 0, in_rc: 1, out_rc: 0
Mon Jun  4 19:25:04 2018 user.debug adblock-[3.4.3]: f_main ::: name: malware, enabled: 1
Mon Jun  4 19:25:10 2018 user.debug adblock-[3.4.3]: f_list ::: mode: merge, cnt: 22542, in_rc: 0, out_rc: 0
Mon Jun  4 19:25:10 2018 user.debug adblock-[3.4.3]: f_main ::: name: malwarelist, enabled: 1
Mon Jun  4 19:25:11 2018 user.debug adblock-[3.4.3]: f_list ::: mode: merge, cnt: 1131, in_rc: 0, out_rc: 0
Mon Jun  4 19:25:11 2018 user.debug adblock-[3.4.3]: f_main ::: name: openphish, enabled: 1
Mon Jun  4 19:25:12 2018 user.debug adblock-[3.4.3]: f_list ::: mode: merge, cnt: 2760, in_rc: 0, out_rc: 0
Mon Jun  4 19:25:12 2018 user.debug adblock-[3.4.3]: f_main ::: name: ransomware, enabled: 1
Mon Jun  4 19:25:13 2018 user.debug adblock-[3.4.3]: f_list ::: mode: merge, cnt: 1899, in_rc: 0, out_rc: 0
Mon Jun  4 19:25:13 2018 user.debug adblock-[3.4.3]: f_main ::: name: reg_cn, enabled: 0
Mon Jun  4 19:25:13 2018 user.debug adblock-[3.4.3]: f_list ::: mode: remove, cnt: 0, in_rc: 1, out_rc: 0
Mon Jun  4 19:25:13 2018 user.debug adblock-[3.4.3]: f_main ::: name: reg_de, enabled: 0
Mon Jun  4 19:25:13 2018 user.debug adblock-[3.4.3]: f_list ::: mode: remove, cnt: 0, in_rc: 1, out_rc: 0
Mon Jun  4 19:25:13 2018 user.debug adblock-[3.4.3]: f_main ::: name: reg_id, enabled: 0
Mon Jun  4 19:25:13 2018 user.debug adblock-[3.4.3]: f_list ::: mode: remove, cnt: 0, in_rc: 1, out_rc: 0
Mon Jun  4 19:25:13 2018 user.debug adblock-[3.4.3]: f_main ::: name: reg_nl, enabled: 0
Mon Jun  4 19:25:13 2018 user.debug adblock-[3.4.3]: f_list ::: mode: remove, cnt: 0, in_rc: 1, out_rc: 0
Mon Jun  4 19:25:13 2018 user.debug adblock-[3.4.3]: f_main ::: name: reg_pl, enabled: 0
Mon Jun  4 19:25:13 2018 user.debug adblock-[3.4.3]: f_list ::: mode: remove, cnt: 0, in_rc: 1, out_rc: 0
Mon Jun  4 19:25:13 2018 user.debug adblock-[3.4.3]: f_main ::: name: reg_ro, enabled: 0
Mon Jun  4 19:25:13 2018 user.debug adblock-[3.4.3]: f_list ::: mode: remove, cnt: 0, in_rc: 1, out_rc: 0
Mon Jun  4 19:25:13 2018 user.debug adblock-[3.4.3]: f_main ::: name: reg_ru, enabled: 0
Mon Jun  4 19:25:13 2018 user.debug adblock-[3.4.3]: f_list ::: mode: remove, cnt: 0, in_rc: 1, out_rc: 0
Mon Jun  4 19:25:13 2018 user.debug adblock-[3.4.3]: f_main ::: name: shalla, enabled: 0
Mon Jun  4 19:25:13 2018 user.debug adblock-[3.4.3]: f_list ::: mode: remove, cnt: 0, in_rc: 1, out_rc: 0
Mon Jun  4 19:25:13 2018 user.debug adblock-[3.4.3]: f_main ::: name: spam404, enabled: 0
Mon Jun  4 19:25:13 2018 user.debug adblock-[3.4.3]: f_list ::: mode: remove, cnt: 0, in_rc: 1, out_rc: 0
Mon Jun  4 19:25:13 2018 user.debug adblock-[3.4.3]: f_main ::: name: sysctl, enabled: 0
Mon Jun  4 19:25:13 2018 user.debug adblock-[3.4.3]: f_list ::: mode: remove, cnt: 0, in_rc: 1, out_rc: 0
Mon Jun  4 19:25:13 2018 user.debug adblock-[3.4.3]: f_main ::: name: whocares, enabled: 0
Mon Jun  4 19:25:13 2018 user.debug adblock-[3.4.3]: f_list ::: mode: remove, cnt: 0, in_rc: 1, out_rc: 0
Mon Jun  4 19:25:13 2018 user.debug adblock-[3.4.3]: f_main ::: name: winspy, enabled: 0
Mon Jun  4 19:25:13 2018 user.debug adblock-[3.4.3]: f_list ::: mode: remove, cnt: 0, in_rc: 1, out_rc: 0
[... cut ...]
Mon Jun  4 19:25:15 2018 user.debug adblock-[3.4.3]: f_list ::: mode: merge, cnt: 2634, in_rc: 0, out_rc: 0
Mon Jun  4 19:25:15 2018 user.debug adblock-[3.4.3]: f_main ::: name: zeus, enabled: 1
Mon Jun  4 19:25:15 2018 user.debug adblock-[3.4.3]: f_list ::: mode: merge, cnt: 375, in_rc: 0, out_rc: 0
Mon Jun  4 19:25:15 2018 user.debug adblock-[3.4.3]: f_hash ::: hash_util: /usr/bin/sha256sum, hash: 01ba4719c80b6fe911b091a7c05124b64eeece964e09c058ef8f9805daca546b, out_rc: 1
root@OpenWrt:~# 

```

dibdot, 2018-06-04:
That's an OOM exception. You have only a 32 MB router, added many list and enabled Force Overall Sort. Disable all additional lists and the sort option and start from there. But no promises, 32MB devices are more or less "unsupported" - at least from adblock perspective.

closed 2018-06-04 by dibdot

icf20, 2018-06-04:
`Force Overall Sort` is enabled 

on the other hand adblocker.sh handle 15k list with no problem