openwrt/packages #6208: adblock: Archer C5 128MB RAM. Blocklist with overall 0 domains loaded successfully.

smartfonreddit, 2018-06-07:
I followed the instructions but it still loads 0 domains. Tried only 1 list and as many as 5 lists. The Archer C5 1200 v1 device has plenty of free RAM. Did a reboot as well.

```
Thu Jun  7 12:33:25 2018 user.debug adblock-[3.4.3]: f_uci  ::: config: dhcp, change: 
Thu Jun  7 12:33:25 2018 user.debug adblock-[3.4.3]: f_uci  ::: config: firewall, change: 
Thu Jun  7 12:33:25 2018 user.debug adblock-[3.4.3]: f_jsnup::: status: running, mode: normal mode, cnt: 0, notify: 0, notify_cnt: 0, notify_pid: -
Thu Jun  7 12:33:25 2018 user.info adblock-[3.4.3]: start adblock processing (reload)
Thu Jun  7 12:33:25 2018 user.debug adblock-[3.4.3]: f_main ::: dns: dnsmasq, fetch_util: /bin/uclient-fetch (libustream-ssl), backup: 0, backup_mode: 0, whitelist_mode: 0, force_srt: 0, force_dns: 0, mem_total: 125
Thu Jun  7 12:33:25 2018 user.debug adblock-[3.4.3]: f_main ::: name: adaway, enabled: 1
Thu Jun  7 12:33:25 2018 user.debug adblock-[3.4.3]: f_list ::: mode: merge, cnt: 409, in_rc: 0, out_rc: 0
Thu Jun  7 12:33:25 2018 user.debug adblock-[3.4.3]: f_main ::: name: adguard, enabled: 1
Thu Jun  7 12:33:28 2018 user.debug adblock-[3.4.3]: f_list ::: mode: merge, cnt: 19676, in_rc: 0, out_rc: 0
Thu Jun  7 12:33:28 2018 user.debug adblock-[3.4.3]: f_main ::: name: bitcoin, enabled: 0
Thu Jun  7 12:33:28 2018 user.debug adblock-[3.4.3]: f_list ::: mode: remove, cnt: 0, in_rc: 1, out_rc: 0
Thu Jun  7 12:33:28 2018 user.debug adblock-[3.4.3]: f_main ::: name: blacklist, enabled: 1
Thu Jun  7 12:33:28 2018 user.debug adblock-[3.4.3]: f_list ::: mode: remove, cnt: 0, in_rc: 1, out_rc: 0
Thu Jun  7 12:33:28 2018 user.debug adblock-[3.4.3]: f_main ::: name: disconnect, enabled: 0
Thu Jun  7 12:33:28 2018 user.debug adblock-[3.4.3]: f_list ::: mode: remove, cnt: 0, in_rc: 1, out_rc: 0
Thu Jun  7 12:33:28 2018 user.debug adblock-[3.4.3]: f_main ::: name: dshield, enabled: 0
Thu Jun  7 12:33:28 2018 user.debug adblock-[3.4.3]: f_list ::: mode: remove, cnt: 0, in_rc: 1, out_rc: 0
Thu Jun  7 12:33:28 2018 user.debug adblock-[3.4.3]: f_main ::: name: feodo, enabled: 0
Thu Jun  7 12:33:28 2018 user.debug adblock-[3.4.3]: f_list ::: mode: remove, cnt: 0, in_rc: 1, out_rc: 0
Thu Jun  7 12:33:28 2018 user.debug adblock-[3.4.3]: f_main ::: name: hphosts, enabled: 0
Thu Jun  7 12:33:28 2018 user.debug adblock-[3.4.3]: f_list ::: mode: remove, cnt: 0, in_rc: 1, out_rc: 0
Thu Jun  7 12:33:28 2018 user.debug adblock-[3.4.3]: f_main ::: name: malware, enabled: 0
Thu Jun  7 12:33:28 2018 user.debug adblock-[3.4.3]: f_list ::: mode: remove, cnt: 0, in_rc: 1, out_rc: 0
Thu Jun  7 12:33:28 2018 user.debug adblock-[3.4.3]: f_main ::: name: malwarelist, enabled: 0
Thu Jun  7 12:33:28 2018 user.debug adblock-[3.4.3]: f_list ::: mode: remove, cnt: 0, in_rc: 1, out_rc: 0
Thu Jun  7 12:33:28 2018 user.debug adblock-[3.4.3]: f_main ::: name: openphish, enabled: 0
Thu Jun  7 12:33:28 2018 user.debug adblock-[3.4.3]: f_list ::: mode: remove, cnt: 0, in_rc: 1, out_rc: 0
Thu Jun  7 12:33:28 2018 user.debug adblock-[3.4.3]: f_main ::: name: ransomware, enabled: 0
Thu Jun  7 12:33:28 2018 user.debug adblock-[3.4.3]: f_list ::: mode: remove, cnt: 0, in_rc: 1, out_rc: 0
Thu Jun  7 12:33:28 2018 user.debug adblock-[3.4.3]: f_main ::: name: reg_cn, enabled: 0
Thu Jun  7 12:33:28 2018 user.debug adblock-[3.4.3]: f_list ::: mode: remove, cnt: 0, in_rc: 1, out_rc: 0
Thu Jun  7 12:33:28 2018 user.debug adblock-[3.4.3]: f_main ::: name: reg_de, enabled: 0
Thu Jun  7 12:33:28 2018 user.debug adblock-[3.4.3]: f_list ::: mode: remove, cnt: 0, in_rc: 1, out_rc: 0
Thu Jun  7 12:33:28 2018 user.debug adblock-[3.4.3]: f_main ::: name: reg_id, enabled: 0
Thu Jun  7 12:33:28 2018 user.debug adblock-[3.4.3]: f_list ::: mode: remove, cnt: 0, in_rc: 1, out_rc: 0
Thu Jun  7 12:33:28 2018 user.debug adblock-[3.4.3]: f_main ::: name: reg_nl, enabled: 0
Thu Jun  7 12:33:28 2018 user.debug adblock-[3.4.3]: f_list ::: mode: remove, cnt: 0, in_rc: 1, out_rc: 0
Thu Jun  7 12:33:28 2018 user.debug adblock-[3.4.3]: f_main ::: name: reg_pl, enabled: 0
Thu Jun  7 12:33:28 2018 user.debug adblock-[3.4.3]: f_list ::: mode: remove, cnt: 0, in_rc: 1, out_rc: 0
Thu Jun  7 12:33:28 2018 user.debug adblock-[3.4.3]: f_main ::: name: reg_ro, enabled: 0
Thu Jun  7 12:33:28 2018 user.debug adblock-[3.4.3]: f_list ::: mode: remove, cnt: 0, in_rc: 1, out_rc: 0
Thu Jun  7 12:33:28 2018 user.debug adblock-[3.4.3]: f_main ::: name: reg_ru, enabled: 0
Thu Jun  7 12:33:28 2018 user.debug adblock-[3.4.3]: f_list ::: mode: remove, cnt: 0, in_rc: 1, out_rc: 0
Thu Jun  7 12:33:28 2018 user.debug adblock-[3.4.3]: f_main ::: name: shalla, enabled: 0
Thu Jun  7 12:33:28 2018 user.debug adblock-[3.4.3]: f_list ::: mode: remove, cnt: 0, in_rc: 1, out_rc: 0
Thu Jun  7 12:33:28 2018 user.debug adblock-[3.4.3]: f_main ::: name: spam404, enabled: 0
Thu Jun  7 12:33:28 2018 user.debug adblock-[3.4.3]: f_list ::: mode: remove, cnt: 0, in_rc: 1, out_rc: 0
Thu Jun  7 12:33:28 2018 user.debug adblock-[3.4.3]: f_main ::: name: sysctl, enabled: 0
Thu Jun  7 12:33:28 2018 user.debug adblock-[3.4.3]: f_list ::: mode: remove, cnt: 0, in_rc: 1, out_rc: 0
Thu Jun  7 12:33:28 2018 user.debug adblock-[3.4.3]: f_main ::: name: whocares, enabled: 0
Thu Jun  7 12:33:28 2018 user.debug adblock-[3.4.3]: f_list ::: mode: remove, cnt: 0, in_rc: 1, out_rc: 0
Thu Jun  7 12:33:28 2018 user.debug adblock-[3.4.3]: f_main ::: name: winspy, enabled: 0
Thu Jun  7 12:33:28 2018 user.debug adblock-[3.4.3]: f_list ::: mode: remove, cnt: 0, in_rc: 1, out_rc: 0
Thu Jun  7 12:33:28 2018 user.debug adblock-[3.4.3]: f_main ::: name: winhelp, enabled: 0
Thu Jun  7 12:33:28 2018 user.debug adblock-[3.4.3]: f_list ::: mode: remove, cnt: 0, in_rc: 1, out_rc: 0
Thu Jun  7 12:33:28 2018 user.debug adblock-[3.4.3]: f_main ::: name: yoyo, enabled: 1
Thu Jun  7 12:33:30 2018 user.debug adblock-[3.4.3]: f_list ::: mode: merge, cnt: 2634, in_rc: 0, out_rc: 0
Thu Jun  7 12:33:30 2018 user.debug adblock-[3.4.3]: f_main ::: name: zeus, enabled: 0
Thu Jun  7 12:33:30 2018 user.debug adblock-[3.4.3]: f_list ::: mode: remove, cnt: 0, in_rc: 1, out_rc: 0
Thu Jun  7 12:33:30 2018 user.debug adblock-[3.4.3]: f_hash ::: hash_util: /usr/bin/sha256sum, hash: 01ba4719c80b6fe911b091a7c05124b64eeece964e09c058ef8f9805daca546b, out_rc: 1
Thu Jun  7 12:33:32 2018 user.debug adblock-[3.4.3]: f_tld  ::: source: /tmp/tmp.kPkkMe/adb_list.overall, cnt: 22719, cnt_srt: 0, cnt_tld: 0
Thu Jun  7 12:33:32 2018 user.debug adblock-[3.4.3]: f_list ::: mode: final, cnt: 0, in_rc: 0, out_rc: 0
Thu Jun  7 12:33:32 2018 user.debug adblock-[3.4.3]: f_hash ::: hash_util: /usr/bin/sha256sum, hash: e3b0c44298fc1c149afbf4c8996fb92427ae41e4649b934ca495991b7852b855, out_rc: 1
Thu Jun  7 12:33:32 2018 user.debug adblock-[3.4.3]: f_dnsup::: cache_util: -, cache_rc: 0, cache_flush: 0, cache_cnt: 0, out_rc: 0
Thu Jun  7 12:33:32 2018 user.debug adblock-[3.4.3]: f_jsnup::: status: enabled, mode: normal mode, cnt: 0, notify: 0, notify_cnt: 0, notify_pid: -
Thu Jun  7 12:33:32 2018 user.info adblock-[3.4.3]: blocklist with overall 0 domains loaded successfully (TP-Link Archer C5 v1, LEDE Reboot 17.01.4 r3560-79f57e422d)
```

dibdot, 2018-06-07:
Please retry with latest adlock snapshot 3.5.2, see here: https://downloads.openwrt.org/snapshots/packages/x86_64/packages/adblock_3.5.2-2_all.ipk

smartfonreddit, 2018-06-07:
Since I'm not experiences in this, could you please provide the steps on how to update from that URL? The UI installer lists 3.4 as the latest. 
Should I first remove the existing adblock before updating?

dibdot, 2018-06-07:
ssh to your router and do this:
```
1. cd /tmp 
2. uclient-fetch https://downloads.openwrt.org/snapshots/packages/x86_64/packages/adblock_3.5.2-2_all.ipk
3. opkg install adblock_3.5.2-2_all.ipk --force-maintainer

```

smartfonreddit, 2018-06-07:
Got it working after the third attempt. Not sure why the adblock entry wasn't appearing on list at first. The Suspend and Resume button also works. Earlier it was always on the Suspend button (as if the service was always enabled). At the moment there are 80,000 entries. 

What does not work, however, if the ad blocking itself. The ads go through on all websites. I added entries to the Blacklist and they weren't blocked. Rebooted the router. Now I can't remove the entries from the blacklist. After pressing "Save" the page refreshes and the entries are still there. 

```
Thu Jun  7 14:03:02 2018 user.debug adblock-[3.5.2]: f_uci  ::: config: dhcp, change: 
Thu Jun  7 14:03:02 2018 user.debug adblock-[3.5.2]: f_uci  ::: config: firewall, change: 
Thu Jun  7 14:03:02 2018 user.debug adblock-[3.5.2]: f_jsnup::: status: running, mode: normal mode, cnt: 0, notify: 0, notify_cnt: 0, notify_pid: -
Thu Jun  7 14:03:02 2018 user.info adblock-[3.5.2]: start adblock processing (start)
Thu Jun  7 14:03:02 2018 user.debug adblock-[3.5.2]: f_main ::: dns: dnsmasq, fetch_util: /bin/uclient-fetch (libustream-ssl), backup: 0, backup_mode: 0, dns_jail: 0, force_srt: 0, force_dns: 0, mem_total: 125, mem_free: 93, max_queue: 4
Thu Jun  7 14:03:02 2018 user.debug adblock-[3.5.2]: f_main ::: name: adaway, enabled: 1
Thu Jun  7 14:03:02 2018 user.debug adblock-[3.5.2]: f_main ::: name: adguard, enabled: 1
Thu Jun  7 14:03:02 2018 user.debug adblock-[3.5.2]: f_main ::: name: bitcoin, enabled: 0
Thu Jun  7 14:03:02 2018 user.debug adblock-[3.5.2]: f_list ::: name: bitcoin, mode: remove, cnt: 0, in_rc: 0, out_rc: 0
Thu Jun  7 14:03:02 2018 user.debug adblock-[3.5.2]: f_main ::: name: blacklist, enabled: 1
Thu Jun  7 14:03:02 2018 user.debug adblock-[3.5.2]: f_main ::: name: disconnect, enabled: 1
Thu Jun  7 14:03:02 2018 user.debug adblock-[3.5.2]: f_list ::: name: blacklist, mode: download, cnt: 4, in_rc: 0, out_rc: 0
Thu Jun  7 14:03:03 2018 user.debug adblock-[3.5.2]: f_list ::: name: adaway, mode: download, cnt: 409, in_rc: 0, out_rc: 0
Thu Jun  7 14:03:04 2018 user.debug adblock-[3.5.2]: f_list ::: name: disconnect, mode: download, cnt: 3788, in_rc: 0, out_rc: 0
Thu Jun  7 14:04:32 2018 user.debug adblock-[3.5.2]: f_list ::: name: adguard, mode: download, cnt: 19676, in_rc: 0, out_rc: 0
Thu Jun  7 14:04:32 2018 user.debug adblock-[3.5.2]: f_main ::: name: dshield, enabled: 0
Thu Jun  7 14:04:32 2018 user.debug adblock-[3.5.2]: f_list ::: name: dshield, mode: remove, cnt: 0, in_rc: 0, out_rc: 0
Thu Jun  7 14:04:32 2018 user.debug adblock-[3.5.2]: f_main ::: name: feodo, enabled: 0
Thu Jun  7 14:04:32 2018 user.debug adblock-[3.5.2]: f_list ::: name: feodo, mode: remove, cnt: 0, in_rc: 0, out_rc: 0
Thu Jun  7 14:04:32 2018 user.debug adblock-[3.5.2]: f_main ::: name: hphosts, enabled: 1
Thu Jun  7 14:04:32 2018 user.debug adblock-[3.5.2]: f_main ::: name: malware, enabled: 0
Thu Jun  7 14:04:32 2018 user.debug adblock-[3.5.2]: f_list ::: name: malware, mode: remove, cnt: 0, in_rc: 0, out_rc: 0
Thu Jun  7 14:04:32 2018 user.debug adblock-[3.5.2]: f_main ::: name: malwarelist, enabled: 0
Thu Jun  7 14:04:32 2018 user.debug adblock-[3.5.2]: f_list ::: name: malwarelist, mode: remove, cnt: 0, in_rc: 0, out_rc: 0
Thu Jun  7 14:04:32 2018 user.debug adblock-[3.5.2]: f_main ::: name: openphish, enabled: 0
Thu Jun  7 14:04:32 2018 user.debug adblock-[3.5.2]: f_list ::: name: openphish, mode: remove, cnt: 0, in_rc: 0, out_rc: 0
Thu Jun  7 14:04:32 2018 user.debug adblock-[3.5.2]: f_main ::: name: ransomware, enabled: 1
Thu Jun  7 14:04:32 2018 user.debug adblock-[3.5.2]: f_main ::: name: reg_cn, enabled: 0
Thu Jun  7 14:04:32 2018 user.debug adblock-[3.5.2]: f_list ::: name: reg_cn, mode: remove, cnt: 0, in_rc: 0, out_rc: 0
Thu Jun  7 14:04:32 2018 user.debug adblock-[3.5.2]: f_main ::: name: reg_cz, enabled: 0
Thu Jun  7 14:04:32 2018 user.debug adblock-[3.5.2]: f_list ::: name: reg_cz, mode: remove, cnt: 0, in_rc: 0, out_rc: 0
Thu Jun  7 14:04:32 2018 user.debug adblock-[3.5.2]: f_main ::: name: reg_de, enabled: 0
Thu Jun  7 14:04:32 2018 user.debug adblock-[3.5.2]: f_list ::: name: reg_de, mode: remove, cnt: 0, in_rc: 0, out_rc: 0
Thu Jun  7 14:04:32 2018 user.debug adblock-[3.5.2]: f_main ::: name: reg_id, enabled: 0
Thu Jun  7 14:04:32 2018 user.debug adblock-[3.5.2]: f_list ::: name: reg_id, mode: remove, cnt: 0, in_rc: 0, out_rc: 0
Thu Jun  7 14:04:32 2018 user.debug adblock-[3.5.2]: f_main ::: name: reg_nl, enabled: 0
Thu Jun  7 14:04:32 2018 user.debug adblock-[3.5.2]: f_list ::: name: reg_nl, mode: remove, cnt: 0, in_rc: 0, out_rc: 0
Thu Jun  7 14:04:32 2018 user.debug adblock-[3.5.2]: f_main ::: name: reg_pl, enabled: 0
Thu Jun  7 14:04:32 2018 user.debug adblock-[3.5.2]: f_list ::: name: reg_pl, mode: remove, cnt: 0, in_rc: 0, out_rc: 0
Thu Jun  7 14:04:32 2018 user.debug adblock-[3.5.2]: f_main ::: name: reg_ro, enabled: 0
Thu Jun  7 14:04:32 2018 user.debug adblock-[3.5.2]: f_list ::: name: reg_ro, mode: remove, cnt: 0, in_rc: 0, out_rc: 0
Thu Jun  7 14:04:32 2018 user.debug adblock-[3.5.2]: f_main ::: name: reg_ru, enabled: 0
Thu Jun  7 14:04:32 2018 user.debug adblock-[3.5.2]: f_list ::: name: reg_ru, mode: remove, cnt: 0, in_rc: 0, out_rc: 0
Thu Jun  7 14:04:32 2018 user.debug adblock-[3.5.2]: f_main ::: name: shalla, enabled: 0
Thu Jun  7 14:04:32 2018 user.debug adblock-[3.5.2]: f_list ::: name: shalla, mode: remove, cnt: 0, in_rc: 0, out_rc: 0
[... cut ...]
Thu Jun  7 14:04:47 2018 user.debug adblock-[3.5.2]: f_hash ::: hash_util: /usr/bin/sha256sum, hash: 11311542d2da4bb4ae43addab73151275ee19a60c3e43d09754509213f60298c, out_rc: 1
Thu Jun  7 14:04:47 2018 user.debug adblock-[3.5.2]: f_dnsup::: cache_util: -, cache_rc: 0, cache_flush: 0, cache_cnt: 0, rc: 0
Thu Jun  7 14:04:47 2018 user.debug adblock-[3.5.2]: f_jsnup::: status: enabled, mode: normal mode, cnt: 87654, notify: 0, notify_cnt: 0, notify_pid: -
Thu Jun  7 14:04:48 2018 user.info adblock-[3.5.2]: blocklist with overall 87654 domains loaded successfully (TP-Link Archer C5 v1, LEDE Reboot 17.01.4 r3560-79f57e422d)
```

smartfonreddit, 2018-06-07:
I was able to get the Blacklist entries removed. Each line had to be removed individually. I was mass-selecting and deleting everything earlier. The ads still go through.

dibdot, 2018-06-08:
OK, thanks - the underlying problem still persist:

 ```
Thu Jun  7 14:04:41 2018 user.debug adblock-[3.5.2]: f_tld  ::: source: /tmp/tmp.aFPPPJ/adb_list.overall, cnt: 87654, cnt_srt: -, cnt_tld: -
```

That means that the merge of all lists works (first cnt), the second sort step dies which is unusual - maybe a RAM defect (but that's only a rough guess). To workaround this problem you can try to install the package `coreutils-sort` and repeat the test.

For further testing, please don't use the luci-app, cause this app is still on 3.4.3 release level - take the cli interface via `/etc/init.d/adblock` only.

To check afterwards if the generated blocklist will be used, do something like this:

```
root@blackhole:~# nslookup doubleclick.net
```

you should receive the NXDOMAIN answer:
```
Server:		127.0.0.1
Address:	127.0.0.1#53

** server can't find doubleclick.net: NXDOMAIN
** server can't find doubleclick.net: NXDOMAIN
```

Repeat this test on your connected client, if it's differ make sure that you use your local dnsserver instance for resolving.

smartfonreddit, 2018-06-08:
It works now, thank you!

coreutils-sort was unavailable from command line so I installed coreutils.

ssh root@192.168.1.1 
password
nslookup doubleclick.net

returned the exact same thing you posted. Restarted the service, gave it a few minutes, and it started to block ads. I took the risk and used the old GUI to do that. The manual Blacklist feature is the only non-functional part as of right now.

dibdot, 2018-06-08:
Thanks for your feedback, I close the ticket by now.

closed 2018-06-08 by dibdot