openwrt/packages #6246: sexpect: a light weight Expect like tool

clarkwang, 2018-06-09:
[sexpect]: https://github.com/clarkwang/sexpect

Recently I developed a lightweigh Expect-like tool [sexpect]. I've tested it on OpenWRT 15.05.1, ramips/mt7620.

I'm wondering if it can be integrated into OpenWRT packages.

deqoder, 2018-09-10:
不错，我正好要用，发现openwrt下没这个东东。Good.

dony71, 2022-04-05:
@clarkwang : can you provide README how to compile in openwrt environment?

clarkwang, 2022-04-05:
> @clarkwang : can you provide README how to compile in openwrt environment?

Just follow openwrt's official doc. sexpect has no dependencies on other packages so it's quite easy to compile.

dony71, 2022-04-05:
@clarkwang manage to cross compile but got error execute binary
i open issue on your github

clarkwang, 2022-04-17:
Pull request merged to master: https://github.com/openwrt/packages/pull/18260