openwrt/packages #6282: base-files: actual failsafe doc link in failsafe mode

bobster-galore, 2018-06-12:
Maintainer: @blogic 
Environment: actual OpenWrt version

Description:
The documentation link in failsafe mode should point to https://openwrt.org/docs/guide-user/troubleshooting/failsafe_and_factory_reset .
http://wiki.openwrt.org/doc/howto/generic.failsafe is deprecated. 
s.a. https://github.com/freifunk-berlin/firmware/issues/572

SvenRoederer, 2018-06-14:
for a patch see: http://lists.infradead.org/pipermail/openwrt-devel/2018-June/012804.html

SvenRoederer, 2018-06-14:
@bobster-galore the "base-files" do not come from the openwrt-packages feed. This issue must be opened against OpenWrt-core, as per https://github.com/openwrt/openwrt this needs to be send via https://bugs.openwrt.org

closed 2018-06-18 by jow-