openwrt/packages #6287: mpd-full __cxa_init_primary_exception _ZNSt15__exception_ptr13exception_ptrC1EPv

DrUlysses, 2018-06-13:
Hello, I can't start mpd-full because of this issues

```
mpd
Error relocating /usr/bin/mpd: __cxa_init_primary_exception: symbol not found
Error relocating /usr/bin/mpd: _ZNSt15__exception_ptr13exception_ptrC1EPv: symbol not found
```

here are the dependencies which are installed

```
Package libc (1.1.16-1) installed in root is up to date.
Package glib2 (2.56.1-1) installed in root is up to date.
Package libcurl (7.60.0-3) installed in root is up to date.
Package libpthread (1.1.16-1) installed in root is up to date.
Package libmpdclient (2.11-1) installed in root is up to date.
Package libstdcpp (5.4.0-1) installed in root is up to date.
Package libflac (1.3.2-1) installed in root is up to date.
Package libmad (0.15.1b-4) installed in root is up to date.
Package libvorbisidec (1.0.3-20150104-3) installed in root is up to date.
Package alsa-lib (1.1.6-1) installed in root is up to date.
Package boost (1.67.0-2) installed in root is up to date.
Package boost-container (1.67.0-2) installed in root is up to date.
Package libexpat (2.2.5-1) installed in root is up to date.
Package libaudiofile (0.3.6-4) installed in root is up to date.
Package libfaad2 (2.8.8-1) installed in root is up to date.
Package libffmpeg-mini (3.2.10-3) installed in root is up to date.
Package libid3tag (0.15.1b-4) installed in root is up to date.
Package libmms (0.6.4-4) installed in root is up to date.
Package libogg (1.3.3-1) installed in root is up to date.
Package libsndfile (1.0.28-1) installed in root is up to date.
Package libvorbis (1.3.6-1) installed in root is up to date.
Package libupnp (1.6.19-2) installed in root is up to date.

```

diizzyy, 2018-06-13:
Platform (router model)?
OpenWrt snapshot, release? Did you compile it yourself?
Please don't cross-post. https://forum.lede-project.org/t/mpd-is-not-starting/15623

DrUlysses, 2018-06-13:
Sorry, I will answer here.
Onion Omega 2 plus
Lede Reboot (17.01-SNAPSHOT, r3497-033367e)
compiled what? I have installed mpd-full using opkg 
I have also updated the packages using
opkg list-upgradable | cut -f 1 -d ' ' | xargs opkg upgrade 
(But I can not update kernel modules, because somewhy I can not update the kernel)

diizzyy, 2018-06-13:
You can't mix and mash snapshot builds, they need to be synced. Your best bet would to upgrade to the latest available snapshot or preferably compile your own firmware and packages.

DrUlysses, 2018-06-13:
So how can I upgrade to the latest snaphot? I thinked the opkg command does that.
Also I think this is not a good idea to compile on the Omega2

diizzyy, 2018-06-13:
sysupgrade
https://wiki.openwrt.org/doc/howto/generic.sysupgrade

thess, 2018-06-14:
@DrUlysses - I have answered your posting in the forum and am closing this issue here as a duplicate of:
https://forum.lede-project.org/t/mpd-is-not-starting/15623

Please continue the discussion there.

closed 2018-06-14 by thess